dsmrreader/dsmr-reader #472: Vraag: DSMR-reader zonder internetverbinding

cranberryNL, 2018-05-03:
Niet zozeer een bug of feature, maar een eenvoudige vraag:

Binnenkort krijg ik de sleutel van onze 'nieuwe' woning uit 1924. Begin juli wordt de meterkast verplaatst en krijg ik een nieuwe e-aansluting incl. slimme meter. Nu wil ik graag d.m.v. een RPi icm DSMR-reader m;n slimme meter uitlezen, maar het duurt tot begin augustus voordat de internetverbinding wordt opgeleverd.
Weet iemand toevallig of DSMR Reader werkt zonder internetverbinding? Ik wil de RPi na het booten gebruik laten maken van m'n hotspot op m'n telefoon, zodat 'ie de juiste datum&tijd heeft. Daarna zou er alleen internet zijn als ik aanwezig ben en m'n hotspot activeer.
Gaat dit überhaupt werken?

swerveshot, 2018-05-04:
Volgens mij (correct me if I'm wrong @dennissiemensma) heb je je internetverbinding alleen nodig ten tijde van de installatie en werken de core componenten prima offline.

Er zijn wel een paar functies in de applicatie die een internet verbinding vereisen, maar die zijn optioneel en zul je voor een verse installatie vermoedelijk niet direct gebruiken, zoals:

- Controle op geupdate versies van DSMR Reader
- Backup naar Dropbox
- Koppeling naar externe APIs zoals mindergas.nl en PVOutput

dennissiemensma, 2018-05-04:
Ik sluit me aan bij swerveshot, volgens mij is er alleen even internet nodig voor de installatie. Echter heb ik het nog nooit utigeprobeerd, dus laat het vooral weten of onze aanname klopt!

cranberryNL, 2018-05-05:
Dank voor jullie reacties! 
T.z.t. hoop ik m'n bevindingen hier te melden.
Ben trouwens erg onder de indruk van DSMR Reader. Het ziet er professioneel uit en kan bogen op een hele goede support! Hulde!

dennissiemensma, 2018-05-24:
@cranberryNL is het gelukt?

cranberryNL, 2018-05-24:
@dennissiemensma Nee, duurt nog wel ff. We hebben nu net een week de sleutel van de woning en we zijn erg druk met slopen.
In de eerste week van juli wordt de meterkast verplaatst en wordt de slimme meter geplaatst. Vanaf dat moment probeer ik zo snel mogelijk een raspberry met DSMR reader in gebruik te nemen.
Wordt vervolgd!

dennissiemensma, 2018-05-24:
Bedankt voor je update! In dat geval veel succes en dan hoor ik tzt wel of het werkt. :]

cranberryNL, 2018-07-12:
Zojuist heb ik een eerste getest gedaan. Ik heb eerst een Wifi-profiel  voor de hotspot m'n telefoon op m'n RPi aangemaakt. Zo weet ik zeker dat de RPi de juiste tijd heeft. 
Daarna heb ik de P1-kabel op de RPi aangesloten. Telegrammen komen correct binnen. Daarna heb ik de hotspot uitgezet en ben ik weer naar huis gegaan. In m'n nieuwe woning draait DSMR Reader nu dus 'offline'. Eén dezer dagen hoop ik tijd te hebben om even te kijken hoe e.e.a. loopt.

dennissiemensma, 2018-07-12:
Bedankt voor je update. Ik ben erg benieuwd. :]

cranberryNL, 2018-07-18:
Inmiddels draait DSMR-reader al enige tijd prima. Als ik in het nieuwe huis ben en m'n hotspotfunctie op m'n telefoon activeer, kan ik via m'n telefoon probleemloos het dashboard bereiken en allerlei informatie opvragen.
Uiteraard werken de functies die afhankelijk van een werkende internetverbinding (zoals backup en upload naar mindergas bijv) niet, maar dat is nogal logisch.
Ik ben positief! Nog krappe 1,5 week en dan is de glasvezelverbinding operationeel en kan de applicatie volledig operationeel zijn.

dennissiemensma, 2018-07-19:
Mooi om te horen, bedankt voor de checks en het uitproberen! Succes met de laatste loodjes daar en geniet er van.

closed 2018-07-19 by dennissiemensma