dsmrreader/dsmr-reader #474: Er zijn te veel onverwerkte telegrammen.

darthf1, 2018-05-20:
### Jouw omgeving
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.16`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 2`

### Bug, wens of iets anders

Hi! Ik zit helaas weer met het probleem van teveel onverwerkte telegrammen, inmiddels zijn het er 258.000. Ik zat voorheen op v1.13 en heb een update gedaan naar 1.16. Dit had geen effect op de backlog. Vervolgens heb ik het DSMR_BACKEND process opnieuw gestart, helaas ook zonder effect.

De laatste entry in de dsmrreader.log is dateert van 24-04 en geeft:```

> [2018-04-24 23:53:19,235] ERROR @ apps | publish_json_dsmr_reading() failed: SSL SYSCALL error: EOF detected
> 
> [2018-04-24 23:53:19,250] ERROR @ apps | publish_split_topic_dsmr_reading() failed: connection already closed
Gezien de datum denk ik niet dat deze error hier aan gerelateerd is, maar ik kan deze alsnog niet thuis brengen.

Hoe kan ik dit het beste debuggen?

dennissiemensma, 2018-05-20:
Bedankt voor je melding. Welke informatie zie je in [de logs van supervisor](http://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html)? En dan specifiek die van `dsmr_backend`?

Daarin zou je moeten zien wat die exact aan het verwerken is.

darthf1, 2018-05-20:
Ik denk dat ik de melding heb gevonden!

```
2018-05-20 22:42:17.605073+02:00: Calling backend services
 >>> Uncaught exception :: failed to add old item to the right sibling while splitting block 836 of index "dsmr_consumption_electricityc_currently_delivered_0f28e53a_uniq"
 >>> <bound method AppConfig._on_backend_called_signal of <AppConfig: dsmr_consumption>> ::   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/dispatch/dispatcher.py", line 210, in send_robust
    response = receiver(signal=self, sender=sender, **named)
```

dennissiemensma, 2018-05-20:
De melding lijkt uit PostgreSQL te komen : ` failed to add old item to the right sibling while splitting block 836 of index "dsmr_consumption_electricityc_currently_delivered_0f28e53a_uniq"`

Ik heb deze echter nog nooit eerder gezien en google wijst naar corrupte indexes. Ik zou niet zo 1-2-3 hoe en of dit op te lossen is. 

* Ik neem aan dat je al de complete Pi een keer herstart hebt om te kijken of dat het oplost?

* En anders kun je nog proberen om een databasedump te maken, de oude database te verwijderen een nieuwe aan te maken en hem weer te importeren. De processen in Supervisor moeten daarvoor wel even uit en het inlezen van de backup kan enige tijd in beslag nemen.

darthf1, 2018-05-21:
Opgelost! Ik was gaan kijken wat er aan de hand was met PostgreSQL, en het blijkt dat deze is geupdate van 9.4 naar 9.6, zonder dat het cluster was geupgrade. Nadat ik het PostgreSQL cluster heb geupdate werkt alles weer perfect!

https://gist.github.com/dmitrykustov/27c673ec4f7abd716912e4c830910019

closed 2018-05-21 by darthf1

dennissiemensma, 2018-05-22:
Top, bedankt voor je update en fijn dat het opgelost is.