dsmrreader/dsmr-reader #475: Notify my android error

dbmnl, 2018-05-21:
*Let op: Het DSMR-reader project is gefocused op het uitlezen van de slimme meter en om die data met jou op zoveel mogelijk manieren te delen. Het inlezen van andere (externe) data in DSMR-reader valt buiten de scope van dit project.*

### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.1.6
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 3
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v4

### Bug, wens of iets anders

Bij mijn wekelijkse rondje op mijn Pi zag ik bij Pi-hole opeens een absurd hoog aantal requests naar notify my android toe gaan. Aangezien DSMR reader de enige is die dit gebruikt op mijn netwerk eens gekeken waarom dat plaatsvindt.

[image]
[image]

Het lijkt plaats te vinden sinds middernacht de 19de.

Ik heb daarom de logs er even bijgepakt en zie het volgende langskomen:

2018-05-21 20:30:58.555907+02:00: Calling backend services
 - Creating new notification containing daily usage.
 >>> Uncaught exception :: Notify API call failed: <html><head>
<meta http-equiv="content-type" content="text/html;charset=utf-8">
<title>404 Not Found</title>
</head>
<body text=#000000 bgcolor=#ffffff>
<h1>Error: Not Found</h1>
</body></html>
 (HTTP404)
 >>> <bound method AppConfig._on_backend_called_signal of <AppConfig: dsmr_notification>> ::   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/dispatch/dispatcher.py", line 210, in send_robust
    response = receiver(signal=self, sender=sender, **named)

  File "/home/dsmr/dsmr-reader/dsmr_notification/apps.py", line 20, in _on_backend_called_signal
    dsmr_notification.services.notify()

  File "/home/dsmr/dsmr-reader/dsmr_notification/services.py", line 104, in notify
    send_notification(notification_api_url, notification_settings.api_key, message, str(_('Daily usage notification')))

  File "/home/dsmr/dsmr-reader/dsmr_notification/services.py", line 62, in send_notification
    raise AssertionError('Notify API call failed: {0} (HTTP{1})'.format(response.text, response.status_code))
Command completed. Sleeping for 1 second(s)...

Ik heb daarom de backend service herstart, maar helaas geeft dit mij nog steeds dezelfde melding.

dbmnl, 2018-05-21:
Nou, 
Ik had het ook zelf kunnen vinden denk ik :) Notify my android is ermee gestopt door de aankomende GDPR...
Je kan dat stukje dus uit je code slopen.

------
https://notifymyandroid.com/

Notify my Android is shut down
With great sadness, and after 8 years of operation, NMA will shutdown on May 24th. Due to unacceptably high risks introduced by the GDPR european regulation, and the majority of our users been europeans, we can't justify operations anymore.v

dennissiemensma, 2018-05-22:
Bedankt voor alle informatie. Ik had al een vermoeden dat er iets niet klopte, gezien ik geen notificaties meer kreeg. Ik zal de feature er dan bij de volgende update uithalen.

closed 2018-05-24 by dennissiemensma

dennissiemensma, 2018-05-25:
Uitgebracht in release v1.17, nogmaals bedankt.