dsmrreader/dsmr-reader #476: Foutmelding & supervisorctl pass vergeten

KiefDelicious, 2018-05-25:
Hoi,

niet zozeer een bug, maar een vervelend probleem. Ik heb in mijn front-end de onderstaande foutmelding. 

`Server Error
Exception:
OperationalError: could not connect to server: Connection refused
	Is the server running on host "localhost" (::1) and accepting
	TCP/IP connections on port 5432?
could not connect to server: Connection refused
	Is the server running on host "localhost" (127.0.0.1) and accepting
	TCP/IP connections on port 5432?
Traceback:
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/core/handlers/base.py", line 126, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/views/generic/base.py", line 69, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/views/generic/base.py", line 89, in dispatch
    return handler(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/views/generic/base.py", line 150, in get
    context = self.get_context_data(**kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/views/dashboard.py", line 30, in get_context_data
    frontend_settings = FrontendSettings.get_solo()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/solo/models.py", line 54, in get_solo
    obj, created = cls.objects.get_or_create(pk=cls.singleton_instance_id)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/manager.py", line 82, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/query.py", line 487, in get_or_create
    return self.get(**lookup), False

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/query.py", line 397, in get
    num = len(clone)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/query.py", line 254, in __len__
    self._fetch_all()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/query.py", line 1179, in _fetch_all
    self._result_cache = list(self._iterable_class(self))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/query.py", line 53, in __iter__
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch, chunk_size=self.chunk_size)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/sql/compiler.py", line 1064, in execute_sql
    cursor = self.connection.cursor()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/base/base.py", line 255, in cursor
    return self._cursor()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/base/base.py", line 232, in _cursor
    self.ensure_connection()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/base/base.py", line 216, in ensure_connection
    self.connect()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/utils.py", line 89, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/base/base.py", line 216, in ensure_connection
    self.connect()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/base/base.py", line 194, in connect
    self.connection = self.get_new_connection(conn_params)

[... 49 lines cut ...]
    return self.Migration._meta.db_table in self.connection.introspection.table_names(self.connection.cursor())
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/base/base.py", line 255, in cursor
    return self._cursor()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/base/base.py", line 232, in _cursor
    self.ensure_connection()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/base/base.py", line 216, in ensure_connection
    self.connect()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/utils.py", line 89, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/base/base.py", line 216, in ensure_connection
    self.connect()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/base/base.py", line 194, in connect
    self.connection = self.get_new_connection(conn_params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/postgresql/base.py", line 168, in get_new_connection
    connection = Database.connect(**conn_params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/psycopg2/__init__.py", line 164, in connect
    conn = _connect(dsn, connection_factory=connection_factory, async=async)
django.db.utils.OperationalError: could not connect to server: Connection refused
        Is the server running on host "localhost" (::1) and accepting
        TCP/IP connections on port 5432?
could not connect to server: Connection refused
        Is the server running on host "localhost" (127.0.0.1) and accepting
        TCP/IP connections on port 5432?

`

dennissiemensma, 2018-05-25:
Hoi, bedankt voor je melding. Het lijkt erop dat je PostgreSQL database niet (goed) draait.

Dat je het wachtwoord van dsmr vergeten bent maakt niet uit, wellicht dat je iets door elkaar haalt. Het enige wat je hoeft te doen is op je Pi te verbinden met SSH (of commandline) als `root` of `pi` user. 
Die twee zouden superuserrechten op je hele systeem moeten hebben en daar kun je dus zowel je DB als Supervisor-processen mee herstarten en ook wisselen naar `dsmr` gebruiker.

Dit zou moeten werken:
* SSH naar je Pi als `root` of `pi`.
* Wanneer je `pi` bent, doe `sudo -s`, nu ben je als het goed is `root`.
* Database (her)starten kan als het goed is met `service postgresql restart`.
* Supervisor benaderen kan als root met `supervisorctl`.
* Wisselen naar `dsmr` gebruiken om je webinterface-wachtwoord van de applicatie te wijzigen, kan met: `su - dsmr`. Hiervoor moet je wel eerst je database weer werkend krijgen.

Je zou daar geen wachtwoord voor nodig moeten hebben.

dennissiemensma, 2018-05-26:
Normaal gesproken haalt die de achterstand in. Echter, als de database voor langere tijd niet gewerkt heeft, was het ook niet mogelijk om metingen op te slaan en zijn die gegevens dus niet aanwezig helaas. 

Mocht je nog ergens tegenaan lopen dan hoor ik het graag.

closed 2018-05-26 by dennissiemensma