dsmrreader/dsmr-reader #477: Waardes van gas-metingen worden vanaf 2 april 2018 gecumuleerd opgenomen

svanzon, 2018-05-29:
*Let op: Het DSMR-reader project is gefocused op het uitlezen van de slimme meter en om die data met jou op zoveel mogelijk manieren te delen. Het inlezen van andere (externe) data in DSMR-reader valt buiten de scope van dit project.*

### Jouw omgeving
* DSMR-reader versie 1.16 => Update uitgevoerd naar 1.17 (vandaag)
* Hardware RaspberryPi 3
* DSMR-protocol versie van slimme meter V4 (Kaifa MA304C)

### Bug
Sinds 2 april 2018 (13:00) is de stand van gasmeter in DSMR reader "opbouwend" van waarde.
De uitgelezen gasstanden heb ik hieronder opgenomen (komen uit functie "exporteer gegevens") ter verduidelijking:

2-4-2018 09:00 | 0,714
2-4-2018 10:00 | 0,58
2-4-2018 11:00 | 0,28
2-4-2018 12:00 | 0
2-4-2018 13:00 | 574,54
2-4-2018 14:00 | 574,604
2-4-2018 15:00 | 574,667
2-4-2018 16:00 | 574,667
2-4-2018 17:00 | 574,667
2-4-2018 18:00 | 574,667
2-4-2018 19:00 | 574,667
2-4-2018 20:00 | 574,699

Zoals kan worden gezien wordt vanaf 13:00 een soort van gecummuleerd saldo weergegeven. 
Die blijft maar steeds verder oplopen. Hierdoor zijn de tabellen niet leesbaar in de browser omgeving (printscreen 1). 
[image]

Hij geeft steeds een verbuik van +/- 600 en stijgende. Per heden is de stand 27-5-2018 22:00 | 618,781.
De export is nu dus ook niet bruikbaar om direct analyses mee te doen, deze moet worden omgebouwd (cumulatieve eruit halen).

Bij het status scherm kan ik niets vinden. En ik heb al een keer geprobeerd om de database opnieuw op te bouwen (net zoals dat wordt gedaan als je het tarief hebt aangepast in het verleden. Dit heeft echter geen verschil uitgemaakt, en dat kan ik ook begrijpen aangezien in de database (zoals ik het interpreteer) de verkeerde waardes staan. 

**Wat wel goed gaat**
Het lijkt erop dat hij mijn dagverbruik wel goed berekent, daar komen namelijk logische verbruiksgetallen in naar voren (printscreen 2).
[image]

Daarnaast geeft hij bij het vergelijken van data ook de juist waarde weer (printscreen 3).
[image]

dennissiemensma, 2018-05-30:
Bedankt voor je uitgebreide omschrijving! Ik heb even bij mijzelf gekeken en zelf ik dit issue (helaas) niet. Dus we zullen even moeten kijken wat er anders is.

* Heb je toevallig een nieuwe slimme meter of gasmeter gehad in de afgelopen periode?

* Je hebt nu een CSV-export gedaan op uur-basis en die klopt niet. Zie je hetzelfde issue wanneer je dat op dag-basis doet?

* Wat krijg je te zien als je het volgende uitvoert?
```
sudo su - postgres
psql dsmrreader
select * from dsmr_consumption_gasconsumption order by read_at desc limit 10;
```

Ter referentie, ik zie dit soort aantallen, waarbij de derde kolom de meterstanden zijn en de vierde het verschil met de volgende (wat je dus ook in je CSV verwacht):

```
dsmrreader=# select * from dsmr_consumption_gasconsumption order by read_at desc limit 10;
  id   |        read_at         | delivered | currently_delivered 
-------+------------------------+-----------+---------------------
 21148 | 2018-05-30 15:00:00+00 |  2535.469 |               0.000
 21147 | 2018-05-30 14:00:00+00 |  2535.469 |               0.018
 21146 | 2018-05-30 13:00:00+00 |  2535.451 |               0.000
 21145 | 2018-05-30 12:00:00+00 |  2535.451 |               0.000
 21144 | 2018-05-30 11:00:00+00 |  2535.451 |               0.000
 21143 | 2018-05-30 10:00:00+00 |  2535.451 |               0.000
 21142 | 2018-05-30 09:00:00+00 |  2535.451 |               0.015
 21141 | 2018-05-30 08:00:00+00 |  2535.436 |               0.000
 21140 | 2018-05-30 07:00:00+00 |  2535.436 |               0.000
 21139 | 2018-05-30 06:00:00+00 |  2535.436 |               0.000
```

svanzon, 2018-06-08:
**Zie hieronder jou vragen en mijn antwoorden.**

> Heb je toevallig een nieuwe slimme meter of gasmeter gehad in de afgelopen periode?

Nee, ik heb eerst gewacht tot ik (eind 2016) een nieuwe meter kreeg, daarna alles aangeschaft en geïnstalleerd. Ik heb die dag volgens mij wel een update uitgevoerd, maar dit weet ik niet zeker en weet ook niet meer welke versie het was (volgens mij iets met 1.4 en een week later was er weer een nieuwe update).
> Je hebt nu een CSV-export gedaan op uur-basis en die klopt niet. Zie je hetzelfde issue wanneer je dat op dag-basis doet?

Dit lijkt wel goed te gaan. Bij maandag 2 april (dag dat het - bij het gas - fout gaat) staat geen "Piektarief stroom" maar dit is ook 2e paasdag, dit klopt daarom in mijn ogen.
[image]

> Wat krijg je te zien als je het volgende uitvoert?

Zie hieronder een printscreen vanuit PUTTY. 
De "consumption_delivered" en "current_delivered" lijkt goed te gaan.
Maar de "read_at" waarde is vreemd, het lijkt erop dat de klok niet goed staat?

Datum / tijd lijkt op Raspberry Pi wel goed te staan: 
`pi@raspberrypi:~ $ date`
`Fri  8 Jun 14:57:03 CEST 2018`

[image]

dennissiemensma, 2018-06-08:
Bedankt voor de info, zou je eens een recent telegram uit je meter willen plaatsen hier? Ik denk dat je gelijk hebt met dat het ergens niet goed gaat met de klok. En dan is de vraag waar dat exact mis gaat.

Je kunt eenvoudig een telegram inzien met: `sudo supervisorctl tail -f dsmr_datalogger`.

dennissiemensma, 2018-06-10:
Bedankt voor je update. Het lijkt allemaal te kloppen qua telegrammen, ook de datum, waarvan ik had gedacht dat die mogelijk op 2019 zou staan. De gasstand verandert inderdaad slechts 1x per uur in jouw geval.

Ik zou graag willen weten waar de afwijking is ontstaan en ook hoeveel gasstanden in de 'toekomst' zijn gedaan.

Je kunt deze queries uitvoeren:
```
sudo su - postgres
psql dsmrreader
```

**Alle dagen en aantal gasmetingen per dag:**
```
select date_trunc('day', read_at) as day, count(id) from dsmr_consumption_gasconsumption group by day order by day desc;
```

**De dagen in de toekomst:**
```
select date_trunc('day', read_at) as day, count(id) from dsmr_consumption_gasconsumption where read_at > now() group by day order by day desc;
```

Zie je daar iets vreemds? En wat komt er uit de tweede query?

svanzon, 2018-06-13:
Bij de eerste query zie ik niets bijzonders. Bij vrijwel alle dagen staat 24 tellingen, soms staat er 23.
Rond 2 april zie ik enkel 24 tellingen staan. Er staat wel 1 regel met 1 januari 2019.

Voorbeeld staat hieronder: 

```
dsmrreader=# select date_trunc('day', read_at) as day, count(id) from dsmr_consu                                                                                                              mption_gasconsumption group by day order by day desc;
          day           | count
------------------------+-------
 2019-01-01 00:00:00+00 |    24
 2018-06-13 00:00:00+00 |    20
 2018-06-12 00:00:00+00 |    24
 2018-06-11 00:00:00+00 |    24
 2018-06-10 00:00:00+00 |    24
 2018-06-09 00:00:00+00 |    24
 2018-06-08 00:00:00+00 |    24
......
 2018-04-05 00:00:00+00 |    24
 2018-04-04 00:00:00+00 |    24
 2018-04-03 00:00:00+00 |    24
 2018-04-02 00:00:00+00 |    24
 2018-04-01 00:00:00+00 |    24
 2018-03-31 00:00:00+00 |    24
 2018-03-30 00:00:00+00 |    24
```

Bij de tweede query zie ik onderstaande staan:

```
dsmrreader=# select date_trunc('day', read_at) as day, count(id) from dsmr_consu                                                                        mption_gasconsumption where read_at > now() group by day order by day desc;
          day           | count
------------------------+-------
 2019-01-01 00:00:00+00 |    24
(1 row)
```

dennissiemensma, 2018-06-13:
Waarschijnlijk kun je nog wel in datalogger-tabel zien of daar ook die metingen van 2019 staan, voor `extra_device_timestamp` (de datum/tijd van de gasmeter). Het kan zijn dat deze wat traag is, gezien dit veld geen index heeft in de database:
```
select timestamp, extra_device_timestamp from dsmr_datalogger_dsmrreading where extra_device_timestamp is not null order by extra_device_timestamp desc limit 10;
```

Verder denk ik dat je het beste de dag van 2019 (dus die 24 foute uren) kan verwijderen uit `dsmr_consumption_gasconsumption` en kijken of de volgende dag je gasverbruik beter gelogd wordt. Hiermee kijk je of je de goede data raakt:
```
select * from dsmr_consumption_gasconsumption where read_at::text > '2019-00-01 00:00:00';
```

En hiermee verwijder je het:
```
delete from dsmr_consumption_gasconsumption where read_at::text > '2019-00-01 00:00:00';
```

dennissiemensma, 2018-06-13:
Mocht je overigens nog alle metingen hebben bewaard, en geen opschoning hebben aanstaan bij de Rententie-instellingen, dan kun je nog overwegen om de tabellen `dsmr_stats_daystatistics` en `dsmr_stats_hourstatistics` leeg te gooien.
 En vervolgens `./manage.py dsmr_stats_regenerate_statistics` uitvoeren om ze opnieuw te genereren. 

Dit kan dus alleen wanneer je geen data hebt laten opschonen.

dennissiemensma, 2018-06-28:
@svanzon ben je hier naderhand nog uitgekomen?

svanzon, 2018-06-28:
Ik moet er nog naar kijken. 

Laat het binnenkort weten.

svanzon, 2018-08-21:
Ik ben er weer mee bezig, maar ik kom er niet uit.0
De metingen van 1-1-2019 heb ik kunnen verwijderen, ik kan ze nu ook niet meer opvragen met:

`select date_trunc('day', read_at) as day, count(id) from dsmr_consumption_gasconsumption where read_at > now() group by day order by day desc;
`

 Dan de volgende stap. Klopt het dat ik om <code>dsmr_stats_daystatistics</code> en <code>dsmr_stats_hourstatistics</code> leeg te maken het volgende moet doen (vanuit een nieuwe sessie startend, ingelogd met user "pi"):

```
sudo su - postgres
psql dsmrreader
truncate dsmr_stats_daystatistics
truncate dsmr_stats_hourstatistics
./manage.py dsmr_stats_regenerate_statistics
```

Het lijkt namelijk niet te werken.

Ik heb in het verleden een keer mijn database opgeschoond omdat ik achter de tarieven heb gewijzigd. Hierbij merkte ik toen dat het opbouwen even duurde, maar na bovenstaande te hebben gedaan zie ik niet dat hij hem nog moet opbouwen.

dennissiemensma, 2018-09-14:
Oke prima dan laat ik het voor nu voor wat het is.

closed 2018-09-14 by dennissiemensma