dsmrreader/dsmr-reader #478: Dank en complimenten!

NL-Toontje, 2018-06-01:
* DSMR-reader versie: v1.1.17
* Hardware: RaspberryPi 3B+
* DSMR-protocol versie van slimme meter: v5

Hallo Dennis,
Via deze weg wil ik je bedanken voor het maken en beschikbaar stellen van dsmr-reader. 
Ik heb het sinds gisteren draaien op een Raspberry Pi 3B+ en via een Chinese kabel leest hij onze Iskra AM550 (3-fasen) uit. Dat gaat perfect, daarvoor ook complimenten.
Om de kabel aan de praat te krijgen heb ik met de FTDI tool 'FT-Prog' ook het signaal RTS moeten inverteren. Zonder dat kreeg ik alleen 'connected' maar geen telegrammen.

De punten die mij tot nu toe zijn opgevallen:
- De tijden onder de grafieken staan in 12-uurs formaat (dus met a.m. en p.m.) Kan dat ook in ons 24-uurs formaat?
- De grafieken bestrijken 30 min. Ik zag in andere tickets dat je al bezig was om te kijken of dit (instelbaar) groter te maken is. Ikzelf zou die optie zeer waarderen.
- Ik heb onder 'Statistics', 'Meter Statistics' vanaf de eerste keer opstarten diverse waardes staan, die in 1,5 dag verder niet opgelopen zijn. Wellicht heeft dat met de installatie te maken of een ander soort glitch. Is er een mogelijkheid die waardes te resetten? Dit ziet er raar uit en ik vraag mij af of ze correct/terecht zijn.
- Je geeft aan dat dit project zich richt op het uitlezen van de slimme meter en dat er geen mogelijkheid gaat komen om waardes van andere bronnen (inverters, watermeters etc.) in te lezen. O.a. omdat het lastig testen is.
Uiteraard respecteer ik die keuze.
Echter, mocht je in de toekomst echt niet meer weten wat je aan dsmr-reader zou kunnen verbeteren of toevoegen, weet dan dat mij het kunnen uitlezen van andere bronnen (evt. via standaard S0) een hele waardevolle toevoeging zou lijken.
Ik zal ook uitleggen waarom:
Ik heb twee inverters, een GoodWe die ik via de GoodWe-portal kan monitoren en een oude SMA die helemaal niet te monitoren is. Het kunnen monitoren van die SMA is de eerste reden.
De tweede reden is dat door het kunnen inlezen van inverter data een compleet beeld ontstaat van het energieverbruik en productie binnen dsmr-reader (waar nu de fase[s] waarop teruggeleverd wordt gesaldeerd wordt/worden weergegeven). Hierdoor wordt dsmr-reader 'the only tool you need'.
Als laatste betekent het veel voor mij wanneer ik mijn data binneshuis kan houden en niet afhankelijk hoef te zijn van externe (Chinese) sites om mijn energieverbruik en productie te monitoren.
Ik hoop dat de status van deze laatste feature wijzigt van 'komt nooit' naar 'komt net voor Pasen en Pinksteren op één dag vallen' (-;

dennissiemensma, 2018-06-02:
Hallo Toontje, bedankt voor je vriendelijke woorden.

**Tijdformaat**
Wellicht dat ik het tijdformaat kan wijzigen, het zit er al lang op deze manier in: https://github.com/dennissiemensma/dsmr-reader/blob/master/dsmrreader/formats/en/formats.py
De meeste gebruikers gebruiken een Nederlandstalige browser (ik neem aan dat je Engelstalig gebruikt), dus is het nooit echt een issue geweest nog, maar het klinkt wel als een goed idee dat gelijk te trekken.

**Grafieken**
Op korte termijn zou ik er een optie voor kunnen maken om het handmatig te wijzigen. Echter schaalt dat niet goed, daarvoor zijn een ander soort grafieken nodig en dat kost aardig wat tijd om dat te wijzigen.

**Statistieken**
De statistieken zijn de waardes die de meter aangeeft. Deze zijn niet te resetten voor zover ik weet. De applicatie werkt ze bij na elk telegram dat uitgelezen wordt, maar doorgaans veranderen ze zelden (wellicht na een stroomstoring).
Voor meer informatie zou je een telegram van jezelf kunnen bekijken wat er precies op staat. De mapping staat hier: https://github.com/dennissiemensma/dsmr-reader/blob/master/dsmr_datalogger/dsmr.py
Of nog beter, in de documentatie van netbeheerders: https://www.netbeheernederland.nl/_upload/Files/Slimme_meter_15_a727fce1f1.pdf (rond pagina 21 daar).

**Externe data**
Het maken en vooral testbaar maken van importeren van data uit externe bronnen is een vrij onmogelijke taak voor mij. Daarnaast heb ik net voldoende vrije tijd om dit project te onderhouden zoals het is (met alle openstaande bugs/wensen tot zover). Vandaar dat ik je vrijwel kan garanderen dat ik hier geen verandering in zie komen. 
Ik begrijp dat het fijn is om alles in één applicatie te hebben, maar dat is voor dit project op dit moment echt te veel gevraagd ;-)

dennissiemensma, 2018-06-05:
De tijdsnotatie is straks opgelost via #479.

closed 2018-06-05 by dennissiemensma

NL-Toontje, 2018-06-05:
Hallo Dennis,
Dank voor de update!
Dat het tijdformaat aan mijn taalinstellingen ligt had ik niet aan gedacht...... Ik ben zo gewend alle software in het Engels te hebben dat ik daar niet meer aan denk......

De grafieken zijn nu ook OK, alleen dat halve uur is wat krap. Als het (te) veel werk is dan begrijp ik dat er meer dringende zaken zijn, maar mocht je ooit iets bouwen waarmee (al dan niet handmatig instelbaar) die periode aan te passen is dan zie ik dat wel een keer. Een 'nice to have' zullen we maar zeggen.

Die statistieken moeten dan ontstaan zijn bij de aanleg van de slimme meter (begin dit jaar). Verder schenk ik er dan ook geen aandacht aan.

En over die externe data begrijp ik jouw standpunt. Het woordje 'vrijwel' is voor mij voldoende. Dat betekent dat de deur niet op slot zit en de sleutel is weggegooid, maar dat die deur theoretisch nog open kan (-;

Dank in ieder geval voor de tijd en moeite die je in dit project stopt!

dennissiemensma, 2018-06-06:
Bedankt voor je reactie. Ik heb #482 aangemaakt voor een quick fix van de grafieken, echter schaalt dat maar tot een X aantal items, maar wie weet helpt het al iets.