dsmrreader/dsmr-reader #481: Memory Leak in dsmr_datalogger

ualex73, 2018-06-06:
* DSMR-reader versie: 1.16
* Hardware : x86_86 & Docker
* DSMR-protocol versie van slimme meter: v5
* MQTT publish van de Dagtotalen en Telegram als JSON

De dsmr_datalogger draait nu iets van 10 dagen en het geheugen gebruikt is opgelopen tot boven de 600MB. Er lijkt dus een memory leak in te zitten ... Na een herstart gaat het weer naar 55-60MB.

Hieronder wat statistieken:
2018-05-24 16:40:00,54768,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-05-25 00:00:01,73812,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-05-26 00:00:01,122424,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-05-27 00:00:01,91628,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-05-28 00:00:02,154240,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-05-29 00:00:01,219876,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-05-30 00:00:01,262760,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-05-31 00:00:02,308408,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-01 00:00:01,323776,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-02 00:00:01,414980,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-03 00:00:01,460768,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-04 00:00:01,506264,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-05 00:00:01,552616,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-06 00:00:01,597992,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-06 09:10:01,611144,python3,-u /home/dsmr/app/manage.py dsmr_datalogger

dennissiemensma, 2018-06-06:
Bedankt voor je ticket. Ik gebruik zelf thuis een scriptje dat alles naar de API stuurt, dus ik zal weer eens terug switchen naar de ingebouwde datalogger om te kijken of ik van hetzelfde last heb, of dat het mogelijk gerelateerd is aan Docker.

Zou je in ieder geval willen upgraden naar de laatste versie, v1.18, zodat we dezelfde versie van Django etc draaien?

ualex73, 2018-06-07:
Na het weekend zal ik upgraden naar 1.18 en je op de hoogte houden.

dennissiemensma, 2018-06-08:
Ik draai nu de datalogger nu bijna 48 uur en ik zie geen groot geheugenverbruik. 

* Begrijp ik het goed dat je elke dag een paar MB geheugenverbruik ziet toenemen?
* Hoe draai je bovenstaande lijst uit, want ik wil dat ook wel even doen voor je, ter referentie.

ualex73, 2018-06-12:
Sorry voor de late reply, want ik was een lang weekend weg.

Ik heb nu 1.18.0 geladen en ik ga het geheugen wederom in de gaten houden.

Ik gebruik het volgende script:

#!/bin/bash

list="dsmr_datalogger dsmr_backend"

for app in $list
do

ps ax -o rss,command |grep $app | sort -nr | head -n 1 | awk -v logfile=/var/tmp/mem-$app.txt '
{
    command="date \"+%F %T\"";
    command | getline ts
    close(command);
    printf "%s,%s,%s,",ts,$1,$2 >> logfile
    for(i=3;i<=NF;i++){printf "%s ", $i >> logfile};
    printf "\n" >> logfile
}'

done

dennissiemensma, 2018-06-12:
Thanks, ik zal hem dan zelf ook proberen.

ualex73, 2018-06-15:
Kleine update (v1.18) van het geheugen gebruik:
2018-06-12 15:55:01,52696,python3,-u /home/dsmr/app/manage.py dsmr_datalogger
2018-06-15 00:15:01,114396,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-15 01:15:01,115320,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-15 02:15:01,116292,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-15 03:15:02,117200,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-15 04:15:01,118240,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-15 05:15:01,119208,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-15 06:15:01,120080,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-15 07:15:01,121140,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-15 08:15:01,122116,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-15 09:15:01,122960,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-15 10:15:01,124048,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-15 11:15:01,125024,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-15 12:15:02,125828,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-15 13:15:01,126964,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-15 14:15:01,127932,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 

Dus groeit iets van 800-1000 bytes per uur.

dennissiemensma, 2018-06-15:
Bedankt voor je update. Dat is behoorlijk wat. Hier groeit het to zover met ongeveer 400 bytes in **drie dagen**:

```
2018-06-12 14:42:32,36832,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
...
2018-06-15 17:41:01,37232,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
```

```
2018-06-12 14:42:32,36832,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-12 15:39:44,36832,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-12 15:41:01,36832,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-12 16:41:01,36832,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-12 17:41:01,36832,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-12 18:41:01,36832,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-12 19:41:01,36832,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-12 20:41:01,36832,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-12 21:41:01,36836,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-12 22:41:01,36836,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-12 23:41:01,36840,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-13 00:41:01,36844,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-13 01:41:01,36856,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-13 02:41:01,36860,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-13 03:41:01,36864,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-13 04:41:01,36872,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-13 05:41:01,36884,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-13 06:41:01,36884,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-13 07:41:01,36888,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-13 08:41:01,36896,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-13 09:41:01,36904,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-13 10:41:01,36904,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-13 11:41:01,36908,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-13 12:41:01,36912,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-13 13:41:01,36920,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-13 14:41:02,36928,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-13 15:41:01,36928,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-13 16:41:01,36932,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-13 17:41:01,36936,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-13 18:41:01,36956,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-13 19:41:01,36956,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-13 20:41:01,36960,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-13 21:41:01,36968,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-13 22:41:01,36972,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-13 23:41:01,36976,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-14 00:41:01,36976,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-14 01:41:01,36984,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-14 02:41:01,37004,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-14 03:41:01,37008,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-14 04:41:02,37012,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
[... cut ...]
2018-06-15 14:41:01,37212,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-15 15:41:01,37220,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-15 16:41:01,37228,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
2018-06-15 17:41:01,37232,/home/dsmr/.virtualenvs/dsmrreader/bin/python3,-u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger 
```

ualex73, 2018-06-16:
Na genoeg uren ziet het er zo uit:

2018-06-15 20:25:01,52184,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-15 20:30:02,52216,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
...
2018-06-16 07:55:01,46604,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-16 08:00:02,46608,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 

Zelfs iets omlaag gegaan ;-) Dus het lijkt aan de MQTT berichten te liggen? Ik ze nu 1 voor 1 aanzetten en aankijken.

dennissiemensma, 2018-06-16:
Bedankt voor je toevoeging. Ik gebruik zelf geen MQTT en die wordt inderdaad aangeroepen in het datalogger-proces. Ik zal eens kijken of ik daar wat kan vinden.

ualex73, 2018-06-16:
Het is maar een heel kort interval, maar ik zie wat geheugen gebruik toename van 08:00 - 11:00 (ik heb een van de twee MQTT weer aangezet rond 08h):
2018-06-16 08:00:02,46608,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-16 09:00:01,49068,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-16 10:00:01,50340,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-16 11:00:01,50932,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-16 11:45:02,52180,python3,-u /home/dsmr/app/manage.py dsmr_datalogger

ualex73, 2018-06-17:
Ik heb nu de andere MQTT aangezet:

2018-06-16 22:00:01,58716,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-16 23:00:01,59308,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-17 00:00:02,60112,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-17 01:00:01,60896,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-17 02:00:01,61544,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-17 03:00:01,62820,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-17 04:00:01,63256,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-17 05:00:01,63916,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-17 06:00:01,64440,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-17 07:00:02,64952,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 
2018-06-17 08:00:01,66988,python3,-u /home/dsmr/app/manage.py dsmr_datalogger 

Dit is als "MQTT: Dagtotalen (als JSON) configuratie" is aangezet (de andere MQTT staat uit). De leak lijkt toch meer in de dagtotalen te zitten?

dennissiemensma, 2018-06-17:
Het kan wel eens in beiden zitten, uberhaupt de MQTT library. Want ook voor JSON telegrammen via JSON zorgen er bij mij voor dat ik iets meer dan 100 bytes per uur erbij krijg, terwijl het eerst 400 bytes in 3 dagen was.

ualex73, 2018-06-18:
Ik ben niet thuis in python, maar zou de 2x mqtt publish per seconde een probleem kunnen zijn? De paho library is mogelijk async?

dennissiemensma, 2018-06-19:
Dat zou inderdaad kunnen. Ik ben zelf niet helemaal thuis in async en de paho lib, dus ik ga de docs weer bekijken om te zien of ik ergens optimalisatie heb gemist.

dennissiemensma, 2018-07-22:
Mogelijk dat de refactoring in #417 later iets oplost voor dit issue. Mocht je qua memory leak hier echt dagelijks last van hebben, dan zou ik nog een tussenoplossing kunnen maken in de vorm van een workaround die het backend-proces na x aantal runs stopt (zodat superviso hem weer start, met schoon geheugen).

ualex73, 2018-07-22:
#417 is van harte welkom natuurlijk 👍 

Ik draai vanaf 1.21.1 op Debian Stretch (Xirixiz versie draait nog op Debian Jessie) en het lijkt erop dat ik nu iets minder een memory leak heb, maar ik zal het in de gaten blijven houden.

ualex73, 2018-07-31:
Ik heb dsmr_mqtt nu ook toegevoegd aan het lijst en mijn docker opnieuw gestart met de development branch.

ualex73, 2018-08-01:
Nee, ik zie geen geheugen toename op dit moment ... Het blijft stabiel 👍

dennissiemensma, 2018-08-02:
Mooi! Ik ga er dan vanuit dat het opgelost is.

ualex73, 2018-08-02:
Net nog gecheckt en memory gebruik is stabiel.