dsmrreader/dsmr-reader #483: CRC check fails

marisabele, 2018-06-24:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie : `v1.19`
* Hardware : `RaspberryPi 3B+`
* DSMR-protocol versie van slimme meter : `v4+`

### Bug, wens of iets anders
* CRC check fails for moust of telegramms
Telegram:
```
/Ene5\T210-D ESMR5.0

1-3:0.2.8(50)
0-0:1.0.0(180624230352S)
0-0:96.1.1(4530303438303030303132333330373138)
1-0:1.8.1(000050.134*kWh)
1-0:1.8.2(000027.00*kWh)
1-0:2.8.1(000000.148*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.215*kW)
1,0:2.7.0(00.000*kW)
0-0:96.7.21(00017)
0-0:96.7.9(00006)
1-0:99.97.0(1)(0-0:96.7.19)(180617150629S)(0000004386*s)
1-0:32.32.0(00003)
1-0:52.32.0(00003)
1-0:72.32.0(00002)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00000)
0-0:96.13.0()
1-0:32.7.0(234.0*V)
1-0:52.7.0(234.0*V)
1-0:72.7.0(236.0*V)
1-0:31.7.0(001*A)
1-0:51.7.0(000*A)
1-0:71.7.0(000*A)
1-0:21.7.0(00.215*kW)
1-0:41.7.0(00.000*kW)
1-0:61.7.0(00.00*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303538353330303138393631303138)
0-1:24.2.1(180624230000S)(00009.493*m3)
!6D8C
```
Telegram CRC: 6d8c calculated CRC: e9fb

dennissiemensma, 2018-06-25:
Bedankt voor je melding en de data! 

Klopt het dat je dit telegram 1-op-1 zo hebt gekopieerd? Normaal is het namelijk vrij moeilijk te zeggen waar het in zit (de checksum zegt alleen dat er iets niet klopt, niet wat), maar bij bovenstaande telegram valt me toevallig op dat op regel 12 het tweede teken een komma is, terwijl dat een streepje hoort te zijn volgens mij:

```
# Telegram
1,0:2.7.0(00.000*kW)

# Verwacht
1-0:2.7.0(00.000*kW)
```

Echter, dat gezegd hebbende, als ik het in mijn ontwikkelomgeving in een test gooi en de komma corrigeer naar een streepje, lijkt de checksum alsnog niet te kloppen.

Zien je andere telegrammen er toevallig ook zo uit?

dennissiemensma, 2018-06-25:
En begrijp ik het goed dat de CRC-check niet 100% van je telegrammen afwijst, maar een (groot) deel?

marisabele, 2018-06-25:
The reason for CRC failure is a bad cable. The scrambled data coming in.

closed 2018-06-25 by marisabele

dennissiemensma, 2018-06-26:
Bedankt voor je update, hopelijk doet een andere kabel het beter!