dsmrreader/dsmr-reader #484: API call om huidige versie terug te geven

ualex73, 2018-06-26:
*(Indien van toepassing)*
* DSMR-reader versie: v1.19
* Hardware: Docker
* DSMR-protocol versie van slimme meter: v5

**Wens:**
Op dit moment (lijkt) het niet makkelijk om de huidige DSMR-Reader versie op te halen. Het lijkt alleen te kunnen door in de "__init__.py" versie te kijken. Zou het mogelijk zijn om een API call toe te voegen om de versie via een REST call op te halen? Dit maakt het dan heel makkelijk voor mij om in Home Assistant mij te waarschuwen dat er een nieuwe versie is ;-)

dennissiemensma, 2018-06-26:
Bedankt voor je verzoek. Bedoel je met 'huidige versie' de versie die jij draait of de nieuwste versie op Github (dus een update-check)?

ualex73, 2018-06-26:
De huidige versie die we draaien, dus niet de github check ... Die github check doe ik nu al via een rest sensor ;-)

dennissiemensma, 2018-07-04:
Uitgebracht in nieuwe release: https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.20.0

ualex73, 2018-07-04:
Dank je, ik heb hem direct in mijn docker gegooid :) https://hub.docker.com/r/ualex73/dsmr-reader-docker/

ualex73, 2018-07-04:
Okee ... Getest met v1.20.0 en ik krijg netjes de versie terug ... Dank je voor de change :-)