dsmrreader/dsmr-reader #488: Het is een tijdje geleden sinds het laatste telegram uitgelezen is.

joopy, 2018-07-04:
Ha ik heb een probleem dat de metingen niet meer worden verwerkt door de applicatie:

Berichten kan ik lezen.
(dsmrreader)dsmr@raspberrypi:~/dsmr-reader $ cu -l /dev/ttyUSB0 -s 115200 --parity=none -E q

/KFM5KAIFA-METER

1-3:0.2.8(42)
0-0:1.0.0(180704162353S)
0-0:96.1.1(4530303039313030303030343038303133)
1-0:1.8.1(010462.573*kWh)
1-0:1.8.2(009903.008*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.281*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00007)
0-0:96.7.9(00004)
1-0:99.97.0(4)(0-0:96.7.19)(180601004226S)(0000002307*s)(160215124602W)(0000013125*s)(140113105630W)(0000001468*s)(000101000001W)(2147483647*s)
1-0:32.32.0(00000)
1-0:32.36.0(00000)
0-0:96.13.1()
0-0:96.13.0()
1-0:31.7.0(001*A)
1-0:21.7.0(00.282*kW)
1-0:22.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303136353631323033373736333133)
0-1:24.2.1(180704160000S)(04380.862*m3)
!18B3

status pagina geeft aan:
 P1 DSMR TELEGRAMMEN 
Het is een tijdje geleden sinds het laatste telegram uitgelezen is. 

Meest recente meting: 1 dag geleden
Klik hier voor meer informatie

GASSTATISTIEKEN 
VOOR HET DASHBOARD-OVERZICHT.
Dataverwerking loopt achter: 1 dag, 2 uur geleden

In de log files kan ik niks terugvinden. Draai nu versie 1.20 maar had de zelfde problemen gisteren op 1.19. Heeft er iemand een idee?

dennissiemensma, 2018-07-04:
Wat staat er in de logs van de datalogger (`dsmr_datalogger`)? Het lijkt erop dat er namelijk daar niets meer binnenkomt.

joopy, 2018-07-04:
een aantal oude telegrammen. Maar ook meldingen zoals:

Detected signal #15, exiting on next run...
Detected signal #15, exiting on next run...
Starting infinite command loop...

dennissiemensma, 2018-07-04:
* Ik neem aan dat je al een keer geprobeerd hebt om de processen te herstarten?

* Zie je toevallig op de Statistieken-pagina het aantal geweigerde telegrammen wegens CRC-check toenemen? Staat ergens halverwege de pagina.

joopy, 2018-07-05:
Ja de processen herstart & ook raspberry herstart

Ik zie geen geweigerde CRC telegrammen ik heb die check uitgezet op crc uitgezet in de applicatie na wat google-en op problemen binnen DSMR. Zitten nu totaal 4.205.436 metingen in de database.

Gisteren nogmaals een herstart en toen heeft de applicatie het even kort gedaan 23:10 - 0:19. Als er eventueel problemen zijn met de raspberry dan moet de webinterface het ook niet doen denk ik?

dennissiemensma, 2018-07-05:
Hmmm, klinkt allemaal heel vreemd. Kan het zijn dat je SD-kaart defect is of begint te raken of gebruik je iets anders?

In de logfile van `dsmr_datalogger` zie je dus geen nieuwe logregels binnenkomen? Of nieuwe telegrammen?

dennissiemensma, 2018-07-05:
Oke ik ben benieuwd, ik zou ook niet zo 1-2-3 een andere oplossing weten. Meestal is het de SD-kaart bij 'vage plotselingen issues', maar ik wil niets uitsluiten.

joopy, 2018-07-06:
Nieuwe raspberry pi 2 van de plank gehaald en nieuwe sd card clean install incl database restore. Eerste keer stopte hij na 25min. Nu loopt hij al een paar uur goed?!
Kan het iets te maken hebben met MQTT JSON messages doorsturen? Mijn home assistant een andere raspberry pi stond vast.. 

Eind goed al goed.

dennissiemensma, 2018-07-06:
Hmm ik heb eerlijk gezegd geen idee. Ik gebruik zelf geen MQTT. Kijk komende tijd even aan of het nu opgelost is, anders zal ik het ticket heropenen om verder te debuggen.

closed 2018-07-06 by dennissiemensma