dsmrreader/dsmr-reader #490: Restore sql backup zonder drop database

Martinvdm, 2018-07-08:
Is er ook een mogelijkheid om een .sql in te lezen zonder drop database. Ik wil metingen uit het verleden terug zetten eigenlijk. Wellicht met PSQL Studio of iets dergelijks?

thanks

dennissiemensma, 2018-07-08:
In principe kun je altijd SQL inlezen zonder drop database. Het hangt er alleen erg vanaf of je data terugzet die al bestaat, want dan faalt de import doorgaans.

Wat voor metingen uit het verleden wil je terugzetten? Een oude dump van dsmr-reader zelf? En overlapt dat bestaande gegevens?

Martinvdm, 2018-07-08:
ok thanks. idd oude gegevens, dit overlapt niet. zit 1 dag tussen zonder gegevens.

dennissiemensma, 2018-07-08:
Dan moet het wel goedkomen. Ik weet overigens niet helemaal zeker of de applicatie onverwerkte metingen achteraf nog goed verwerkt, helemaal mocht je het periodiek opschonen van metingen aan hebben staan, maar je kunt het altijd proberen. Eventueel vooraf nog een extra backup maken.

Ik hoor het graag als je er toch niet uit komt.

Martinvdm, 2018-07-08:
ik ga het proberen. zou mooi zijn als het in de GUI mogelijk is :)

dennissiemensma, 2018-07-08:
Is het gelukt? 

Het importeren van backup's via de webinterface zou wellicht kunnen, maar ik introduceer het liefste zo min mogelijk functionaliteit dat de applicatie stuk kan maken. ;-)

Martinvdm, 2018-07-08:
Ik snap het. Ik moet er nog voor gaan zitten ;)

Martinvdm, 2018-07-20:
```
cat /home/docker/dsmrdb/restore.sql | docker exec -i dsmrdb psql -U postgres
```

zie netjes alle ALTER_TABLE voor bij komen, maar zie geen backup data terug in de gui.
restart frontend en db maakt niets uit

dennissiemensma, 2018-07-21:
Gaf de import van data uberhaupt output op de terminal?

Output zoals:

```
COPY 1
 setval 
--------
      1
(1 row)

COPY 942
 setval 
--------
   1208
(1 row)

COPY 22302
 setval 
--------
  25184
(1 row)

COPY 57
 setval 
--------
     57
(1 row)

COPY 21876
 setval 
--------
  21895
(1 row)

COPY 1
 setval 
--------
      1
(1 row)
```

dennissiemensma, 2018-09-06:
@Martinvdm ben je hier nog uitgekomen of blijf je tegen issues aanlopen?

xirixiz, 2018-10-01:
> ```
> cat /home/docker/dsmrdb/restore.sql | docker exec -i dsmrdb psql -U postgres
> ```
> zie netjes alle ALTER_TABLE voor bij komen, maar zie geen backup data terug in de gui.
> restart frontend en db maakt niets uit. restore is 18MB en edited met notepad zie ik zeker weten de data erin.

The role postgres doesn't exist in the dsmrdb docker container. Use the following steps to backup and/or restore data.

Also you could easily create a backup. Values depend on docker/docker-compose user and database variables:
docker-compose stop dsmr
docker exec -t dsmrdb pg_dumpall -c -U dsmrreader > dsmrreader.sql
docker-compose start dsmr

Or drop the database and restore a backup. Values depend on docker/docker-compose user and database variables:
docker-compose stop dsmr
docker exec -t dsmrdb dropdb dsmrreader -U dsmrreader
docker exec -t dsmrdb createdb -O dsmrreader dsmrreader -U dsmrreader
cat dsmrreader.sql | docker exec -i dsmrdb psql -U dsmrreader
docker-compose start dsmr

dennissiemensma, 2018-10-01:
Bedankt voor je aanvulling. @Martinvdm zie bovenstaande.

dennissiemensma, 2018-10-22:
Mocht ik nog iets voor je kunnen doen dan hoor ik het graag en heropen ik het issue weer voor je.

closed 2018-10-22 by dennissiemensma