dsmrreader/dsmr-reader #491: MQTT stales DSMR when MQTT server isn't available.

joopy, 2018-07-08:
When using MQTT with DSMR (eg home assistant) when Home Assistant isn't working 'crash stuck' (100% cpu) DSMR datalogger stales. Tested and can confirm this is the case.

dennissiemensma, 2018-07-08:
Bedankt voor het uitproberen en je melding. Ik zal dan kijken of ik het zelf ook kan reproduceren.

Bedoel je dat je Home Assistant laat crashen of iets om het na te bootsen?

dennissiemensma, 2018-07-22:
@joopy Bedoel je dat je Home Assistant laat crashen of iets, om het na te bootsen? Of zet je server waar HA op draait uit?

joopy, 2018-07-22:
wanneer HA crasht dus mqtt niet beschikbaar dan loopt ook dsmr vast. Zijn twee aparte raspberries.

dennissiemensma, 2018-07-31:
De volgende release bevat een complete refactoring van MQTT. Het MQTT-verhaal is daarmee losgetrokken van de backend en zou niet meer in de weg moeten zitten.

Ik wacht even de volgende release af, zodat je die dan kan uitproberen.

joopy, 2018-08-01:
Ben heel benieuwd!

joopy, 2018-08-02:
Dennis ik heb versie 1.23 draaien voor nu alles goed! Hou je op de hoogte.

dennissiemensma, 2018-08-05:
Voor nu sluit ik deze. Mocht je toch nog tegen issues aanlopen, dan hoor ik het graag!

closed 2018-08-05 by dennissiemensma