dsmrreader/dsmr-reader #494: Extend Usage statistics to include return

helmo, 2018-07-13:
The 'Usage statistics' statistics block on /statistics now shows the highest and lowest wattage for consumption, but not for returned power.

The highest return (per phase) might make a nice addition. The lowest value may be less usefull, as atleast solar and wind will have 0 as minimum.

dennissiemensma, 2018-07-13:
Bedankt voor je ideeen. Hiervoor mag ik sowieso eerst #296 realiseren.

dennissiemensma, 2018-08-05:
Ik zal kijken of ik hier iets voor kan maken. De bestaande opzet doet een rechtstreekse zoekopdracht op alle metingen. 

Echter, sinds de invoering van het automatisch opschonen van data, is dat geen betrouwbare manier meer (en traag). Dus ik zal dit sowieso op de schop gooien.
Waarschijnlijk wordt het iets dat, net zoals bij meterstatistieken, bij het verwerken van elke meting kijkt wat de hoogste en laagste waarden ooit zijn. En dat apart opslaat. Dat maakt het bekijken van de gegevens ook stukken sneller.

dennissiemensma, 2018-08-06:
Ik heb hiervoor iets gemaakt dat het vanaf de volgende release structureel bijhoudt. Bestaande hoogtepunten worden helaas niet meegenomen. Daar staat wel tegenover dat de laden van die statistieken straks instant is.

dennissiemensma, 2018-08-06:
Gemerged naar `development`.

closed 2018-08-06 by dennissiemensma