dsmrreader/dsmr-reader #497: Kleinigheidje: missende vertalingen

jorisvervuurt, 2018-07-16:
Een kleinigheidje: na de update naar v1.21, merkte ik op dat op de Trends-pagina een aantal (Nederlandse) vertalingen missen in de onderste twee energieverbruik-grafieken. Er staat ‘electricity1’ en ‘electricity2’, waar ik ‘Elektriciteit 1 (daltarief)’ en ‘Elektriciteit 2 (piektarief)’ zou verwachten.

Helemaal onbelangrijk: tijdens het laden van de grafieken zie ik kort ‘loading’ staan in plaats van ‘laden’. 😛

dennissiemensma, 2018-07-16:
Bedankt voor je opmerkingen. Ik zal kijken of ik die vertalingen kan oplossen.

dennissiemensma, 2018-07-16:
Ik heb het opgelost in 1d68500c792bab255da9db77ee886af77024aeca. Gaat zo mee met de patch release die ik al klaar had staan.

closed 2018-07-16 by dennissiemensma

jorisvervuurt, 2018-07-16:
Top, dankjewel!