dsmrreader/dsmr-reader #501: q/r: lijn grafiek bij geen teruglevering

Bl4ckviper, 2018-07-20:
Dennis, 

Mooie nieuwe grafieken. Echter 1 vraag. Ik zie dat als er geen teruglevering is de groene lijn toch ook getoond wordt op de lijn van het verbruik.

Andersom is dit niet het geval. Zie screenshot.
Is dit by design?
[image]

dennissiemensma, 2018-07-20:
Bedankt voor je terugkoppeling. Het is niet bewust, maar wel iets waar ik nog eigenlijk een keuze in moet maken.

De oorzaak is dat ik tegen de grafiek zeg "stapel ze maar" en daarbij is de teruglevering altijd bovenop en wordt die (blijkbaar) altijd getoond.

De stapel-optie kan ik ook uitzetten en dan worden de grafieken door elkaar heen getoond. Het contrast is dan minder, maar er zitten dan geen lijnen door elkaar heen. Zie hieronder een voorbeeld van dezelfde data en let dan ook op de schaal links:

**Gestapeld**
[image]

**Normaal**
[image]

Ik denk dat ik het stapelen voor de levering/teruglevering-grafiek dan uitzet en voor fases actief laat.

dennissiemensma, 2018-07-20:
Je kunt zelf ook even kijken of het beter is wanneer je het stapelen uitzet voor je grafiek. 

Deze file kun je dan aanpassen: 
`/var/www/dsmrreader/static/dsmr_frontend/js/dsmr-reader/dashboard/electricity.js` 

Door op regels 60 en 68, de regels met `stack` uit te commenten of weg te halen:
[image]

Vervolgens zou je, na een harde refresh in je browser, een ander soort grafieken moeten zien. Laat maar weten of dat beter werkt (ik heb zelf geen zonnepanelen en heb dit dan ook niet gezien). Ik denk dat ik het uiteindelijk sowieso doorvoer voor die grafiek.

Bl4ckviper, 2018-07-20:
zal dit straks eens probren en je laten weten hoe t ging.

dennissiemensma, 2018-07-20:
Dat ziet er een stuk beter uit zo ja, ik zal het dan doorvoeren voor de volgende release voor die grafiek. Tot die tijd kun je dit rustig laten staan in je bestand.

dennissiemensma, 2018-07-21:
Ik heb het doorgevoerd in b9f766e32524bed0d22bf198c15bf16ab3d5ee15 voor de volgende release, nogmaals bedankt voor de headsup.

Voor het testen met teruglevering (wat ik naboots) heb ik de tool voor het genereren van nep-data wat aangepast, zodat die de werkelijkheid beter reflecteert en ik dit mogelijk een volgende keer eerder zie bij ontwikkelen.

closed 2018-07-21 by dennissiemensma

dennissiemensma, 2018-07-22:
Uitgebracht in `v1.22`.