dsmrreader/dsmr-reader #504: Updates using webinterface

dennissiemensma, 2018-07-21:
Once again try to implement #397, updating using the webinterface.

Executing the update-script is quite easy, however it should never timeout and keep running in the background, even if the webinterface times out or becomes unavailable.

Test these options:
* Update using the `dsmr_backend` process. It will signal itself to reload its process, however when it reaches the point of sending the `SIGHUP` to itself, the update is finished anyway.

* ~~Update using `screen`, running it in the background standalone. Polling whether the PID has finished.~~ This requires installation of `screen`, so I don't think it's a realistic option.

jorisvervuurt, 2018-07-22:
Maybe it’s a bit overkill, but wouldn’t it be easy to simply run a second HTTP server on a different port, that provides API endpoints both for checking if an update is available and actually starting the update (running the deploy script and providing live progress updates through e.g. Server-Sent Events)?

Using simple client-side AJAX, you could then call those API endpoints and simply reload the page when the update has finished. Just a thought. ;-)

dennissiemensma, 2018-07-22:
Bedankt voor je suggestie. Het draaien van een tweede server zou kunnen, maar werkt dan niet voor bestaande installaties. Dan zou iedereen weer handmatig wijzigingen moeten doen in Nginx en Supervisor (net zoals mijn suggestie voor `screen`). Daarnaast weet ik niet of een tweede webserver het issue qua timeouts oplost, want het proces moet blijven draaien in één request.

Het gebruiken van XHR/AJAX is echter wel nog een goede toevoeging. Ik zal kijken of ik het updaten in stapjes kan doen, zodat er ook inderdaad een voortgang te zien is.

jorisvervuurt, 2018-07-22:
Klopt; een extra proces toevoegen vergt dan wel wat handmatige acties. Misschien dat je het als optionele extra kan toevoegen die standaard uit staat. Als iemand dan graag gebruik wil maken van de functie, kun je in de installatiehandleiding de benodigde stappen beschrijven.

Ik ben niet goed genoeg thuis in Python, maar het lijkt mij dat er wel een mogelijkheid zal zijn om een achtergrondproces/thread te starten? Als je die dan op de een of andere manier (ik weet niet wat gebruikelijk is in Python, IPC?) ergens de update-status laat wegschrijven, kun je die vanuit een ander proces uitlezen.

Het request start dat proces enkel, waarna het onafhankelijk z’n werk kan doen. Met AJAX kun je dan de status van het proces opvragen en in de UI tonen. ;-)

dennissiemensma, 2018-07-22:
Ja precies, ik snap waar je op doelt. Ik zal t.z.t. eens wat uitproberen.

dennissiemensma, 2018-09-18:
Ik laat dit toch voor wat het is. Inmiddels is de backlog aan tickets zeer beperkt en verwacht ik sowieso geen ingrijpende releases meer. Los van Python-versies en dependencies die bijgewerkt moeten worden.

closed 2018-09-18 by dennissiemensma