dsmrreader/dsmr-reader #505: SSL/TLS support for MQTT

dennissiemensma, 2018-07-22:
From #417:
> Kunnen we ook SSL/TLS ondersteuning toevoegen aan MQTT? Security wordt steeds belangrijker ;-) (al mijn andere devices doen all MQTT met TLS v1.1/1.2)

dennissiemensma, 2018-07-30:
I've managed to make this work, I'll add three choices:
* Insecure (as is)
* Secure, but no certificate validation (for self-signed)
* Secure, with validation (most secure, but requires a 'real' certificate)

Connecting insecure to MQTTS results in (mosquitto logs):
```
OpenSSL Error: error:140760FC:SSL routines:SSL23_GET_CLIENT_HELLO:unknown protocol
```

Connecting to a self-signed certificate, with the most strict validation will result in:
```
OpenSSL Error: error:14094418:SSL routines:ssl3_read_bytes:tlsv1 alert unknown ca
OpenSSL Error: error:140940E5:SSL routines:ssl3_read_bytes:ssl handshake failure
```

ualex73, 2018-07-31:
Vluchtig getest met no certificate validation, maar toen stopte MQTT met werken (kreeg oneindig veel errors in de mosquitto log) ... Ik ga vanavond verder testen, maar eerst ffies mqtt fixen, want deze weigert om op te starten (nadat ik hem op none-secure heb gezet).

ualex73, 2018-07-31:
Nope, mijn MQTT is nu echt stuk :-(

Mijn mosquitto logging loopt vol met:
1533027238: New connection from 172.18.0.6 on port 1883.
1533027239: Socket error on client <unknown>, disconnecting.

ualex73, 2018-07-31:
Ik heb het nogmaals geprobeerd en nu werkt het wel goed, mijn password manager paste de username/password aan :-( Ik kan alleen niet de valid certificate testen, want ik heb alleen self signed certificaten.

Dus de MQTT/MQTTS support is approved :+1:

dennissiemensma, 2018-07-31:
Top, fijn om te horen dat het uiteindelijk gelukt is en voor jou werkt. Bedankt voor het proberen! Ik zal dit dan terugmergen voor de volgende release. Tot die tijd kun je het blijven draaien.

dennissiemensma, 2018-07-31:
@ualex73 je kunt nu wisselen naar de `development` branch voor nu. Daar zit zowel de refactoring als MQTTS in.

```
sudo su - dsmr
git fetch
git checkout -b development origin/development
./deploy.sh
```

closed 2018-07-31 by dennissiemensma