dsmrreader/dsmr-reader #506: Fasen-grafiek hangt op 'loading'

TampertK, 2018-07-22:
Een schone installatie van DSMR-Reader 1.21.1 toont geen informatie per fase meer in het Dashboard. De fasen-grafiek hang op 'loading', ook na reboots. Na update naar 1.22.0 zelfde probleem.

Onder 'Statistics' ->  'Usage Statistics' staat wel het hoogste verbruik per fase, dus de informatie komt wel binnen.

In het verleden heeft de exact zelfde opstelling in mijn situatie wel gewerkt onder versie 1.19.0

Raspberry Pi3 met Debian Buster.

dennissiemensma, 2018-07-22:
Bedankt voor je melding. Kun je aangeven:
* Of je foutmeldingen krijgt in de console van je browser? (F12 en dan pagina refreshen)
* Of je alleen 3-fasen hebt voor verbruik of ook voor teruglevering?
* De andere grafieken op het dashboard wél goed werken?

dennissiemensma, 2018-07-22:
En aanvullend, krijg je iets te zien als je deze URL aanroept op de applicatie?
* `http://hostname of ip van je Pi` `/xhr/electricity?phases=True`
Als het goed is zie je een shitload aan cijfers.

En als je de bovenste entry bekijkt, die weergegevens wordt op de volgende URL, zijn dan de onderste 6 velden (goed) gevuld?:
* `http://hostname of ip van je Pi` `/admin/dsmr_consumption/electricityconsumption/`

Bij mij is het bijvoorbeeld (geen zonnepanelen):
[image]

TampertK, 2018-07-22:
Dank voor je snelle response :)

- Error: ja, zie:

> jquery-3.3.1.min.js:2 GET http://192.168.1.5/xhr/electricity?phases=True 500 (Internal Server Error)

- Alleen verbruik, geen teruglevering
- Per minuut
- Andere grafiek (totaal verbruik) werkt wel goed
- Aanroepen van de gevraagde URL geeft de volgende error:

> Server Error
> Exception:
> TypeError: float() argument must be a string or a number, not 'NoneType'
> Traceback:
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/core/handlers/base.py", line 126, in _get_response
>     response = wrapped_callback(request, *callback_args, **callback_kwargs)
> 
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/views/generic/base.py", line 69, in view
>     return self.dispatch(request, *args, **kwargs)
> 
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/views/generic/base.py", line 89, in dispatch
>     return handler(request, *args, **kwargs)
> 
>   File "/home/dsmr/dsmr-reader/dsmr_frontend/views/dashboard.py", line 120, in get
>     data['phases_delivered']['l1'].append(float(current.phase_currently_delivered_l1) * 1000)

dennissiemensma, 2018-07-22:
Bedankt voor je terugkoppeling. Ik ben die melding al eerder tegengekomen en als het goed is lost dit het op, wil je dit eens uitproberen?

```
sudo su - dsmr
git fetch
git checkout -b 506-phases-error origin/506-phases-error
./post-deploy.sh
```

Als dat klaar is zou je (na een harde refresh) geen fout meer moeten krijgen op die URL. En als dat goed werkt, zou de grafiek ook weer goed moeten werken. Ik hoor graag of dit het oplost.

TampertK, 2018-07-22:
Zowel de URL en de grafiek werken weer!

dennissiemensma, 2018-07-22:
Fijn, bedankt voor het melden en testen! Ik heb het doorgevoerd in een bugfix-release. 

Je kunt nu weer wisselen naar de hoofd-branch, daar zit de fix inmiddels in:

```
sudo su - dsmr
git checkout master

# Let op, ander script
./deploy.sh
```

closed 2018-07-22 by dennissiemensma