dsmrreader/dsmr-reader #507: Omgaan met nieuwe energiemeter

ewoutdenheijer, 2018-07-23:
Hallo Dennis,

Ik heb vorige maand een nieuwe meter gekregen en het viel mij op dat dsmr-reader hier niet zo goed mee om kan gaan. Het gebruik en de teruglevering worden als negatieve waarden weggeschreven. Technisch gezien klopt dat wel want de tellerstand gaat omlaag, maar er is in het echt geen negatief gebruik of negatieve productie mogelijk tenzij er iets goed fout zit in de meter.

[image]

Zou de applicatie hier mee om kunnen gaan door bijvoorbeeld verlagingen te negeren bij het berekenen?

In de praktijk komt het denk ik nog wel eens voor dat een meter vervangen wordt. Ik mijn geval ben ik al aan de derde meter toe. De communicatiemodule gaat steeds kapot.

dennissiemensma, 2018-07-23:
Bedankt voor je melding. Je bent [niet de eerste](https://github.com/dennissiemensma/dsmr-reader/issues?q=is%3Aissue+meter+vervangen+is%3Aclosed), maar ik heb er nog geen goede oplossing voor. 

Wellicht dat ik er later nogmaals iets voor bedenk, want het moet ook werken voor gebruikers die verhuizen (Waarbij de meterstand niet perse lager of nul is).

Tot dat moment kun je in de admin interface de dag (gisteren) opzoeken bij `Dagstatistieken` en `Uurstatistieken`. De absurde getallen kun je daar op nul zetten of zelf berekenen. Ik verwacht je dat je slechts 1 dag en 1 uur hoeft te wijzigen.

ewoutdenheijer, 2018-07-23:
Bedankt voor je snelle reactie!

Wellicht kan je een wijziging in het serienummer van de meter detecteren en dan het verschil tussen die twee berichten negeren.

dennissiemensma, 2018-07-23:
Daar heb je gelijk in. Echter de uitdaging voor dit issue zit hem meer in het overal aanpassen/splitsen van de meters. Zowel in de backend als frontend. 
Vandaar dat ik het tot dusver even gelaten heb voor wat het is, gezien de handmatige 'fix' relatief weinig werk is.

eterpstra, 2018-11-08:
Hmmm het heeft niet alleen met aanpassen van meter te maken. Ik heb zojuist een backup terugezet van maanden logging (pi was gecrashed), echter staat nu ook alles bij mij in de min bij realtime data dashboard. De dag standen in het archief staan wel netjes / normaal weergeven.

dennissiemensma, 2018-11-08:
Bedankt voor je melding. In principe gebeurt dat eenmalig, al zou ik dat niet verwachten bij het terugzetten van een backup eerlijk gezegd. Laat even weten of het morgen nog steeds speelt en zo ja dan kun je er denk ik het beste een los issue van maken.

dennissiemensma, 2020-02-20:
Ik laat dit voor wat het is. Het is ontzettend lelijk en ik wil het al jaren oplossen, maar in de praktijk is het slechts 5 minuten handmatig fixwerk voor iets wat (per gebruiker gezien) zelden voorkomt.

Voor het voorkomen van negatief verbruik (bij nieuwe meter) kan ik wel iets maken dat alles op 0 zet, maar dat moet dan maar net opvallen bij de gebruiker, zodat die het daarna goed handmatig invult. En andersom (verhuizen naar een huis met een veel hogere meterstand) is het al helemaal niet betrouwbaar te detecteren.

Alle logica voor het berekenen van uur-, dag- en range-totalen is gebaseerd op een aggregatie/range van gemeten verbruik. Als ik per meetpunt moet bijhouden en checken of het nog dezelfde meter is, wordt het vrij complex. Ook als ik simpelweg ga bijhouden dat er ergens een meterwisseling heeft plaatsgevonden moet ik dingen knippen.

Ik laat het daarom dus voor wat het is, voor de ontwikkeltijd die dit kost en het een feature is waar een gebruiker gemiddeld slechts 1x per X jaren tegenaan loopt...

closed 2020-02-20 by dennissiemensma