dsmrreader/dsmr-reader #515: Missing mqtt values

Martinvdm, 2018-08-07:
*Let op: Het DSMR-reader project is gefocused op het uitlezen van de slimme meter en om die data met jou op zoveel mogelijk manieren te delen. Het inlezen van andere (externe) data in DSMR-reader valt buiten de scope van dit project.*

### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.23`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `Docker`

* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v.2`

### Bug  ###

Issue not getting all MQTT values. getting only

dsmr/reading/electricity_delivered_2 06163.036
dsmr/reading/electricity_delivered_1 08867.681
dsmr/reading/electricity_returned_2 07502.092

But should be a lot more as defined in:
MQTT: Telegram (per gesplitst topic) 
whitch is enabled in configuration.
for examle: inncluding Gas, returned_1.
Was not before upgrade.

koelkast88, 2018-08-07:
I do miss the following MQTT (geslitst topic):

Versie: v1.23
Docker on Ubuntu
DSMR v4
electricity_currently_delivered = dsmr/reading/electricity_currently_delivered
electricity_delivered_1 = dsmr/reading/electricity_delivered_1
electricity_delivered_2 = dsmr/reading/electricity_delivered_2

dennissiemensma, 2018-08-07:
Bedankt voor jullie melding. Bedoelen jullie `MQTT: Telegram (per split topic)`? (er zijn meerdere split-topics).

* Wellicht willen jullie even in de broker-instellingen (`MQTT: Broker/connection configuration`), onderaan de pagina, de checkbox `Enable debug logging` aanzetten. 
* En dan kijken of jullie meer informatie zien in de logfile `/var/log/supervisor/dsmr_mqtt.log`. Wanneer ik het lokaal probeer, zie ik onderstaande berichten en *lijkt* het goed. Wat zien jullie?

Als ik dit instel:
```
[mapping]
electricity_delivered_1 = dsmr/telegram/electricity_delivered_1
electricity_returned_1 = dsmr/telegram/electricity_returned_1
electricity_currently_delivered = dsmr/telegram/electricity_currently_delivered
```

Lijkt die dit te doen (formaat wat aangepast):
```
MQTT | Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r0, m157), 
'b'dsmr/telegram/electricity_currently_delivered'', ... (6 bytes)
MQTT | Paho client: on_publish(userdata, mid) None 157

MQTT | Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r0, m158), 
'b'dsmr/telegram/electricity_returned_1'', ... (10 bytes)
MQTT | Paho client: on_publish(userdata, mid) None 158

MQTT | Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r0, m159), 
'b'dsmr/telegram/electricity_delivered_1'', ... (10 bytes)
MQTT | Paho client: on_publish(userdata, mid) None 159

Sleeping 1 sec(s)
```

En ik zie ze ook met een client die de broker uitleest: 
```
mosquitto_sub -t 'dsmr/telegram/electricity_currently_delivered'

# Nep-data
02.248
03.145
00.000
00.000
```

Idem voor gas (`extra_device_delivered = dsmr/telegram/extra_device_delivered` bij mij).

dennissiemensma, 2018-08-07:
Kan het zijn dat bovenin de config het woordje `[mapping]` mist? Ik zie dat dat ook iets stuk kan maken.

Jullie kunnen nog proberen om je huidige config even in een kladblokje te bewaren. En vervolgens het blok met de default-instellingen op de `MQTT: Telegram (per split topic) configuration` pagina te gebruiken om te kijken of dat iets verandert?

Ik hoor graag. Het is vast iets heel kleins wat ik over het hoofd zie, waardoor ik het hier niet zo snel even kan reproduceren.

Martinvdm, 2018-08-07:
Correct. gebruik "MQTT: Telegram (per gesplitst topic)" zoals in de preview in configuration staat aangegeven zou dit : 
```
[mapping]
id = dsmr/reading/id
timestamp = dsmr/reading/timestamp
electricity_delivered_1 = dsmr/reading/electricity_delivered_1
electricity_returned_1 = dsmr/reading/electricity_returned_1
electricity_delivered_2 = dsmr/reading/electricity_delivered_2
electricity_returned_2 = dsmr/reading/electricity_returned_2
electricity_currently_delivered = dsmr/reading/electricity_currently_delivered
electricity_currently_returned = dsmr/reading/electricity_currently_returned
phase_currently_delivered_l1 = dsmr/reading/phase_currently_delivered_l1
phase_currently_delivered_l2 = dsmr/reading/phase_currently_delivered_l2
phase_currently_delivered_l3 = dsmr/reading/phase_currently_delivered_l3
phase_currently_returned_l1 = dsmr/reading/phase_currently_returned_l1
phase_currently_returned_l2 = dsmr/reading/phase_currently_returned_l2
phase_currently_returned_l3 = dsmr/reading/phase_currently_returned_l3
extra_device_timestamp = dsmr/reading/extra_device_timestamp
extra_device_delivered = dsmr/reading/extra_device_delivered
```

Echter krijg ik veel minder in MQTT voorbij.

ik zie met " mosquitto_sub -v -t "#" alles wel voorbij komen echter voor dsmr maar 3 regels.
waar vind ik die debug logging? of kijk ik er over heen?

dennissiemensma, 2018-08-07:
Het staat in op de pagina waar je de broker instelt, dus de hostname, poort, optioneel SSL en credentials.

Martinvdm, 2018-08-07:
in dsmr_mqtt-stdout---supervisor-PHSIT4.log
```
Sleeping  sec(s)

MQTT | Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r0, m10), 'b'dsmr/reading/electricity_delivered_2'', ... (9 bytes)
MQTT | Paho client: on_publish(userdata, mid) None 10
MQTT | Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r0, m11), 'b'dsmr/reading/electricity_delivered_1'', ... (9 bytes)
MQTT | Paho client: on_publish(userdata, mid) None 11
MQTT | Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r0, m12), 'b'dsmr/reading/electricity_returned_2'', ... (9 bytes)
MQTT | Paho client: on_publish(userdata, mid) None 12
Sleeping 1 sec(s)
```

Martinvdm, 2018-08-07:
@koelkast88 gebruik jij ook docker?

dennissiemensma, 2018-08-07:
Bedankt voor je input. Ik heb net de configuratie, die je hierboven hebt geplakt, geprobeerd en dan krijg ik (zoals verwacht) meer dan 3 topics te zien.

Gezien jullie beiden op Docker draaien zou dat een reden kunnen zijn, al zie ik niet hoe dat uit zou moeten maken. Degene die destijds in #417 getest heeft, draait volgens mij ook Docker. Maar die heb ik er nog niet over gehoord.

Wat ik nog kan doen is een aftakking maken waarbij er wat meer debugging geprint wordt in de logs waar de metingen binnenkomen. 

* Gebruik je gewoon de ingebouwde `dsmr_datalogger` of voer je metingen in via de API?

Martinvdm, 2018-08-07:
nee gebruik de ingebouwde data logger. ik zie ook niet in waarom docker zou uitmaken, immers publish je naar mqtt. het is echter wel vreemd dat ik nu na de upgrade maar 3 waardes krijg en mis ik dus de rest. ook mosquitto ziet ze niet, dus worden echt ook niet gepublished. wellicht kan @koelkast88 nog wat zinnigs zeggen.

Martinvdm, 2018-08-07:
als ik overigens "onbewerkte string - raw" aanzet krijg ik het volgende in mosquitto:
```
dsmr/raw /XMX5XMXABCE100091062

0-0:96.1.1(4B414145303031343034313732363133)
1-0:1.8.1(08867.807*kWh)
1-0:1.8.2(06163.628*kWh)
1-0:2.8.1(02926.730*kWh)
1-0:2.8.2(07502.092*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(0000.57*kW)
1-0:2.7.0(0000.00*kW)
0-0:96.13.1()
0-0:96.13.0()
0-1:96.1.0(3238303131303038333134323639363133)
0-1:24.1.0(03)
0-1:24.3.0(180807230000)(08)(60)(1)(0-1:24.2.0)(m3)
(06794.334)
!
```

hier zit wel alles in volgens mij

dennissiemensma, 2018-08-07:
Ik heb een branch gemaakt met wat extra prints. Deze zouden zichtbaar moeten zijn in de datalogger, omdat die de metingen aanmaakt (en die de berichten voor MQTT dan aanmaakt).

```
sudo su - dsmr
git fetch
git checkout -b 515-missing-mqtt-debugging origin/515-missing-mqtt-debugging
./deploy.sh
```

Zoiets zou je dan moeten zien in de `dsmr_datalogger` er tussendoor:
```
0-1:24.2.1(180807230000W)(01704.092*m3)
!783C
[MQTT] Queue message: dsmr/reading/electricity_currently_returned -> 01.966
[MQTT] Queue message: dsmr/reading/phase_currently_returned_l2 -> 00.661
[MQTT] Queue message: dsmr/reading/phase_currently_delivered_l3 -> 00.000
[MQTT] Queue message: dsmr/reading/electricity_returned_2 -> 000000.000
[MQTT] Queue message: dsmr/reading/timestamp -> 2018-08-07T21:14:42Z
[MQTT] Queue message: dsmr/reading/id -> 8134273
[MQTT] Queue message: dsmr/reading/electricity_currently_delivered -> 00.000
[MQTT] Queue message: dsmr/reading/electricity_delivered_2 -> 003408.183
[MQTT] Queue message: dsmr/reading/phase_currently_delivered_l2 -> 00.000
[MQTT] Queue message: dsmr/reading/electricity_delivered_1 -> 005680.305
[MQTT] Queue message: dsmr/reading/extra_device_delivered -> 01704.092
[MQTT] Queue message: dsmr/reading/phase_currently_delivered_l1 -> 00.000
[MQTT] Queue message: dsmr/reading/electricity_returned_1 -> 000000.000
[MQTT] Queue message: dsmr/reading/phase_currently_returned_l1 -> 00.014
[MQTT] Queue message: dsmr/reading/phase_currently_returned_l3 -> 01.291
[MQTT] Queue message: dsmr/reading/extra_device_timestamp -> 2018-08-07T21:00:00Z
Sleeping 1 sec(s)
```

Terug:
```
sudo su - dsmr
git checkout master
./deploy.sh
```

Martinvdm, 2018-08-07:
ik zal als ik tijd hebt (lastig :) ) eens kijken. json en raw string doen het iig wel.

dennissiemensma, 2018-08-07:
* Als je ze daar wel allemaal ziet, dan zit er ergens een fout in het mechanisme van het publiceren.
* Als je ze daar niet allemaal ziet (dus max 3 verschillende), dan zit er een fout in het triggeren van de berichten.

Ik kan het niet te laat maken, dus ik lees het morgenavond wel als jullie iets wijzer zijn geworden. Als jullie nu echt veel data missen, dan zouden jullie t.z.t. nog terug kunnen naar de vorige release.

Martinvdm, 2018-08-07:
yep, zoals gezegd, json en raw zie ik ze wel, zie eerdere logging.
json:
```
dsmr/json {"electricity_returned_2": "07502.092", "electricity_delivered_1": "08867.831", "electricity_delivered_2": "06163.628", "phase_currently_delivered_l3": null, "electricity_currently_returned": "0000.00", "id": 276563, "electricity_returned_1": "02926.730", "phase_currently_delivered_l2": null, "phase_currently_delivered_l1": null, "extra_device_timestamp": "2018-08-07T21:00:00Z", "timestamp": "2018-08-07T21:15:58.856Z", "extra_device_delivered": "06794.334", "electricity_currently_delivered": "0000.57"}
```

gesplits topic is dit de output:
```
dsmr/reading/electricity_delivered_2 06163.628
dsmr/reading/electricity_delivered_1 08867.831
dsmr/reading/electricity_returned_2 07502.092
```

dennissiemensma, 2018-08-08:
Ik ben niet helemaal duidelijk geweest. Ik doelde eigenlijk op de output in de datalogger logs, n.a.v. [dit comment](https://github.com/dennissiemensma/dsmr-reader/issues/515#issuecomment-411205170).

Baseje, 2018-08-08:
Ik mis inderdaad ook veel items waaronder;
dsmr/reading/electricity_currently_delivered

dennissiemensma, 2018-08-08:
@Baseje bedankt voor je melding. Draai je toevallig ook op Docker?

Baseje, 2018-08-08:
> Als je overigens in de gelegenheid bent om wat dieper te debuggen, dat kan op deze manier<https://github.com/dennissiemensma/dsmr-reader/issues/515#issuecomment-411205170>.

Ja, draai op docker.
Zal straks debuggen en de logs delen!

koelkast88, 2018-08-08:
Beste @dennissiemensma , ik zie geen kans om nu te debuggen helaas. Even aandacht voor andere dingen nodig, dan moet ik er mega in gaan duiken (ben lerende). Hopelijk komen jullie eruit! 👍

Baseje, 2018-08-09:
Waar kan ik de logs vinden in de container?
Enig idee?

dennissiemensma, 2018-08-09:
Mijn kennis van Docker is ook helaas zeer beperkt. Ik zou verwachten dat de datalogger logs op de standaard plek in `/var/log/supervisor/` staan.

En anders kan wellicht @ualex73 jullie bijstaan. Die is heeft veel meer ervaring met de Docker setup en hoe het werkt.

ualex73, 2018-08-09:
Yups, in Docker is het dezelfde lokatie "/var/log/supervisor".

Om in de Docker container te gaan, gebruik ik het volgende commando:
docker exec -it dsmr /bin/bash

Dan (filename kan anders zijn natuurlijk):
tail -f dsmr_mqtt-stdout---supervisor-55iZtV.log

Met debug aan, ik zie het als volgt:
MQTT | Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r0, m259), 'b'dsmr/meter-stats/rejected_telegrams'', ... (3 bytes)
MQTT | Paho client: on_publish(userdata, mid) None 259
MQTT | Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r0, m260), 'b'dsmr/meter-stats/electricity_tariff'', ... (1 bytes)
MQTT | Paho client: on_publish(userdata, mid) None 260
MQTT | Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r0, m261), 'b'dsmr/day-totals'', ... (97 bytes)
MQTT | Paho client: on_publish(userdata, mid) None 261
MQTT | Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r0, m262), 'b'dsmr/reading/electricity_currently_returned'', ... (6 bytes)
MQTT | Paho client: on_publish(userdata, mid) None 262
MQTT | Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r0, m263), 'b'dsmr/reading/electricity_delivered_1'', ... (10 bytes)
MQTT | Paho client: on_publish(userdata, mid) None 263
MQTT | Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r0, m264), 'b'dsmr/reading/phase_currently_delivered_l2'', ... (6 bytes)
MQTT | Paho client: on_publish(userdata, mid) None 264
MQTT | Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r0, m265), 'b'dsmr/reading/phase_currently_delivered_l3'', ... (6 bytes)
MQTT | Paho client: on_publish(userdata, mid) None 265
MQTT | Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r0, m266), 'b'dsmr/reading/electricity_delivered_2'', ... (10 bytes)
MQTT | Paho client: on_publish(userdata, mid) None 266
MQTT | Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r0, m267), 'b'dsmr/reading/electricity_returned_2'', ... (10 bytes)
MQTT | Paho client: on_publish(userdata, mid) None 267
MQTT | Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r0, m268), 'b'dsmr/reading/electricity_currently_delivered'', ... (6 bytes)
MQTT | Paho client: on_publish(userdata, mid) None 268
MQTT | Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r0, m269), 'b'dsmr/reading/extra_device_timestamp'', ... (20 bytes)
MQTT | Paho client: on_publish(userdata, mid) None 269
MQTT | Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r0, m270), 'b'dsmr/reading/id'', ... (8 bytes)
MQTT | Paho client: on_publish(userdata, mid) None 270
MQTT | Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r0, m271), 'b'dsmr/reading/timestamp'', ... (20 bytes)
MQTT | Paho client: on_publish(userdata, mid) None 271
MQTT | Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r0, m272), 'b'dsmr/reading/extra_device_delivered'', ... (9 bytes)
MQTT | Paho client: on_publish(userdata, mid) None 272
MQTT | Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r0, m273), 'b'dsmr/reading/phase_currently_delivered_l1'', ... (6 bytes)
MQTT | Paho client: on_publish(userdata, mid) None 273
MQTT | Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r0, m274), 'b'dsmr/reading/electricity_returned_1'', ... (10 bytes)
MQTT | Paho client: on_publish(userdata, mid) None 274
MQTT | Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r0, m275), 'b'dsmr/telegram'', ... (384 bytes)
MQTT | Paho client: on_publish(userdata, mid) None 275
Sleeping 1 sec(s)

Dus alles lijkt bij mij goed te zijn ...

BTW: ik draai wel mijn eigen docker image: https://hub.docker.com/r/ualex73/dsmr-reader-docker/

dennissiemensma, 2018-08-09:
Fijn, bedankt voor je toelichting en uitproberen!

dennissiemensma, 2018-08-09:
In dat geval heb ik zelf geen clue waar dit ik kan zitten. Er is wel duidelijk iets aan de hand, maar ik kan het niet reproduceren en het komt blijkbaar ook weer niet bij iedereen voor... :/

Het enige wat ik helaas kan doen is wachten op jullie debug-output van [deze branch](https://github.com/dennissiemensma/dsmr-reader/issues/515#issuecomment-411205170).

Martinvdm, 2018-08-09:
ik probeer je debugging:
bij .deploy gaat het fout.
```
root@b0dec4677ccc:/home/dsmr/app# sudo su - dsmr
dsmr@b0dec4677ccc:~/app$ git fetch
remote: Counting objects: 12, done.
remote: Compressing objects: 100% (12/12), done.
remote: Total 12 (delta 4), reused 8 (delta 0), pack-reused 0
Unpacking objects: 100% (12/12), done.
From https://github.com/dennissiemensma/dsmr-reader
 * [new branch]      515-missing-mqtt-debugging -> origin/515-missing-mqtt-debugging
dsmr@b0dec4677ccc:~/app$ git checkout -b 515-missing-mqtt-debugging origin/515-missing-mqtt-debugging
D       dsmrreader/provisioning/django/postgresql.py
D       dsmrreader/provisioning/nginx/dsmr-webinterface
Previous HEAD position was ec983aa... Fixed invalid FAQ link in changelog.
Branch 515-missing-mqtt-debugging set up to track remote branch 515-missing-mqtt-debugging from origin.
Switched to a new branch '515-missing-mqtt-debugging'
dsmr@b0dec4677ccc:~/app$ ./deploy.sh

 --- You are currently running version:
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "dsmrreader/__init__.py", line 17, in <module>
    from django.utils.version import get_version
ImportError: No module named django.utils.version

     [!] FAILED to call Django (did you activate the 'dsmrreader' VirtualEnv before running?)

[!] Halted in pre-deployment: Either user ABORTED or script FAILED
```

dennissiemensma, 2018-08-09:
Je zult de virtualenv moeten activeren. Bij de installatie is dat onderdeel van de `.bashrc` van de `dsmr` user, zodat dat automatisch gaat.

Handmatig is het:
```
sudo su - dsmr
source ~/.virtualenvs/dsmrreader/bin/activate
cd ~/dsmr-reader

...
```

Martinvdm, 2018-08-10:
Dit wekt helaas niet in een docker container. vermoedelijk wordt daar geen virtualenv gebruikt

ualex73, 2018-08-10:
Anders kan je mijn dockers lokaal hercompileren.

Haal Docker files op:
`git clone https://github.com/ualex73/dsmr-reader-docker.git`

Verander in 'Dockerfile' de volgende regels:
`ARG TAG="v1.23.0"`
`#ARG TAG="-b development origin/development"`
In:
`#ARG TAG="v1.23.0"`
`ARG TAG="-b 515-missing-mqtt-debugging origin/515-missing-mqtt-debugging"`

En run dan:
`docker build -t ualex73/dsmr-reader-docker:test .`

Dan heb je een nieuw image met deze specfieke branch ;-)

Martinvdm, 2018-08-10:
i am using Xirixiz docker imager. I don't want to break my dsmr setup. i think my debugging will stop here as i don't have the knowledge changing all those things.

ualex73, 2018-08-10:
Mine docker is a fork of Xirixiz, so it should be a drop in place (i only changed the image name when I switched). I forked it, because then I am faster with updates 😜

Martinvdm, 2018-08-10:
ok, though i am using docker-compose, i am not familiair with those git an docker commands

Baseje, 2018-08-10:
[dsmr_mqtt-stdout---supervisor-5__2xy.log](https://github.com/dennissiemensma/dsmr-reader/files/2277137/dsmr_mqtt-stdout---supervisor-5__2xy.log)
Hier is de volledige log,
Download hem maar, kan ik hem daarna verwijderen.

Weet niet wat er precies instaat 👍

dennissiemensma, 2018-08-10:
@Baseje bedankt, je kunt het verwijderen hoor. 

Echter staat hier niet de debugging in die ik zoek, het moet het process zijn dat het triggert, in de meeste gevallen is dat de `dsmr_datalogger`. Ben je in de gelegenheid om die te delen? Die zou in dezelfde map moeten staan.

De regels die ik zoek zijn [degene uit dit comment](https://github.com/dennissiemensma/dsmr-reader/issues/515#issuecomment-411205170). Ik hoef niet de complete log, de laatste paar honderd regels zijn prima. 
Je mag dat deel eventueel mailen naar github @ dennissiemensma.nl, dan staan de gegevens ook niet publiekelijk op Github.

jhoekstra, 2018-08-10:
Hi Dennis,

I have the same problem, using the program directly on a nuc with debian 9. I've checked out the debug version above and enabled debugging on mqtt.
[dsmr_datalogger.log](https://github.com/dennissiemensma/dsmr-reader/files/2279385/dsmr_datalogger.log)
I've attached a snippet of the dsmr_datalogger and dsmr_mqtt logfiles for roughly 10 seconds length each. I notice that the dsmr/reading/electricity_currently_delivered is not  in both the datalogger and mqtt log.

That's probably the hint you're looking for?

dennissiemensma, 2018-08-10:
@jhoekstra bedankt! Klopt het dat je in de logs geen ` End of publish_split_topic_dsmr_reading()` tegenkomt? Die mis in nog.

Ik heb nu nog wat extra debugging toegevoegd om verder te zoomen op de plek waar het mogelijk mis gaat. Wil je de branch lokaal bijwerken en dan nog een snippet van (alleen) de datalogger plakken?

Dit zou afdoende moeten zijn en er zou dan meer debugging moeten komen:
```
sudo su - dsmr
./deploy.sh
```

Als het goed is staat er dan iets als `Checking k:` soms, of (waar ik benieuwd naar ben), `(!) k NOT in topic_mapping`.

jhoekstra, 2018-08-11:
Hi Dennis,

I've updated and took a look again and now I do see the dsmr/reading/electricity_currently_delivered being pushed to mqtt.

[dsmr_datalogger.log](https://github.com/dennissiemensma/dsmr-reader/files/2280114/dsmr_datalogger.log)

I see I need to add the phase_currently_returned_l[1-3] to the mqtt mapping.

Backstory: I'm using this in my domotica to provide a dashboard where current house-usage is shown.

dennissiemensma, 2018-08-11:
@jhoekstra thanks!

Was the log captured **before** you did any changes to the configuration? Because the (missing) `L1-L3` values, are just the phases and should not affect the missing `electricity_currently_delivered` field.

In the log it already seems okay, but I'm not sure whether you've changed anything before capturing it:
```
[MQTT] Queue debug   | publish_split_topic_dsmr_reading(): Checking k: electricity_currently_delivered
[MQTT] Queue message | publish_split_topic_dsmr_reading(): dsmr/reading/electricity_currently_delivered -> 00.160
```

You can try going back to the master version to see if things still work now:

```
sudo su - dsmr
git checkout master
./deploy.sh
```

jhoekstra, 2018-08-11:
Hi Dennis,

I haven't changed configuration to make electricity_currently_delivered work, only the change you committed yesterday was pulled in by ./deploy.sh.
I tried reverting to master, the electricity_currently_delivered wasn't pushed to mqtt anymore. Going to the beta the values came back in mosquitto.

The L1-L3 values were just an observation of values not being pushed to mqtt yet which still need to be mapped, I just added these to "Telegram (per split topic) configuration" successfully.
Now after switching back to master the electricity_currently_delivered  still works, might it have to do with updating the mappingfile which overrides the default values?

dennissiemensma, 2018-08-12:
@jhoekstra thanks for trying and sharing. I don't see how the branch could have fixed anything, as [it only adds print statements](https://github.com/dennissiemensma/dsmr-reader/compare/515-missing-mqtt-debugging), but it's good to hear things are working fine for you now.

dennissiemensma, 2018-08-12:
@Baseje ik ben nog benieuwd naar [dit verhaal](https://github.com/dennissiemensma/dsmr-reader/issues/515#issuecomment-412135207).

Verder zou je nog kunnen kijken of het toevoegen van deze waarden iets oplost, als ik het verhaal van jhoekstra hierboven zo lees:

```
phase_currently_delivered_l1 = dsmr/reading/phase_currently_delivered_l1
phase_currently_delivered_l2 = dsmr/reading/phase_currently_delivered_l2
phase_currently_delivered_l3 = dsmr/reading/phase_currently_delivered_l3
phase_currently_returned_l1 = dsmr/reading/phase_currently_returned_l1
phase_currently_returned_l2 = dsmr/reading/phase_currently_returned_l2
phase_currently_returned_l3 = dsmr/reading/phase_currently_returned_l3
```

Baseje, 2018-08-13:
Waar zou ik dat moeten invullen?
En moet ik die deploy.sh dan nog doen?

dennissiemensma, 2018-08-13:
* Je kunt de waarden invullen bij `MQTT: Telegram (per split topic) configuration` (als ze nog niet bestaan). De relatieve URL in de applicatie is: `/admin/dsmr_mqtt/splittopictelegrammqttsettings/`.

* Voor de extra debugging kun je inderdaad weer gewoon `./deploy.sh` uitvoeren. Deze werkt de code binnen de branch bij naar de laatste versie én herlaadt de applicatie. Vervolgens zou je in de log van `dsmr_datalogger` de debugging moeten zien.

Baseje, 2018-08-13:
Voordat ik iets aanpas staat het standaard zo;
[mapping]
# READING FIELD = TOPIC PATH
id = dsmr/reading/id
timestamp = dsmr/reading/timestamp
electricity_delivered_1 = dsmr/reading/electricity_delivered_1
electricity_returned_1 = dsmr/reading/electricity_returned_1
electricity_delivered_2 = dsmr/reading/electricity_delivered_2
electricity_returned_2 = dsmr/reading/electricity_returned_2
electricity_currently_delivered = dsmr/reading/electricity_currently_delivered
electricity_currently_returned = dsmr/reading/electricity_currently_returned
phase_currently_delivered_l1 = dsmr/reading/phase_currently_delivered_l1
phase_currently_delivered_l2 = dsmr/reading/phase_currently_delivered_l2
phase_currently_delivered_l3 = dsmr/reading/phase_currently_delivered_l3
extra_device_timestamp = dsmr/reading/extra_device_timestamp
extra_device_delivered = dsmr/reading/extra_device_delivered

Hier mis je dus;
phase_currently_returned_l1 = dsmr/reading/phase_currently_returned_l1
phase_currently_returned_l2 = dsmr/reading/phase_currently_returned_l2
phase_currently_returned_l3 = dsmr/reading/phase_currently_returned_l3

Als ik vervolgens de deploy.sh probeer te starten krijg ik;
--- You are currently running version:
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "dsmrreader/__init__.py", line 17, in <module>
    from django.utils.version import get_version
ImportError: No module named django.utils.version

     [!] FAILED to call Django (did you activate the 'dsmrreader' VirtualEnv before running?)

[!] Halted in pre-deployment: Either user ABORTED or script FAILED

Zal wel iets gigantisch verkeerd doen haha

dennissiemensma, 2018-08-13:
Oke, in dat geval eerst maar proberen of het toevoegen van die drie regels het oplost. Wil je eens:
* Die regels toevoegen
* Vervolgens handmatig de applicatie herstarten: `sudo supervisorctl restart all` (kan eventjes duren)
* Kijken of dat iets verandert aan de missende waardes?

Baseje, 2018-08-17:
deploy.sh is me nog niet gelukt tot heden...

Uniwatcher, 2018-08-18:
I have installed DSRM reader directly via the fast method on a Raspberri Pi3. Not via docker.  I'm able to read the values from the day-consumption topic (Day consumption (per split topic config). But listening on the reading topic doesn't give me any values (Telegram (per split topic) configuration). I have enabled ebug logging.

Uniwatcher, 2018-08-18:
This is my dsmr_mqtt.log. MQTT broker is configured correctly. I have a hostname. I get the values of the "day-consumption" topic so that's correct

`dsmr_backend.mixins.StopInfiniteRun
MQTT | No hostname found in settings, restarting in a minute...
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/__init_$
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/__init_$
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/base.py$
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/base.py$
    output = self.handle(*args, **options)
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 35, in handle
    self.data = self.initialize()
  File "/home/dsmr/dsmr-reader/dsmr_mqtt/management/commands/dsmr_mqtt.py", line 20, in initialize
    self.mqtt_client = dsmr_mqtt.services.broker.initialize()
  File "/home/dsmr/dsmr-reader/dsmr_mqtt/services/broker.py", line 19, in initialize
MQTT | No hostname found in settings, restarting in a minute...
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/__init_$
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/__init_$
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/base.py$
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/base.py$
    output = self.handle(*args, **options)
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 35, in handle
    self.data = self.initialize()
  File "/home/dsmr/dsmr-reader/dsmr_mqtt/management/commands/dsmr_mqtt.py", line 20, in initialize
    self.mqtt_client = dsmr_mqtt.services.broker.initialize()
  File "/home/dsmr/dsmr-reader/dsmr_mqtt/services/broker.py", line 19, in initialize
    raise StopInfiniteRun()`

Baseje, 2018-08-20:
Blijf de melding krijgen bij het aftrappen van de deploy.sh.

> --- You are currently running version:
> Traceback (most recent call last):
>   File "<string>", line 1, in <module>
>   File "dsmrreader/__init__.py", line 17, in <module>
>     from django.utils.version [...]

Wanneer verwacht je dit verwerkt te hebben in een nieuwe versie, kan ik namelijk simpel mijn container rebuilden met de nieuwste image waarmee het is opgelost.

snoekieboe, 2018-08-20:
Ik ervaar exact dezelfde issues met DSMR reader 1.23 op een Raspberry pi 3 ( geen docker). Het gedrag is zeer wisselvallig. Soms werkt het rapporteren van de verschillende mqtt berichten (telegram uitgesplits per topic) maar soms stopt het er ook vanzelf weer mee. De enige waardes die dan nog worden gerapporteerd is " Voltage sag count". alle andere topics blijven dan stil........ herstarten van de applicatie of de gehele RPI maakt geen verschil. ook het aanpassen/opnieuw configureren van de MQTT settings maakt geen verschil.

snoekieboe, 2018-08-20:
After running ./deploy.sh once more... the following commits were pulled:

rom https://github.com/dennissiemensma/dsmr-reader
 * [new branch]      467-pvoutput-upload-sync -> origin/467-pvoutput-upload-sync
 * [new branch]      514-archive-compare-pages-in-echarts -> origin/514-archive-compare-pages-in-echarts
 * [new branch]      515-missing-mqtt-debugging -> origin/515-missing-mqtt-debugging
   14484ea..8403567  development -> origin/development

This now seems to have restored all MQTT topics and they are updating just fine.......

Martinvdm, 2018-08-20:
reacreated the dsmr container (i am using docker). now i am missing more values then before. only value i get in mosquitto is:
dsmr/reading/electricity_delivered_2 06198.663

dennissiemensma, 2018-08-21:
Bedankt voor jullie reacties. Ik ben momenteel niet in de gelegenheid om inhoudelijk iets te doen, dus ik kom daar eind deze week op terug.

dennissiemensma, 2018-08-24:
Ik ben inmiddels terug van vakantie. Alvast even een korte reactie. 

Willen jullie proberen of het ophogen van `DSMRREADER_MQTT_MAX_MESSAGES_IN_QUEUE` in de `settings.py` iets verandert aan de situatie?

Dat gaat als volgt:
```
sudo su - dsmr
vi dsmrreader/settings.py
```

* Onderaan de geopende file deze regel toevoegen, als die nog niet in de file staat (standaard staat die er niet en is de waarde `100`):
```
DSMRREADER_MQTT_MAX_MESSAGES_IN_QUEUE = 1000
```

* Tot slot het `dsmr_mqtt` proces herstarten
```
# Terug naar root of pi user
CTRL + D 

sudo supervisorctl restart dsmr_mqtt
```

Ik hoor graag of dat iets verandert aan de situatie. Zo nee, dan is er weer een oorzaak uitgesloten.

dennissiemensma, 2018-08-24:
Waarschijnlijk kan ik overigens vanavond even wat inhoudelijker reageren, want er staan inmiddels wat diverse comments.

dennissiemensma, 2018-08-24:
@Baseje, @Martinvdm, @Uniwatcher 

Willen jullie [bovenstaande](https://github.com/dennissiemensma/dsmr-reader/issues/515#issuecomment-415664893) nog even proberen met `DSMRREADER_MQTT_MAX_MESSAGES_IN_QUEUE`? Het maakt niet uit of je v1.23 of de debugging-branch draait.

----

@Baseje je kunt die missende configuratie uit je comment gewoon toevoegen en dan het `dsmr_mqtt` proces herstarten om te kijken of het helpt:
```
phase_currently_returned_l1 = dsmr/reading/phase_currently_returned_l1
phase_currently_returned_l2 = dsmr/reading/phase_currently_returned_l2
phase_currently_returned_l3 = dsmr/reading/phase_currently_returned_l3
```

---

@snoekieboe bedankt voor je melding. Ik begrijp dus dat het inmiddels (zomaar) is opgelost?

snoekieboe, 2018-08-25:
Het leek inderdaad "zomaar" opgelost maar gisteren zijn de MQTT berichten weer " zomaar" gestopt.
Het is niet zo dat er helemaal geen MQTT berichten meer verzonden worden, maar de actuele gebruiksstatistieken komen nog maar eens per paar minuten binnen.

Ik heb nu de MQTT message queue op 1000 gezet en het MQTT proces herstart....
Resultaat:

* Eerst een ware stortvloed aan MQTT berichten
* Een paar minuten niets
* nu stabiel iedere seconden een update m.b.t. het actueel verbruik

Ik kijk het weer even aan en laat het weten

Martinvdm, 2018-08-25:
Hoi Dennis, ik heb DSMRREADER_MQTT_MAX_MESSAGES_IN_QUEUE op 1000 gezet en en reboot gegeven. supervisorctl zit er bij docker niet in.
Waar ik eerst nog slechts 1 enkele gesplitst topic doorkreeg: dsmr/reading/electricity_delivered_2 krijg ik er nu geen enkele meer. 
Overigens blijft het vreemd dat de Day_Totals wel doorkomen. ook uit en aanzetten van gesplitst topic maakt geen verschil.

dennissiemensma, 2018-08-25:
@Martinvdm welke versie draai je momenteel? `v1.23` of `v1.23.1b1`?

Het deployen van de applicatie lukte je eerder niet, maar als de [checkout wel gelukt is}(https://github.com/dennissiemensma/dsmr-reader/issues/515#issuecomment-411835354), zou die na een herstart in ieder geval de debug-code moeten draaien.

Martinvdm, 2018-08-25:
draai v1.23

Martinvdm, 2018-08-25:
klopt idd. heb nogmaals even je git 515-missing-mqtt-debugging gedraaid, zonder deploy en dan een reboot. draai nu 1.23.1b1

[18 more replies]

closed 2018-08-27 by dennissiemensma