dsmrreader/dsmr-reader #516: Werkende omgeving is gestopt.

asyscogithub, 2018-08-08:
Ik krijg geen verbinding meer met de database

Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 216, in ensure_connection
    self.connect()

Wanneer ik Postgresql opstart met

pg_createcluster 9.6 main --start

krijg ik

Error: cluster configuration already exists

Als je mij kan helpen zou dat geweldig zijn.

dennissiemensma, 2018-08-08:
Bedankt voor je vraag. Ik mis even wat context.

* Heb je een upgrade gedaan of is dit zomaar uit zichzelf gebeurd?
* Draai je gewoon op Raspbian of in Docker?

Het aanmaken van een nieuw cluster is niet nodig om Postgresql te starten. Je zou kunnen proberen om hem zo te starten:
```
sudo /etc/init.d/postgresql start
```

Ik krijg dan zoiets te zien:
```
[ ok ] Starting postgresql (via systemctl): postgresql.service
```

Daarna kun je kijken of de database draait:

```sudo ps faux | grep postgres```

Ik krijg dan zoiets te zien:
```
root     26807  0.0  0.0   4376   540 pts/0    S+   14:12   0:00                      \_ grep postgres
postgres 26657  0.2  2.2 196112 21132 ?        S    14:11   0:00 /usr/lib/postgresql/9.6/bin/postgres -D /var/lib/postgresql/9.6/main -c config_file=/etc/postgresql/9.6/main/postgresql.conf
postgres 26659  0.0  0.2 196112  2376 ?        Ss   14:11   0:00  \_ postgres: 9.6/main: checkpointer process   
postgres 26660  0.0  0.4 196112  4212 ?        Ss   14:11   0:00  \_ postgres: 9.6/main: writer process   
postgres 26661  0.0  0.2 196112  2376 ?        Ss   14:11   0:00  \_ postgres: 9.6/main: wal writer process   
postgres 26662  0.0  0.6 196548  5988 ?        Ss   14:11   0:00  \_ postgres: 9.6/main: autovacuum launcher process   
postgres 26663  0.0  0.3  53508  3364 ?        Ss   14:11   0:00  \_ postgres: 9.6/main: stats collector process   
postgres 26772  0.1  1.3 197064 13260 ?        Ss   14:11   0:00  \_ postgres: 9.6/main: dsmrreader dsmrreader ::1(41056) idle
postgres 26799  0.8  1.4 197064 13316 ?        Ss   14:11   0:00  \_ postgres: 9.6/main: dsmrreader dsmrreader ::1(41088) idle
postgres 26805 11.0  1.4 197104 13708 ?        Ss   14:12   0:00  \_ postgres: 9.6/main: dsmrreader dsmrreader ::1(41100) idle
```

Als de DB draait dan kun je voor de zekerheid nog even de hele applicatie herladen met:
```
sudo supervisorctl restart all
```

asyscogithub, 2018-08-08:
Hoi Dennis,

Dit is uit zichzelf gebeurd, maar er zijn ook andere die bij het kastje kunnen. Die zie dit echter niet doen. Het is een installatie op een debian raspian distributie op een Odroid-XU4. Dit heeft 2 jaar probleemloos gedraaid. Maar goed, als ik de start draai zie ik inderdaad de melding, maar de grep levert niets op. Als ik in /var/log/postgresql/ kijk, is er een lege log file…

Enig idee?

Groeten,

Casper Leenheer

dennissiemensma, 2018-08-08:
* Is er ook geen logfile op deze locatie: `/var/log/postgresql/postgresql-9.6-main.log`?

* Draai je toevallig op een **SD-kaart** die al een tijdje gebruikt wordt? De meeste vreemd/plotselinge issues die ik doorkrijg (en zelf meemaak) zijn falende/corrupt geraakte SD-kaartjes.

* Heb je wel data in de database-folder? `sudo du -msc /var/lib/postgresql/`

Ik krijg bijvoorbeeld:
```
280	/var/lib/postgresql/
280	total
```

asyscogithub, 2018-08-08:
1.  Ja, maar met 0 bytes
  2.  Ik gebruik geen sd maar een ecc disk, maar goed ook die kan kapot. Ik draai ook nog Home Assistant op dit device zonder een probleem…
  3.  Ja, die heb ik. Dit staat er in. Ik heb ook een 9.4 folder.

root@odroidxu4:/var/lib/postgresql/9.6/main# ls
base          pg_dynshmem   pg_replslot   pg_stat_tmp  PG_VERSION
global        pg_logical    pg_serial     pg_subtrans  pg_xlog
pg_clog       pg_multixact  pg_snapshots  pg_tblspc    postgresql.auto.conf
pg_commit_ts  pg_notify     pg_stat       pg_twophase  postmaster.opts

dennissiemensma, 2018-08-08:
Als je geen SD-kaart gebruikt dan, acht ik ook de kans klein dat het de disk is.

* Staan de rechten van `/var/lib/postgresql/` en onderliggende mappen op de juiste `postgres` user?

* Geeft dit output? `systemctl status postgresql.service`
Mijn output:
```
● postgresql.service - PostgreSQL RDBMS
   Loaded: loaded (/lib/systemd/system/postgresql.service; enabled; vendor preset: enabled)
   Active: active (exited) since Wed 2018-08-08 14:11:08 UTC; 3h 14min ago
  Process: 26677 ExecStart=/bin/true (code=exited, status=0/SUCCESS)
 Main PID: 26677 (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/postgresql.service

Aug 08 14:11:08 raspberrypi systemd[1]: Starting PostgreSQL RDBMS...
Aug 08 14:11:08 raspberrypi systemd[1]: Started PostgreSQL RDBMS.
```

* Staat hier nog wat interessants in? `journalctl -u postgresql.service`

dennissiemensma, 2018-08-08:
En aanvullend:

* Wat geeft `pg_lsclusters` als output?

dennissiemensma, 2018-08-08:
En tot slot wat ik nog kan bedenken:

* Dubbelcheck of je schijf niet vol zit met `df -h` (check `Use%`). Andere applicaties moeten er ook last van hebben als dit zo, maar soms is het niet heel triviaal en zorgt het voor vreemde dingen die "opeens" gebeuren. Goed om uit te sluiten als sanity check.

asyscogithub, 2018-08-09:
Hi Dennis,

Status is active

  Process: 1526 ExecStart=/bin/true (code=exited, status=0/SUCCESS)

Dat is al iets. Geen entries in de journal.

Pg_lsclusters geeft:

Ver Cluster Port Status Owner    Data directory               Log file
9.6 main    5432 down   postgres /var/lib/postgresql/9.6/main /var/log/postgresql/postgresql-9.6-main.log

Die is dus down.

Df -use is 65% dus daar ligt het niet aan. 5 GB vrij…

Groeten,

Casper

dennissiemensma, 2018-08-09:
* Had je nog de rechten gecheckt?

Verder heb ik geen dingen meer voor je om te proberen. Je zou nog kunnen kijken of er updates zijn voor postgres en anders hem te verwijderen en herinstalleren. 

Zorg er wel voor dat je je datafolders bewaart, of in ieder geval een SQL-backup hebt die je kunt terugzetten. In het laatste geval kun je ook de datafolders verwijderen (move ze liever) en na herinstallatie de SQL weer importeren.

dennissiemensma, 2018-08-24:
@asyscogithub ben je hier nog verder uitgekomen?

asyscogithub, 2018-08-27:
Ik heb er verder geen tijd meer ingestapt door andere prioriteiten. Maar ik denk dat ik de boel opnieuw ga installeren. In ieder geval bedankt voor de support!

dennissiemensma, 2018-08-27:
Prima, bedankt voor je update. Dan sluit ik deze voor nu.

closed 2018-08-27 by dennissiemensma