dsmrreader/dsmr-reader #517: Verkeerd wachtwoord voor toevoegen dsmr_mqtt?

maikvitesse, 2018-08-16:
### Jouw omgeving
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.23.0`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 3`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v5`

### Vraag
Domme vraag wellicht: ik probeer de ondersteuning voor MQTT te behouden na updaten van DSMR naar versie V1.23.0. Ik doe dit volgens de manier in de FAQ maar het gaat al mis bij de eerste stap.

_Voer nu de volgende commando’s uit als root/sudo-user:_

Ik doe dit met sudo su - dsmr, maar daarna vraagt hij een wachtwoord. Kan me niet herinneren dat ik hiervoor ooit een wachtwoord heb gehad?

Inloggen met de standaard root gebruiker geeft een Authenticatiefout terwijl ik daar wel op ingelogd ben (zie afbeelding).

[image]

dennissiemensma, 2018-08-16:
Bedankt voor je vraag. Je hoeft niet eerst 'sudo su - dsmr' te doen in dit geval.
De commando's voor deze upgrade kun je gewoon als 'pi' user doen. Dan zou het beter moeten gaan.

maikvitesse, 2018-08-16:
Ergens wist ik dat het een stomme vraag zou zijn.
Logisch ook wel dit, haha
Het werk in ieder geval! enorm bedankt voor je snelle reactie

closed 2018-08-16 by maikvitesse

dennissiemensma, 2018-08-24:
Ik was niet in de geledenheid om eerder te reageren, maar fijn dat het opgelost is!