dsmrreader/dsmr-reader #518: Feature request: aflezen telegraaf + gegevens over voltages

cranberryNL, 2018-08-23:
Ik gebruik nu een krappe 2 maanden DSMR-reader en ben er erg over te spreken!

Twee zaken zou ik graag willen aandragen om wellicht toe te voegen:
1. Het bekijken van ruwe data van een P1 telegram (wellicht zit dit er al in, maar ik heb ik niet goed genoeg gezocht)
2. Statistieken m.b.t. het voltage van de fase(n). De netspanning kan behoorlijk varieren per fase, buurt, tijdstip enz. O.a. in geval van zonnepanelen gebeurt het wel eens dat de spanning op een fase zo hoog wordt, dat de omvormer uitschakelt. Het lijkt me een nuttige toevoeging.
Concreet denk aan dan aan bijv. actuele voltage op de fases (zoals in het dashboard ook het actuele opgenomen/teruggeleverde vermogen worden weergegeven) en max/avg/min voltage over bepaalde periode enz.

helmo, 2018-08-29:
Voltages klinkt als een leuke toevoeging, al heb ik die velden nog niet in de ruwe data kunnen herkennen.

Is dat een optioneel veld wat per merk anders is?

cranberryNL, 2018-08-29:
@helmo in mijn telegrammen zie ik de voltages van de 3 fasen terug:
1-0:32.7.0(235.2*V)
1-0:52.7.0(234.8*V)
1-0:72.7.0(234.5*V)

Het zou kunnen zijn dat deze gegevens pas bij v5 van de DSMR zijn gedefinieerd.

glsf91, 2018-11-27:
Voor voltages heb ik ook belangstelling. Dezelfde reden als cranberryNL

dennissiemensma, 2018-11-28:
Bedankt voor het aangeven van je belangstelling. Ik zal de feature sowieso toevoegen, maar ik weet nog niet op welke termijn.

cranberryNL, 2019-01-10:
> 
> 
> Bedankt voor je vragen en goed om te horen dat je er zo tevreden over bent.
> 
> Wat betreft de features:
> 
>     * Het bekijken van de ruwe data is helaas niet mogelijk, omdat de applicatie eigenlijk alleen de data er uit haalt die het gebruikt en opslaat. De rest van het telegram wordt simpelweg genegeerd. Wat je wél kan doen, als alternatief, is via MQTT DSMR-reader de ruwe telegrammen ergens anders heen kan laten sturen en wellicht opslaan. Of je kunt de telegrammen bekijken via de logfiles van `dsmr_datalogger`.
> 
>     * Het uitlezen en opslaan van voltages is een begrijpelijke wens. Echter denk ik dat ik dat pas kan toevoegen wanneer er heel veel vraag naar is, gezien elk stukje extra data aardig wat werk vereist op alle plekken in de applicatie. Ik verwacht daar op korte termijn in ieder geval geen verandering in.

dennissiemensma, 2019-04-25:
Het kunnen inzien van (de laatste) telegram komt in de volgende release. Het deel mbt voltages volgt waarschijnlijk later, gezien daar nog wat werk in zit (data, grafieken, statistieken).

dennissiemensma, 2019-05-20:
Gereleased in [v2.1.0](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v2.1.0)

Woutch, 2019-09-15:
Hoi, 

Ik zoek ook een optie om de spanning te bekijken, mijn meterstatistieken zijn leeg momenteel. Ik zou dus ofwel de telegrammen willen kunnen zien. Of liefst nog kunnen bekijken via api ofzo, dan kan ik ze in een ander programma voorlopig in het oog houden.

Ik heb hetzelfde probleem dat de omvormer zeer veel uit valt op zonnige dagen en zelfs meer dan de helft van de rijd niets staat te doen. Dit zou handig zijn om dan de gegevens door te geven aan de netbeheerder. Deze heb ik trouwens ook deze week ingeschakeld om het probleem langs de straatkant op te lossen. Want ze vallen veel te veel uit hier.

Nu ga ik geregeld zelf eens kijken op mijn slimme meter als er problemen zijn en de spanning zit altijd makkelijk boven de 250V...

Groetjes,
Wouter

dennissiemensma, 2019-09-25:
Heeft een van jullie een voorbeeld waar de voltages in staan? Ik lees wisselende dingen in de documentatie van Netbeheer.

Het is mij onduidelijk welke van de twee varianten het is:
```
1-0:32.7.0.255(220.1*V)
1-0:52.7.0.255(220.2*V)
1-0:72.7.0.255(220.3*V)
```

```
1-0:32.7.0(220.1*V)
1-0:52.7.0(220.2*V)
1-0:72.7.0(220.3*V)
```

glsf91, 2019-09-25:
> Heeft een van jullie een voorbeeld waar de voltages in staan? Ik lees wisselende dingen in de documentatie van Netbeheer.
>

In mijn geval:
1-0:32.7.0(238.8*V)
1-0:52.7.0(238.1*V)
1-0:72.7.0(237.8*V)

cranberryNL, 2019-09-25:
Uit een zojuist opgevangen telegram van mij:
1-0:32.7.0(237.0*V)
1-0:52.7.0(239.1*V)
1-0:72.7.0(236.5*V)

Betreft een 3fase meter DSMR5

From: Dennis Siemensma 
Sent: Wednesday, September 25, 2019 9:51 PM
To: dennissiemensma/dsmr-reader 
Cc: cranberryNL ; State change 
Subject: Re: [dennissiemensma/dsmr-reader] Aflezen telegram + gegevens over voltages (#518)

Heeft een van jullie een voorbeeld waar de voltages in staan? Ik lees wisselende dingen in de documentatie van Netbeheer.

Het is mij onduidelijk welke van de twee varianten het is:

1-0:32.7.0.255(220.1*V)
1-0:52.7.0.255(220.2*V)
1-0:72.7.0.255(220.3*V)
1-0:32.7.0(220.1*V)
1-0:52.7.0(220.2*V)
1-0:72.7.0(220.3*V)

Woutch, 2019-09-25:
Bij mij enkel deze:
1-0:32.7.0(251.0*V)

Heb dan ook maar een 1-fase aansluiting.

dennissiemensma, 2019-09-25:
Bedankt, dan weet ik genoeg!

dennissiemensma, 2019-09-27:
Ik ben nog op zoek naar een goede manier om het weer te geven. Zowel lijn als staaf is so-so.
Lijn:
[image]

Staaf:
[image]

Woutch, 2019-09-27:
Hoi,

Momenteel probeer ik het bij te houden via mijn domotica. Ik heb dan een lijngrafiek die begint rond 220v en stopt op 260v. Ik weet dat mijn omvormer uitvalt op 254v. En door op 220v te starten zie ik meer details.

Hoe meer ik teruglever hoe hoger de spanning.

Ik heb trouwens de netbeheerder hiervan op de hoogte gebracht en die gaan een afspraak maken om alles te komen nakijken. Maar dit kan nog paar weken duren door de drukte heb ik te horen gekregen.

Hopelijk kunnen ze dit in de winter ook nakijken als er minder opgewekt word en dit wss ook geen probleem is. :-)

Vandaar dat ik dit wil bijhouden om te kunnen "bewijzen" wat er gebeurt.

Groetjes,
Wouter

glsf91, 2019-09-27:
Lijn vind ik veel duidelijker als staaf. Ik weet niet of het mogelijk is maar ik zou de schaal van de y-as auto maken. Een verschil van een paar volt op een schaal van 300V is niet echt duidelijk.
Als dat niet kan zou ik een schaal hanteren van 200V-270V oid.

dennissiemensma, 2019-09-27:
Ik heb de optie gevonden voor autoschalen, dus dat lijkt me zeker een goede suggestie!

Voorbeeld (met iets minder fluctuerende nepdata):
[image]

Dan doe ik hem in de eerste instantie zo en moeten jullie straks zelf maar een tijdje testdraaien of dat bevalt. Ik heb helaas een dsmr v4 meter, dus ik kan het zelf alleen simuleren.

cranberryNL, 2019-09-27:
Een lijn is inderdaad duidelijker dan een staafdiagram. Zeker de autoscale-optie is mooi.

Waardevolle toevoeging, waarvoor dank!!

From: Dennis Siemensma 
Sent: Friday, September 27, 2019 10:20 PM
To: dennissiemensma/dsmr-reader 
Cc: cranberryNL ; State change 
Subject: Re: [dennissiemensma/dsmr-reader] Aflezen gegevens over voltages (#518)

Ik heb de optie gevonden voor autoschalen, dus dat lijkt me zeker een goede suggestie!

Voorbeeld (met iets minder fluctuerende nepdata):

Dan doe ik hem in de eerste instantie zo en moeten jullie straks zelf maar een tijdje testdraaien of dat bevalt. Ik heb helaas een dsmr v4 meter, dus ik kan het zelf alleen simuleren.

dennissiemensma, 2019-09-27:
Ik draai de ontwikkelversie nu zelf even een dag of wat. Als dat goed gaat, kunnen meer liefhebbers het proberen.

dennissiemensma, 2019-09-29:
Liefhebbers om het uit te proberen?

```
sudo su - dsmr
git fetch
git checkout -b development origin/development
./post-deploy.sh
```

Eventueel terug naar de vorige versie:
```
sudo su - dsmr
sh dsmrreader/provisioning/downgrade/2.5.0-2.4.0.sh
git checkout master
./post-deploy.sh
```

cranberryNL, 2019-09-30:
Dankje!

Zojuist geüpdate naar deze versie.  Heb geen problemen of afwijkingen geconstateerd. Wel viel me op dat de autoscaling van y-as (nog) niet werkt:

Is dit een setting in de configuratie of is deze nog niet ingebouwd in de huidige rc? 

From: Dennis Siemensma 
Sent: Sunday, September 29, 2019 1:01 PM
To: dennissiemensma/dsmr-reader 
Cc: cranberryNL ; State change 
Subject: Re: [dennissiemensma/dsmr-reader] Aflezen gegevens over voltages (#518)

Liefhebbers om het uit te proberen?

sudo su - dsmr
git fetch
git checkout -b development origin/development
./post-deploy.sh
Eventueel terug naar de vorige versie:

sudo su - dsmr
sh dsmrreader/provisioning/downgrade/2.5.0-2.4.0.sh
git checkout master
./post-deploy.sh

cranberryNL, 2019-09-30:
Waarschijnlijk te vroeg geroepen. Autoscaling werkt prima nu:

From: Dennis Siemensma 
Sent: Sunday, September 29, 2019 1:01 PM
To: dennissiemensma/dsmr-reader 
Cc: cranberryNL ; State change 
Subject: Re: [dennissiemensma/dsmr-reader] Aflezen gegevens over voltages (#518)

Liefhebbers om het uit te proberen?

sudo su - dsmr
git fetch
git checkout -b development origin/development
./post-deploy.sh
Eventueel terug naar de vorige versie:

sudo su - dsmr
sh dsmrreader/provisioning/downgrade/2.5.0-2.4.0.sh
git checkout master
./post-deploy.sh

dennissiemensma, 2019-09-30:
Het kan zijn dat er nog wat auto-scaling mist bij updates en dat die alleen initieel goed laadt. En dat je nu na een tijdje draaien genoeg verschillende waardes hebt.

Ik zal eens testen of dat ook zo is. En of het te fixen is.

dennissiemensma, 2019-09-30:
Ik neem deze morgen mee in de volgende release.

closed 2019-09-30 by dennissiemensma

dennissiemensma, 2019-10-01:
Uitgebracht in [v2.5](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v2.5.0)

cranberryNL, 2019-10-03:
Vanavond heb ik onderstaande uitgevoerd om weer terug te keren naar de normale branch. Helaas werkt e.e.a. niet meer. Ik zie nu dit als ik naar DSMR-reader browse:

Server Error
Sorry, something unexpected happened. 
Exception:
ProgrammingError: column dsmr_datalogger_dataloggersettings.track does not exist
LINE 1: SELECT "dsmr_datalogger_dataloggersettings"."id", "dsmr_data...
                                                          ^
Traceback:
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/handlers/base.py", line 113, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/views/generic/base.py", line 71, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/views/generic/base.py", line 97, in dispatch
    return handler(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/views/generic/base.py", line 158, in get
    context = self.get_context_data(**kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/views/dashboard.py", line 30, in get_context_data
    context_data['datalogger_settings'] = DataloggerSettings.get_solo()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/solo/models.py", line 54, in get_solo
    obj, created = cls.objects.get_or_create(pk=cls.singleton_instance_id)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/manager.py", line 82, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/query.py", line 538, in get_or_create
    return self.get(**kwargs), False

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/query.py", line 402, in get
    num = len(clone)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/query.py", line 256, in __len__
    self._fetch_all()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/query.py", line 1242, in _fetch_all
    self._result_cache = list(self._iterable_class(self))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/query.py", line 55, in __iter__
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch, chunk_size=self.chunk_size)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 1100, in execute_sql
    cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/utils.py", line 67, in execute
    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/utils.py", line 76, in _execute_with_wrappers
    return executor(sql, params, many, context)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/utils.py", line 89, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)

Wat zou ik het beste kunnen doen?

From: Dennis Siemensma 
Sent: Tuesday, October 1, 2019 5:30 PM
To: dennissiemensma/dsmr-reader 
Cc: cranberryNL ; State change 
Subject: Re: [dennissiemensma/dsmr-reader] Aflezen gegevens over voltages (#518)

Degenen die gewisseld waren van branch kunnen nu dit doen:

sudo su - dsmr
git fetch
git checkout master
./post-deploy.sh

cranberryNL, 2019-10-03:
Toen ik ivm aanpassing logging level ./post-deploy.sh uitvoerde, zag ik een aantal keren deze melding voorbij komen:
ValueError: dictionary update sequence element #0 has length 1; 2 is required

Wellicht geeft dit een idee vwb de oorzaak?

From: Dennis Siemensma 
Sent: Tuesday, October 1, 2019 5:30 PM
To: dennissiemensma/dsmr-reader 
Cc: cranberryNL ; State change 
Subject: Re: [dennissiemensma/dsmr-reader] Aflezen gegevens over voltages (#518)

Degenen die gewisseld waren van branch kunnen nu dit doen:

sudo su - dsmr
git fetch
git checkout master
./post-deploy.sh

dennissiemensma, 2019-10-04:
Welke versie staat er nu rechtsbovenin?

Zou je anders nog een normale deploy willen proberen? Wellicht is de branchinformatie lokaal bij je toch nog outdated.

```
sudo su - dsmr
./deploy.sh
```

cranberryNL, 2019-10-04:
Mijn dank is groot. Alles werkt weer na een redeploy.

Bij het begin van de ./deploy.sh viel me op dat ik blijkbaar een aantal versies achterliep:
--- You are currently running version:
2.1

Het is in ieder geval weer opgelost en daar ben ik erg blij mee. Dank voor de snelle hulp!

From: Dennis Siemensma 
Sent: Friday, October 4, 2019 7:27 AM
To: dennissiemensma/dsmr-reader 
Cc: cranberryNL ; State change 
Subject: Re: [dennissiemensma/dsmr-reader] Aflezen gegevens over voltages (#518)

Welke versie staat er nu rechtsbovenin?

Zou je anders nog een normale deploy willen proberen? Wellicht is de branchinformatie lokaal bij je toch nog outdated.

sudo su - dsmr
./deploy.sh

neverstoppedaskingwhy, 2019-10-06:
Hoi Dennis, bedankt voor het toevoegen van de spanning, een echte aanvulling! 

Misschien is het een idee voor de toekomst, om onder de statistics ook de all time highest voltage en all time lowest voltage, te tonen (all time lowest zou natuurlijk niet onder een bepaalde minimum waarde mogen komen, bijv. 0)

dennissiemensma, 2019-10-07:
Ik zal er over nadenken. Voor nu houd ik het alleen bij de huidige statistieken. 

Je kunt het in je eigen systeem checken met:
```
sudo su - postgres
psql dsmrreader

# Hoogste
select timestamp, phase_voltage_l1 from dsmr_datalogger_dsmrreading order by phase_voltage_l1 desc limit 1;
select timestamp, phase_voltage_l2 from dsmr_datalogger_dsmrreading order by phase_voltage_l2 desc limit 1;
select timestamp, phase_voltage_l3 from dsmr_datalogger_dsmrreading order by phase_voltage_l3 desc limit 1;

# Laagste
select timestamp, phase_voltage_l1 from dsmr_datalogger_dsmrreading order by phase_voltage_l1 asc limit 1;
select timestamp, phase_voltage_l2 from dsmr_datalogger_dsmrreading order by phase_voltage_l2 asc limit 1;
select timestamp, phase_voltage_l3 from dsmr_datalogger_dsmrreading order by phase_voltage_l3 asc limit 1;
```