dsmrreader/dsmr-reader #519: Installatie van Postgres (met back-up terugzetten lukt niet)

svanzon, 2018-09-14:
*Let op: Het DSMR-reader project is gefocused op het uitlezen van de slimme meter en om die data met jou op zoveel mogelijk manieren te delen. Het inlezen van andere (externe) data in DSMR-reader valt buiten de scope van dit project.*

### Jouw omgeving
* DSMR-reader versie: Nog niet van toepassing, ben bezig met de handleiding (14-9-2018)
* Hardware RaspberryPi 3
* DSMR-protocol versie van slimme meter V4 (Kaifa MA304C)

### Bug, wens of iets anders
* Ik heb de stappen van de installatie gevolg en heb direct een database back-up proberen terug te zetten (gebruik zoals aanbevolen postgres) => Ofwel enkel stap 1. (zie tevens de logfile waarbij je mijn stappen kan zien - [Logfiles installatie DSMR reader (14-9-2018).pdf](https://github.com/dennissiemensma/dsmr-reader/files/2381425/Logfiles.installatie.DSMR.reader.14-9-2018.pdf) -). Hieronder kort de commando's die ik heb ingegeven:

sudo apt-get install -y postgresql postg

- resql-server-dev-all
- pi@raspberrypi:~ $ sudo sudo -u postgres createuser -DSR dsmrreader
- pi@raspberrypi:~ $ sudo sudo -u postgres createdb -O dsmrreader dsmrreader
- pi@raspberrypi:~ $ sudo sudo -u postgres psql -c "alter user dsmrreader with password 'dsmrreader';"

Dit lijkt allemaal goed te gaan, vervolgens ook het starten van de back-up (was even aan het puzzelen met "mounten". Maar dit lukte, en de restore begon.
Echter vervolgens typte ik iets in het cmd scherm (ls) waarna er een foutmelding kwam.

> WARNING:  terminating connection because of crash of another server process
> DETAIL:  The postmaster has commanded this server process to roll back the current transaction and exit, because another server process exited abnormally and possibly corrupted shared memory.
> HINT:  In a moment you should be able to reconnect to the database and repeat your command.
> server closed the connection unexpectedly
>         This probably means the server terminated abnormally
>         before or while processing the request.
> invalid command \.
> no connection to the server
> connection to server was lost
> pi@raspberrypi:~/synology $ ls
> dsmr.sql.gz  @eaDir  oud
> pi@raspberrypi:~/synology $ zcat /home/pi/synology/dsmr.sql.gz | sudo sudo -u postgres psql dsmrreader
> psql: FATAL:  the database system is in recovery mode
> pi@raspberrypi:~/synology $ zcat /home/pi/synology/dsmr.sql.gz | sudo sudo -u postgres psql dsmrreader
> psql: FATAL:  the database system is in recovery mode
> pi@raspberrypi:~/synology $ zcat /home/pi/synology/dsmr.sql.gz | sudo sudo -u postgres psql dsmrreader
> psql: could not connect to server: Bestand of map bestaat niet
>         Is the server running locally and accepting
>         connections on Unix domain socket "/var/run/postgresql/.s.PGSQL.5432"?

Bovenstaande staat ook in de bijlage.
Hierna heb ik geen verbinding meer kunnen maken, kan jij me helpen zonder dat ik opnieuw met mijn Raspbian Image moet beginnen.

dennissiemensma, 2018-09-14:
Ik denk dat je nog kunt proberen om het cluster of de datamap van postgresql helemaal weg te gooien. Ik zag namelijk ergens:
```
sudo pg_createcluster 9.6 main --start
Error: cluster configuration already exists
```
Wellicht postgresql stoppen/deinstalleren en alles verwijderen onder:
* `/etc/postgresql`
* `/var/lib/postgresql/`
Daarna nog een versie installatie proberen. Als je dan nog een fout krijgt dan kun je de logfile bekijken in `/var/log/postgresql/`.

Van zelfsprekend verwijder je hiermee ook alle data in de database, maar ik neem aan dat je nog ergens een backup hebt.

svanzon, 2018-10-22:
> Mocht ik nog iets voor je kunnen doen dan hoor ik het graag en heropen ik het issue weer voor je.

Sorry voor het uitblijven van een reactie. Het was niet helemaal meer gelukt, maar wel half geloof ik.

Maar was opnieuw begonnen met een image. Heb toen een hoop geleerd, en vooral geduld hebben 😊.

 

In de terminal lijkt het alsof je altijd door kan gaan, maar het blijft een computer die tijd nodig heeft.