dsmrreader/dsmr-reader #521: Uitlezen datalogger op andere PI

conradhagemans, 2018-09-14:
ik heb aan onze slimme-meter een RasPI hangen met een datalogger.
Via ser2net kan ik deze via een IP adres benaderen.

Kan ik dit ook gebruiken via deze dsmr-reader software?

dennissiemensma, 2018-09-14:
Is je vraag of DSMR-reader ondersteuning heeft of het uitlezen van een IP-datalogger? 

Momenteel wordt data-input via ofwel de API ofwel een seriële poort ondersteund. Die laatste kun je wel zelf instellen op een bepaald pad waar de datalogger draait, bijvoorbeeld:
* `/dev/ttyUSB0`
* `/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A5Z5VGV5-if00-port0` (als je poorten wisselen kun je eentje pakken uit `/dev/serial/by-id/`)

conradhagemans, 2018-09-14:
Beste Dennis, Dank voor het snelle antwoord.
Ik vond op #187 een vergelijkbare oplossing als waar ik naar zocht.
Ik lees nu op een IP poort van de RasPI die aan de slimme meter hangt het telegram elke 10 seconden uit en stuur ik via de API naar binnen. Ik moet wel cd CRC check uitzetten omdat deze een fout geeft. De telegrams worden verder wel netjes opgeslagen in de database.

conradhagemans, 2018-09-14:
Ik zie nu pas dat de gas standen niet meekomen. Dus ergens wordt het inlezen van het telegram afgebroken. Dat is vreemd. Ik gebruik ook Home Assistant om de meterstanden uit telezen en daarin wordt het telegram wel goed ge-parsed.

conradhagemans, 2018-09-15:
Het python3 script geeft deze output:

```
Listening at socket...
DATA LINE: b'/XMX5LGBBFFB231211304\r'
DATA LINE: b'\r'
DATA LINE: b''
Listening at socket...
DATA LINE: b'1-3:0.2.8(42)\r'
DATA LINE: b'0-0:1.0.0(180915015751S)\r'
DATA LINE: b''
Listening at socket...
DATA LINE: b'0-0:96.1.1(4530303035303031383434333232383134)\r'
DATA LINE: b'1-0:1.8.1(008070.889*kWh)\r'
DATA LINE: b'1-0:2.8.1(000616.439*kWh)\r'
DATA LINE: b'1-0:1.8.2(006654.461*kWh)\r'
DATA LINE: b'1-0:2.8.2(001937.006*kWh)\r'
DATA LINE: b''
Listening at socket...
DATA LINE: b'0-0:96.14.0(0001)\r'
DATA LINE: b'1-0:1.7.0(00.417*kW)\r'
DATA LINE: b'1-0:2.7.0(00.000*kW)\r'
DATA LINE: b''
Listening at socket...
DATA LINE: b'0-0:96.7.21(00007)\r'
DATA LINE: b'0-0:96.7.9(00005)\r'
DATA LINE: b'1-0:99.97.0(5)(0-0:96.7.19)'
Listening at socket...
DATA LINE: b'(170324110847W)(0036374999*s)'
Listening at socket...
DATA LINE: b'(160128105905W)(0024436867*s)'
Listening at socket...
DATA LINE: b'(151124214533W)(0000000330*s)'
Listening at socket...
DATA LINE: b'(151124213933W)(0000003569*s)'
Listening at socket...
DATA LINE: b'(150529101014S)(0000008072*s)\r'
DATA LINE: b'1-0:32.32.0(00002)\r'
DATA LINE: b'1-0:52.32.0(00001)\r'
DATA LINE: b'1-0:72.32.0(00001)\r'
DATA LINE: b'1-0:32.36.0(00000)\r'
DATA LINE: b'1-0:52.36.0(00000)\r'
DATA LINE: b'1-0:72.36.0(00000)\r'
DATA LINE: b''
Listening at socket...
DATA LINE: b'0-0:96.13.1()\r'
DATA LINE: b''
Listening at socket...
DATA LINE: b'0-0:96.13.0()\r'
DATA LINE: b'1-0:31.7.0(001*A)\r'
DATA LINE: b'1-0:51.7.0(001*A)\r'
DATA LINE: b'1-0:71.7.0(001*A)\r'
DATA LINE: b'1-0:21.7.0(00.219*kW)\r'
DATA LINE: b'1-0:41.7.0(00.139*kW)\r'
DATA LINE: b'1-0:61.7.0(00.058*kW)\r'
DATA LINE: b'1-0:22.7.0(00.000*kW)\r'
DATA LINE: b'1-0:42.7.0(00.000*kW)\r'
DATA LINE: b'1-0:62.7.0(00.000*kW)\r'
DATA LINE: b'0-1:24.1.0(003)\r'
DATA LINE: b''
Listening at socket...
DATA LINE: b'0-1:96.1.0(4730303032333430313931373434353135)\r'
DATA LINE: b''
Listening at socket...
DATA LINE: b'0-1:24.2.1(180915010000S)(09746.582*m3)\r'
DATA LINE: b''
Listening at socket...
DATA LINE: b'!3CA1\r'

/XMX5LGBBFFB231211304

1-3:0.2.8(42)
0-0:1.0.0(180915015751S)

0-0:96.1.1(4530303035303031383434333232383134)
1-0:1.8.1(008070.889*kWh)
1-0:2.8.1(000616.439*kWh)
1-0:1.8.2(006654.461*kWh)
1-0:2.8.2(001937.006*kWh)

0-0:96.14.0(0001)
1-0:1.7.0(00.417*kW)
1-0:2.7.0(00.000*kW)

0-0:96.7.21(00007)
0-0:96.7.9(00005)
1-0:99.97.0(5)(0-0:96.7.19)
(170324110847W)(0036374999*s)
(160128105905W)(0024436867*s)
(151124214533W)(0000000330*s)
(151124213933W)(0000003569*s)
(150529101014S)(0000008072*s)
1-0:32.32.0(00002)
1-0:52.32.0(00001)
1-0:72.32.0(00001)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00000)

0-0:96.13.1()

0-0:96.13.0()
1-0:31.7.0(001*A)
1-0:51.7.0(001*A)
1-0:71.7.0(001*A)
1-0:21.7.0(00.219*kW)
1-0:41.7.0(00.139*kW)
1-0:61.7.0(00.058*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
0-1:24.1.0(003)

0-1:96.1.0(4730303032333430313931373434353135)

0-1:24.2.1(180915010000S)(09746.582*m3)

!3CA1
```

En biedt deze aan de API aan.  Waarom wordt niet alle data ingelezen? En waarom gaat de checksum mis?

Voor de volledigheid hierbij het script:

`The telegram is sent by this Python-script to the API of dsmr-reader (https://github.com/dennissiemensma/dsmr-reader)
'''
from datetime import datetime
from time import sleep
import socket

import requests

HOST = '192.168.1.xxx'
PORT = XXXX

API_SERVERS = (
    ('http://192.168.1.x:86/api/v1/datalogger/dsmrreading', 'MY_API_KEY'),
)

def main():
    while True:
        telegram = read_telegram()
        print(datetime.now())
        print('Read telegram:')
        print(telegram)

        for current_server in API_SERVERS:
            api_url, api_key = current_server
            print(datetime.now())
            print('Sending telegram to:', api_url)
            send_telegram(telegram, api_url, api_key)

        sleep(10)

def read_telegram():
    telegram_start_seen = False
    telegram = ''

    s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
    s.connect((HOST, PORT))

    while True:
        print('Listening at socket...')
        socket_data = s.recv(1024)

        for data in socket_data.split(b"\n"):
            print("DATA LINE:", data)

            try:
                data = str(data, 'utf-8')
            except TypeError:
                pass

            if data.startswith('/'):
                telegram_start_seen = True

            if telegram_start_seen:
                telegram += data + "\n"

            if data.startswith('!') and telegram_start_seen:
                s.close()
                print ( telegram )
                return telegram

def send_telegram(telegram, api_url, api_key):
    response = requests.post(
        api_url,
        headers={'X-AUTHKEY': api_key},
        data={'telegram': telegram},
    )

    if response.status_code not in (200, 201):
        print('[!] API error: {}'.format(response.text))

if __name__ == '__main__':
    main()
`

dennissiemensma, 2018-09-15:
Wil je eens kijken in `logs/dsmrreader.log`? Staan daar regels zoals `Rejected telegram (base64 encoded): ...`?

Je mag hier wel een van die regels neerzetten, dan kan ik er naar kijken wat exact binnenkomt op de API, en mogelijk ook zien waarom die faalt.

conradhagemans, 2018-09-15:
Beste Dennis,
Super dat je zo snel reageert. Ik ben er vanavond opnieuw ingedoken. Ik kom er niet uit.
Allereerst het telegram dat ik naar de API stuur:

`/XMX5LGBBFFB231211304

1-3:0.2.8(42)
0-0:1.0.0(180916012802S)
0-0:96.1.1(4530303035303031383434333232383134)
1-0:1.8.1(008085.696*kWh)
1-0:2.8.1(000618.574*kWh)
1-0:1.8.2(006654.461*kWh)
1-0:2.8.2(001937.006*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.489*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00007)
0-0:96.7.9(00005)
1-0:99.97.0(5)(0-0:96.7.19)
1-0:32.32.0(00002)36374999*s)(160128105905W)(0024436867*s)(151124214533W)(0000000330*s)(151124213933W)(0000003569*s)(150529101014S)(0000008072*s)
1-0:52.32.0(00001)
1-0:72.32.0(00001)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00000)
0-0:96.13.1()
0-0:96.13.0()
1-0:31.7.0(001*A)
1-0:51.7.0(001*A)
1-0:71.7.0(001*A)
1-0:21.7.0(00.231*kW)
1-0:41.7.0(00.205*kW)
1-0:61.7.0(00.053*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303032333430313931373434353135)
0-1:24.2.1(180916010000S)(09748.562*m3)
!6C2E`

Volgens mij ziet dat er nu netjes uit. Ik maak dat met het volgende Python programma:

`'''
This script reads a telegram that is being transmitted by a ESP8266 (with ESPEasy firmware) connected to a P1-port of a smartmeter.
Source of the firmware is this page: http://www.esp8266thingies.nl/wp/software/

The telegram is sent by this Python-script to the API of dsmr-reader (https://github.com/dennissiemensma/dsmr-reader)
'''
from datetime import datetime
from time import sleep
import socket

import requests

HOST = '192.168.1.XXX'
PORT = 8899

API_SERVERS = (
    ('http://192.168.1.XXX:86/api/v1/datalogger/dsmrreading', 'MY_API_KEY'),
)

def main():
    while True:
        telegram = read_telegram()
        print(datetime.now())
        print('Read telegram:')
        print(telegram)

        for current_server in API_SERVERS:
            api_url, api_key = current_server
            print(datetime.now())
            print('Sending telegram to:', api_url)
            send_telegram(telegram, api_url, api_key)

        sleep(10)

def read_telegram():
    telegram_start_seen = False
    telegram = ''

    s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
    s.connect((HOST, PORT))

    while True:
        print('Listening at socket...')
        socket_data = s.recv(1024)

        for data in socket_data.split(b"\n"):
            print("DATA LINE:", data)

            try:
                data = str(data, 'utf-8')
            except TypeError:
                pass

            if data.startswith('/'):
                telegram_start_seen = True

            if telegram_start_seen and data !='':
                telegram += data

            if telegram_start_seen and data !='' and data[0:1] != '(':
                telegram += "\n"

            if data.startswith('!') and telegram_start_seen:
                s.close()
                print ( telegram )
                return telegram

def send_telegram(telegram, api_url, api_key):
    response = requests.post(
        api_url,
        headers={'X-AUTHKEY': api_key},
        data={'telegram': telegram},
    )

    if response.status_code not in (200, 201):
        print('[!] API error: {}'.format(response.text))

if __name__ == '__main__':
    main()
`
Als CRC check aan staat krijg ik in de logfile de volgende melding:

`[2018-09-16 01:30:36,362] ERROR @ services | CRC mismatch: 65026 / FE02 (telegram) != 19498 / 4C2A (calculated)
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services.py", line 188, in telegram_to_reading
    verify_telegram_checksum(data=data)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services.py", line 139, in verify_telegram_checksum
    telegram_checksum, hex_telegram_checksum, calculated_checksum, hex_calculated_checksum
dsmr_datalogger.exceptions.InvalidTelegramError: CRC mismatch: 65026 / FE02 (telegram) != 19498 / 4C2A (calculated)
[2018-09-16 01:30:56,389] WARNING @ services | Rejected telegram (base64 encoded): b'L1hNWDVMR0JCRkZCMjMxMjExMzA0DQoNCgoxLTM6MC4yLjgoNDIpDQowLTA6MS4wLjAoMTgwOTE2MDEzMDUyUykNCgowLTA6OTYuMS4xKDQ1MzAzMDMwMzUzMDMwMzEzODM0M
zQzMzMyMzIzODMxMzQpDQoxLTA6MS44LjEoMDA4MDg1LjcyMCprV2gpDQoxLTA6Mi44LjEoMDAwNjE4LjU3NCprV2gpDQoxLTA6MS44LjIoMDA2NjU0LjQ2MSprV2gpDQoxLTA6Mi44LjIoMDAxOTM3LjAwNiprV2gpDQoKMC0wOjk2LjE0LjAoMDAwMSkNCjEtMDoxLjcuMCgwMC40ODcqa1c
pDQoxLTA6Mi43LjAoMDAuMDAwKmtXKQ0KCjAtMDo5Ni43LjIxKDAwMDA3KQ0KMC0wOjk2LjcuOSgwMDAwNSkNCjEtMDo5OS45Ny4wKDUpKDAtMDo5Ni43LjE5KQooMTcwMzI0MTEwODQ3VykoMDAzNjM3NDk5OSpzKQooMTYwMTI4MTA1OTA1VykoMDAyNDQzNjg2NypzKQooMTUxMTI0MjE0N
TMzVykoMDAwMDAwMDMzMCpzKQooMTUxMTI0MjEzOTMzVykoMDAwMDAwMzU2OSpzKQooMTUwNTI5MTAxMDE0UykoMDAwMDAwODA3MipzKQ0KMS0wOjMyLjMyLjAoMDAwMDIpDQoxLTA6NTIuMzIuMCgwMDAwMSkNCjEtMDo3Mi4zMi4wKDAwMDAxKQ0KMS0wOjMyLjM2LjAoMDAwMDApDQoxLTA
6NTIuMzYuMCgwMDAwMCkNCjEtMDo3Mi4zNi4wKDAwMDAwKQ0KCjAtMDo5Ni4xMy4xKCkNCgowLTA6OTYuMTMuMCgpDQoxLTA6MzEuNy4wKDAwMSpBKQ0KMS0wOjUxLjcuMCgwMDEqQSkNCjEtMDo3MS43LjAoMDAxKkEpDQoxLTA6MjEuNy4wKDAwLjIzNCprVykNCjEtMDo0MS43LjAoMDAuM
Tk4KmtXKQ0KMS0wOjYxLjcuMCgwMC4wNTMqa1cpDQoxLTA6MjIuNy4wKDAwLjAwMCprVykNCjEtMDo0Mi43LjAoMDAuMDAwKmtXKQ0KMS0wOjYyLjcuMCgwMC4wMDAqa1cpDQowLTE6MjQuMS4wKDAwMykNCgowLTE6OTYuMS4wKDQ3MzAzMDMwMzIzMzM0MzAzMTM5MzEzNzM0MzQzNTMxMzU
pDQoKMC0xOjI0LjIuMSgxODA5MTYwMTAwMDBTKSgwOTc0OC41NjIqbTMpDQoKITM2QkI='
`

dennissiemensma, 2018-09-16:
Ik denk dat het komt doordat er extra linefeeds (`\n`) door het script in het telegram worden gezet, die niet volgens specificatie zijn. De API verwacht een 1-op-1 kopie van het telegram, inclusief alle tekens, exclusief eventuele toevoegingen.
Als neveneffect krijgt je daardoor ook altijd een falende CRC-check, omdat de CRC gemaakt is zonder die extra linefeeds en DSMR-reader neemt alle data (dus ook de extra linefeeds) mee in de CRC-vergelijking.

Volgens de bovenstaande base64 string krijgt de applicatie de data zo (visueel) binnen. Je meterserienummer heb ik even weggehaald:
```
/XMX5LGBBFFB231211304\r\n
\r\n
\n
1-3:0.2.8(42)\r\n
0-0:1.0.0(180916013052S)\r\n
\n
0-0:96.1.1(xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx)\r\n
1-0:1.8.1(008085.720*kWh)\r\n
1-0:2.8.1(000618.574*kWh)\r\n
1-0:1.8.2(006654.461*kWh)\r\n
1-0:2.8.2(001937.006*kWh)\r\n
\n
0-0:96.14.0(0001)\r\n
1-0:1.7.0(00.487*kW)\r\n
1-0:2.7.0(00.000*kW)\r\n
\n
0-0:96.7.21(00007)\r\n
0-0:96.7.9(00005)\r\n
1-0:99.97.0(5)(0-0:96.7.19)\n
(170324110847W)(0036374999*s)\n
(160128105905W)(0024436867*s)\n
(151124214533W)(0000000330*s)\n
(151124213933W)(0000003569*s)\n
(150529101014S)(0000008072*s)\r\n
1-0:32.32.0(00002)\r\n
1-0:52.32.0(00001)\r\n
1-0:72.32.0(00001)\r\n
1-0:32.36.0(00000)\r\n
1-0:52.36.0(00000)\r\n
1-0:72.36.0(00000)\r\n
\n
0-0:96.13.1()\r\n
\n
0-0:96.13.0()\r\n
1-0:31.7.0(001*A)\r\n
1-0:51.7.0(001*A)\r\n
1-0:71.7.0(001*A)\r\n
1-0:21.7.0(00.234*kW)\r\n
1-0:41.7.0(00.198*kW)\r\n
1-0:61.7.0(00.053*kW)\r\n
1-0:22.7.0(00.000*kW)\r\n
1-0:42.7.0(00.000*kW)\r\n
1-0:62.7.0(00.000*kW)\r\n
0-1:24.1.0(003)\r\n
\n
0-1:96.1.0(xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx)\r\n
\n
0-1:24.2.1(180916010000S)(09748.562*m3)\r\n
\n
!36BB
```

Overal waar alleen een `\n` staat en **geen** `\r\n` hoort geen `\n` te staan. Als ik bovenstaand telegram handmatig verifieer in de applicatie, faalt de CRC-check. 

Echter, wanneer ik het formaat aanpas naar hoe het hoort, dan faalt de CRC-check niet meer (wederom serienummers even weggehaald):

```
/XMX5LGBBFFB231211304\r\n
\r\n
1-3:0.2.8(42)\r\n
0-0:1.0.0(180916013052S)\r\n
0-0:96.1.1(xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx)\r\n
1-0:1.8.1(008085.720*kWh)\r\n
1-0:2.8.1(000618.574*kWh)\r\n
1-0:1.8.2(006654.461*kWh)\r\n
1-0:2.8.2(001937.006*kWh)\r\n
0-0:96.14.0(0001)\r\n
1-0:1.7.0(00.487*kW)\r\n
1-0:2.7.0(00.000*kW)\r\n
0-0:96.7.21(00007)\r\n
0-0:96.7.9(00005)\r\n
1-0:99.97.0(5)(0-0:96.7.19)(170324110847W)(0036374999*s)(160128105905W)(0024436867*s)(151124214533W)(0000000330*s)(151124213933W)(0000003569*s)(150529101014S)(0000008072*s)\r\n
1-0:32.32.0(00002)\r\n
1-0:52.32.0(00001)\r\n
1-0:72.32.0(00001)\r\n
1-0:32.36.0(00000)\r\n
1-0:52.36.0(00000)\r\n
1-0:72.36.0(00000)\r\n
0-0:96.13.1()\r\n
0-0:96.13.0()\r\n
1-0:31.7.0(001*A)\r\n
1-0:51.7.0(001*A)\r\n
1-0:71.7.0(001*A)\r\n
1-0:21.7.0(00.234*kW)\r\n
1-0:41.7.0(00.198*kW)\r\n
1-0:61.7.0(00.053*kW)\r\n
1-0:22.7.0(00.000*kW)\r\n
1-0:42.7.0(00.000*kW)\r\n
1-0:62.7.0(00.000*kW)\r\n
0-1:24.1.0(003)\r\n
0-1:96.1.0(xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx)\r\n
0-1:24.2.1(180916010000S)(09748.562*m3)\r\n
!36BB\n
```

Dus de data in het telegram lijkt juist, alleen het formaat niet. Je zult het ergens in die hoek moeten zoeken. In principe zou [dit script (link)](https://github.com/dennissiemensma/dsmr-reader/blob/v1.23.1/dsmr_datalogger/scripts/dsmr_datalogger_api_client.py#L60) afdoende moeten zijn als basis.

In je script staat iets met:
```
if telegram_start_seen and data !='' and data[0:1] != '(':
	telegram += "\n"
```
Waardoor die willekeurig linefeeds lijkt toe te voegen. Als je ervoor zorgt dat het script dezelfde API input geeft als het laatste telegram hierboven, verwacht ik dat de CRC-check niet meer faalt.

conradhagemans, 2018-09-18:
Dank voor alle hulp. Ik heb een script draaien waaruit nu geldige telegrammen komen. Ik weet niet waarom de telegrammen die uit de RasPI komen zoveel lege regels bevat enz. Ik gebruik een oude P1-ETH die ik een paar jaar gelden heb gekocht bij: [http://www.smartmeterdashboard.nl/](http://www.smartmeterdashboard.nl/)

```
`def read_telegram():
    telegram_start_seen = False
    telegram = ''

    s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
    s.connect((HOST, PORT))

    while True:
        print('Listening at socket...')
        socket_data = s.recv(1024)

        for data in socket_data.split(b'\n'):
            print("DATA LINE:", data)

            try:
                data = str(data, 'utf-8')
            except TypeError:
                pass

            if data.startswith('/'):
                telegram_start_seen = True
            if data[0:1] =='!':
               data = data[0:5]
            if telegram_start_seen and data !='':
                telegram += data
            # Zorg dat de 1-0:99.97.0 regel er ook goed inkomt
            if telegram_start_seen and data !='' and data[0:1] != '(' and data[-5:] !='7.19)':
                telegram += "\n"
            if telegram_start_seen and data[-8:] =="8072*s)\r":
                telegram += "\n"

            if data.startswith('!') and telegram_start_seen:
                s.close()
                return telegram
`
```

Dit script gaat de fout in zodra op deze regel er een code bijkomt:
`
1-0:99.97.0(5)(0-0:96.7.19)(170324110847W)(0036374999*s)(160128105905W)(0024436867*s)(151124214533W)(0000000330*s)(151124213933W)(0000003569*s)(150529101014S)(0000008072*s)\r\n` 

Gelukkig komt dat niet vaak voor.
Nu ik alles heb draaien ga ik proberen 3 jaar historie in de database krijgen. Daarover vragen in een ander topic.

dennissiemensma, 2018-09-18:
Fijn om te horen dat het gelukt is.

Ik weet niet of alle uitzonderingen qua if-checks nodig zijn, maar als het script nu werkt, vooral niet meer aanzitten zou ik zeggen. :]

closed 2018-09-18 by dennissiemensma