dsmrreader/dsmr-reader #522: Raspberry Pi 3 B+ supported

hjkoster400d, 2018-09-17:
Gezien de raspberry Pi 3B+ een andere chipset heeft ben ik benieuwd of DSMR reader hier ook op werkt. Heeft iemand ervaring met DSMR reader op de nieuwe raspberry?
Alvast bedankt

dennissiemensma, 2018-09-17:
Bedankt voor je vraag. Ik heb het zelf nog niet kunnen proberen, maar als er gewoon Raspbian OS, of een afgeleide daarvan op draait, ga ik er vanuit dat het geen probleem is.

hjkoster400d, 2018-09-17:
Bedankt voor je antwoord, morgen komt de nieuwe raspberry binnen. Zodra ik tijd heb zal ik de DSMR reader erop zetten en laat ik hier nog wel even weten of het werkt. In ieder geval erg bedankt voor je software! Gr Rik

hjkoster400d, 2018-09-18:
Vanavond de nieuwe raspberry geïnstalleerd en hij draait als een zonnetje. Mocht er toch nog iets voorvallen dan laat ik het hier natuurlijk wel even weten. Bedankt

dennissiemensma, 2018-09-18:
Goed om te horen en jij bedankt voor het proberen. Veel plezier ermee

closed 2018-09-18 by dennissiemensma