dsmrreader/dsmr-reader #523: Automatische gas consumption dashboard

cranberryNL, 2018-09-25:
Ik mis in het dashboard de automatische update van de grafische weergave van het gasverbruik. Voor elektriciteit werkt het prima, maar voor gas dus niet.
'k Heb geen optie terug kunnen vinden die dit zou regelen, dus ik vermoed òf een bug òf een ontbrekende feature òf een probleem met mijn config (v1.23 op een RPi3).

dennissiemensma, 2018-09-25:
Bedankt voor je melding. Dit is in zekere zin bewust, want ik had niet verwacht dat iemand het dashboard langer dan een paar minuten open hield, laat staan voor een kwartier of een uur voor de volgende gas-update.

cranberryNL, 2018-09-25:
Haha, dan ben ik zo'n gek die het dashboard continu open heb staan in m'n browser en regelmatig ff een blik werp op de verbruikscijfers ;)

Mocht je het zien zitten om het automatisch bijwerken toe te voegen, dan zou dat heel mooi zijn!

dennissiemensma, 2018-10-06:
Ik heb iets simpels gemaakt, wat de gehele pagina na 15 minuten refreshed. Soort compremis.

closed 2018-10-06 by dennissiemensma

cranberryNL, 2018-10-10:
Super, dankje Dennis! Ik zie het wel verschijnen in (één van) de volgende release(s).