dsmrreader/dsmr-reader #525: Stap 7 error

Btje1977, 2018-09-27:
*Let op: Het DSMR-reader project is gefocused op het uitlezen van de slimme meter en om die data met jou op zoveel mogelijk manieren te delen. Het inlezen van andere (externe) data in DSMR-reader valt buiten de scope van dit project.*

### Jouw omgeving
* Hardware Orangepizero

### Bug, wens of iets anders
installatie fout bij stap 7

  ----------------------------------------
  Failed building wheel for psycopg2
  Running setup.py clean for psycopg2
  Running setup.py bdist_wheel for psycopg2-binary ... error
  Complete output from command /home/dsmr/.virtualenvs/dsmrreader/bin/python3 -u                                   -c "import setuptools, tokenize;__file__='/tmp/pip-install-_un10_z5/psycopg2-bi                                  nary/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace                                  ('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /                                  tmp/pip-wheel-85yf3pas --python-tag cp35:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-3.5
  creating build/lib.linux-armv7l-3.5/psycopg2
  copying lib/_json.py -> build/lib.linux-armv7l-3.5/psycopg2
  copying lib/tz.py -> build/lib.linux-armv7l-3.5/psycopg2
  copying lib/psycopg1.py -> build/lib.linux-armv7l-3.5/psycopg2
  copying lib/extensions.py -> build/lib.linux-armv7l-3.5/psycopg2
  copying lib/__init__.py -> build/lib.linux-armv7l-3.5/psycopg2
  copying lib/_range.py -> build/lib.linux-armv7l-3.5/psycopg2
  copying lib/_ipaddress.py -> build/lib.linux-armv7l-3.5/psycopg2
  copying lib/extras.py -> build/lib.linux-armv7l-3.5/psycopg2
  copying lib/sql.py -> build/lib.linux-armv7l-3.5/psycopg2
  copying lib/errorcodes.py -> build/lib.linux-armv7l-3.5/psycopg2
  copying lib/pool.py -> build/lib.linux-armv7l-3.5/psycopg2
  creating build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/dbapi20.py -> build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/test_with.py -> build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/testutils.py -> build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/test_cursor.py -> build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/test_module.py -> build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/test_notify.py -> build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/test_replication.py -> build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/test_types_extras.py -> build/lib.linux-armv7l-3.5/psycopg2/test                                  s
  copying tests/test_transaction.py -> build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/test_types_basic.py -> build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/test_psycopg2_dbapi20.py -> build/lib.linux-armv7l-3.5/psycopg2/                                  tests
  copying tests/test_extras_dictcursor.py -> build/lib.linux-armv7l-3.5/psycopg2                                  /tests
  copying tests/test_ipaddress.py -> build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/test_copy.py -> build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/test_async.py -> build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/testconfig.py -> build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/test_bugX000.py -> build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/test_dates.py -> build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/__init__.py -> build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/test_fast_executemany.py -> build/lib.linux-armv7l-3.5/psycopg2/                                  tests
  copying tests/dbapi20_tpc.py -> build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/test_cancel.py -> build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/test_bug_gc.py -> build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/test_errcodes.py -> build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/test_sql.py -> build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/test_connection.py -> build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/test_green.py -> build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/test_lobject.py -> build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/test_quote.py -> build/lib.linux-armv7l-3.5/psycopg2/tests
  copying tests/test_async_keyword.py -> build/lib.linux-armv7l-3.5/psycopg2/tes                                  ts
  Skipping optional fixer: buffer
  Skipping optional fixer: idioms
  Skipping optional fixer: set_literal
  Skipping optional fixer: ws_comma
  running build_ext
  building 'psycopg2._psycopg' extension
  creating build/temp.linux-armv7l-3.5
  creating build/temp.linux-armv7l-3.5/psycopg
  arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-protot                                  ypes -g -fdebug-prefix-map=/build/python3.5-haFsHd/python3.5-3.5.3=. -fstack-pro                                  tector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -                                  fPIC -DPSYCOPG_DEFAULT_PYDATETIME=1 -DPSYCOPG_VERSION=2.7.5 (dt dec pq3 ext) -DP                                  G_VERSION_NUM=90610 -I/usr/include/python3.5m -I/home/dsmr/.virtualenvs/dsmrread                                  er/include/python3.5m -I. -I/usr/include/postgresql -I/usr/include/postgresql/9.                                  6/server -c psycopg/psycopgmodule.c -o build/temp.linux-armv7l-3.5/psycopg/psyco                                  pgmodule.o -Wdeclaration-after-statement
[... 75 lines cut ...]
    Skipping optional fixer: ws_comma
    running build_ext
    building 'psycopg2._psycopg' extension
    creating build/temp.linux-armv7l-3.5
    creating build/temp.linux-armv7l-3.5/psycopg
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prot                                  otypes -g -fdebug-prefix-map=/build/python3.5-haFsHd/python3.5-3.5.3=. -fstack-p                                  rotector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2                                   -fPIC -DPSYCOPG_DEFAULT_PYDATETIME=1 -DPSYCOPG_VERSION=2.7.5 (dt dec pq3 ext) -                                  DPG_VERSION_NUM=90610 -I/usr/include/python3.5m -I/home/dsmr/.virtualenvs/dsmrre                                  ader/include/python3.5m -I. -I/usr/include/postgresql -I/usr/include/postgresql/                                  9.6/server -c psycopg/psycopgmodule.c -o build/temp.linux-armv7l-3.5/psycopg/psy                                  copgmodule.o -Wdeclaration-after-statement
    In file included from psycopg/psycopgmodule.c:27:0:
    ./psycopg/psycopg.h:34:20: fatal error: Python.h: No such file or directory
     #include <Python.h>
                        ^
    compilation terminated.

    It appears you are missing some prerequisite to build the package from sourc                                  e.

    You may install a binary package by installing 'psycopg2-binary' from PyPI.
    If you want to install psycopg2 from source, please install the packages
    required for the build and try again.

    For further information please check the 'doc/src/install.rst' file (also at
    <http://initd.org/psycopg/docs/install.html>).

    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

    ----------------------------------------
Command "/home/dsmr/.virtualenvs/dsmrreader/bin/python3 -u -c "import setuptools                                  , tokenize;__file__='/tmp/pip-install-_un10_z5/psycopg2/setup.py';f=getattr(toke                                  nize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec                                  (compile(code, __file__, 'exec'))" install --record /tmp/pip-record-s706awpa/ins                                  tall-record.txt --single-version-externally-managed --compile --install-headers                                   /home/dsmr/.virtualenvs/dsmrreader/include/site/python3.5/psycopg2" failed with                                   error code 1 in /tmp/pip-install-_un10_z5/psycopg2/

dennissiemensma, 2018-09-28:
Bedankt voor je melding. Het lijkt mis te gaan op het punt waarbij de externe requirements voor DSMR-reader niet goed geinstalleerd kunnen worden. Ik beheer daar zelf niet de code voor, maar ik help je graag op weg naar een oplossing.

* Welk OS draai je? Gewoon Raspbian?

* Ik ben een dergelijke melding ook wel eens op mn werk tegengekomen. Daarbij miste er vaak de Python Development packages. Zie ook bijvoorbeeld: https://stackoverflow.com/questions/21530577/fatal-error-python-h-no-such-file-or-directory. Wil je eens kijken of het installeren van `python3-dev` helpt? 

Dus: `sudo apt-get install python3-dev` en dan weer wisselen naar de `dsmr` user en opnieuw proberen de requirements te installeren.

Btje1977, 2018-09-29:
Dat werkte dankjewel.

Hij stop nu bij stap 10 sudo supervisorctl

Melding is : 
root@verbruiksmonitor:~# sudo cp /home/dsmr/dsmr-reader/dsmrreader/provisioning/supervisor/dsmr-reader.conf /etc/supervisor/conf.d/
root@verbruiksmonitor:~# sudo supervisorctl
unix:///var/run/supervisor.sock no such file

Btje1977, 2018-09-29:
De reread geeft vervolgens:

error: <class 'socket.error'>, [Errno 111] Connection refused: file: /usr/lib/python2.7/socket.py line: 228

dennissiemensma, 2018-09-29:
Ik denk dat Supervisor dan ofwel niet goed geinstalleerd is, ofwel niet draait. Je kunt even in de processlist kijken of die draait en anders de service proberen te installeren.

dennissiemensma, 2018-09-30:
Je zult dat als `root` of `pi` user moeten uitvoeren, omdat je anders onvoldoende rechten hebt.

Btje1977, 2018-09-30:
raar doe ik het nogmaals pakt hij de eerste regen wel daarna weer fout. Met sudo voert hij het toch als root uit?

(dsmrreader) root@verbruiksmonitor:~# sudo cp /home/dsmr/dsmr reader/dsmrreader/provisioning/supervisor/dsmr-reader.conf /etc/supervisor/conf.d/
(dsmrreader) root@verbruiksmonitor:~# sudo supervisorctl
unix:///var/run/supervisor.sock refused connection
supervisor>

Btje1977, 2018-09-30:
Gestert maar blijft mekkeren:

(dsmrreader) root@verbruiksmonitor:~# service supervisor restart
(dsmrreader) root@verbruiksmonitor:~# service supervisor start
(dsmrreader) root@verbruiksmonitor:~#
(dsmrreader) root@verbruiksmonitor:~#
(dsmrreader) root@verbruiksmonitor:~# supervisorctl reread
error: <class 'socket.error'>, [Errno 111] Connection refused: file: /usr/lib/python2.7/socket.py line: 228
(dsmrreader) root@verbruiksmonitor:~# sudo supervisorctl reread
error: <class 'socket.error'>, [Errno 111] Connection refused: file: /usr/lib/python2.7/socket.py line: 228
(dsmrreader) root@verbruiksmonitor:~#

dennissiemensma, 2018-09-30:
* Zie je Supervisor uberhaupt draaien in de processlist?
* Als je de configfile voor Supervisor bedoelt, die zou in `/etc/supervisor/` moeten staan.

Btje1977, 2018-10-01:
ik zal eens kijken met het volgende commando: ps -aux | less

Als ik hem start krijg ik geen foutmelding. vandaar dat ik dacht dat de supervisor wel draaide.

dennissiemensma, 2018-10-01:
En krijg je inmiddels wat te zien in de processlist qua supervisor?

dennissiemensma, 2018-10-17:
@Btje1977 ben je hier nog verder mee gekomen?

dennissiemensma, 2018-10-20:
Ik sluit deze voor nu. Ik hoor het graag als je alsnog hulp nodig hebt.

closed 2018-10-20 by dennissiemensma

Btje1977, 2019-02-10:
Hi,

ik ben weer opnieuw begonnen en krijg de volgende error:

 creating build/temp.linux-armv7l-3.5/psycopg
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall
-Wstrict-prototypes -g
-fdebug-prefix-map=/build/python3.5-QMv3zq/python3.5-3.5.3=.
-fstack-protector-strong -Wformat -Werror=format-security -Wdate-time
-D_FORTIFY_SOURCE=2 -fPIC -DPSYCOPG_DEFAULT_PYDATETIME=1
-DPSYCOPG_VERSION=2.7.6.1 (dt dec pq3 ext) -DPG_VERSION_NUM=90610
-I/usr/include/python3.5m
-I/home/dsmr/.virtualenvs/dsmrreader/include/python3.5m -I.
-I/usr/include/postgresql -I/usr/include/postgresql/9.6/server -c
psycopg/psycopgmodule.c -o
build/temp.linux-armv7l-3.5/psycopg/psycopgmodule.o
-Wdeclaration-after-statement
    In file included from psycopg/psycopgmodule.c:27:0:
    ./psycopg/psycopg.h:34:20: fatal error: Python.h: No such file or
directory
     #include <Python.h>
                        ^
    compilation terminated.

    It appears you are missing some prerequisite to build the package from
source.

    You may install a binary package by installing 'psycopg2-binary' from
PyPI.
    If you want to install psycopg2 from source, please install the packages
    required for the build and try again.

    For further information please check the 'doc/src/install.rst' file
(also at
    <http://initd.org/psycopg/docs/install.html>).

    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

    ----------------------------------------
Command "/home/dsmr/.virtualenvs/dsmrreader/bin/python3 -u -c "import
setuptools,
tokenize;__file__='/tmp/pip-install-qqv_6ud_/psycopg2/setup.py';f=getattr(tokenize,
'open', open)(__file__);code=f.read().replace('\r\n',
'\n');f.close();exec(compile(code, __file__, 'exec'))" install --record
/tmp/pip-record-q90swfrn/install-record.txt
--single-version-externally-managed --compile --install-headers
/home/dsmr/.virtualenvs/dsmrreader/include/site/python3.5/psycopg2" failed
with error code 1 in /tmp/pip-install-qqv_6ud_/psycopg2/

dit na:

pip3 install -r dsmrreader/provisioning/requirements/base.txt -r
dsmrreader/provisioning/requirements/postgresql.txt

geinstalleerd op armbian.

Mvg Bram

jeroenpeters1986, 2019-02-10:
Hoi @Btje1977, het lijkt erop dat je een pakket mist met de python devlopment files.
Ik zie dat je python3 gebruikt, wat je kunt doen is het volgende command uitvoeren:
`sudo apt-get install python3-dev` 
Daarna kun je hopelijk weer verder met de stappen.

Laat maar even weten of het lukt. Succes!

Btje1977, 2019-02-12:
Hoi,

dat stuk ging goed thanx.

Nu zit ik vast bij de supervisor:

root@orangepizero:~# sudo supervisorctl
unix:///var/run/supervisor.sock no such file
supervisor> reread
error: <class 'socket.error'>, [Errno 2] No such file or directory: file:
/usr/lib/python2.7/socket.py line: 228
supervisor>

Dat kreeg ik de vorige keer ook niet lekker draaien.

Hoe check ik of supervisor draait?

Btje1977, 2019-02-12:
Na uitvoeren van het volgende:

```
sudo touch /var/run/supervisor.sock
sudo chmod 777 /var/run/supervisor.sock
sudo service supervisor restart
```

Ben ik verder maar geeft hij supervisor.sock refused connection bij
uitvoeren  sudo supervisorctl

bramvandekerkhof@gmail.com>:

Btje1977, 2019-02-12:
Het bleek dat er geen logfile was aangemaakt.

Hij draait nu, cool.

Nu MQTT aan de praat krijgen.

Groeten Bram

jeroenpeters1986, 2019-02-12:
Klinkt goed Bram, succes!

Btje1977, 2019-02-12:
mmm

als ik de config wil aanpassen moet ik een ww invullen. welk?

   -

   Open the site’s vhost in /etc/nginx/sites-enabled/dsmr-webinterface and
   *uncomment* the following lines (remove the ##):

   ##    satisfy    any;##    allow      192.168.1.0/24;##    deny
  all;##    auth_basic "Restricted application";##
auth_basic_user_file /etc/nginx/htpasswd;

   -

   Now make sure you didn’t insert any typo’s by running:

   sudo service nginx configtest

   -

   And reload with:

   sudo service nginx reload

You should be prompted for login credentials the next time your browser
accesses the application. For more information regarding this topic, see
the Nginx docs
<https://www.nginx.com/resources/admin-guide/restricting-access/>.

Heb ik daarna gedaan maar geen prompt.

Btje1977, 2019-02-12:
Ik heb dus geen idee wat ik bij de django login moet invullen.

Mvg Bram

Btje1977, 2019-02-12:
Dag Dennis,

alles nog eens nagelezen en gevonden wat ik moet invullen.

Mvg Bram