dsmrreader/dsmr-reader #526: Preserve write cycles on sdcard

hjkoster400d, 2018-09-28:
Ik gebruik sinds een paar weken die DSMR reader op een raspberry pi 3 B+ en het bevalt uitstekend. Wat een mooi kwalitatief stukje software is dit.

Een groot nadeel van de raspberry is echter de schrijf-cycles op de sdcard zijn beperkt. Zijn er voorbeelden van oplossingen waarbij de sdcard zoveel mogelijk wordt ontzien? bijvoorbeeld door de database op een usb stick te zetten? Of zijn er andere componenten die ook veel disk-io hebben?

Wat ook handig zou zijn is een UI element waarmee de backups eenvoudig weer kunnen worden teruggezet (i.p.v via de cli). Is dat iets wat mogelijk is?

Gr Rik

dennissiemensma, 2018-09-29:
De nieuwe release heeft standaard minder logging aanstaan.

dennissiemensma, 2018-09-30:
Ik zal hetzelfde ook nog doorvoeren voor de datalogger, gezien die ook een hoop output geeft.

hjkoster400d, 2018-10-02:
Heb de update uitgerold. Thx voor je moeite!!

dennissiemensma, 2018-10-02:
In de volgende release komt nog een aanvulling hierop. Qua postgresql-database kon ik niet zo 1-2-3 iets vinden om de I/O te verminderen.

Je zou nog kunnen spelen met het verhogen van `DSMRREADER_BACKEND_SLEEP` en `DSMRREADER_DATALOGGER_SLEEP`: https://dsmr-reader.readthedocs.io/nl/latest/settings.html#dsmrreader-backend-sleep
Hierdoor draaien de backend en de datalogger minder vaak (ze wachten dan gewoon X seconden), wat uiteindelijk weer resulteert in minder metingen en I/O. Ik weet echter niet of dat significant iets uit maakt.

closed 2018-10-02 by dennissiemensma