dsmrreader/dsmr-reader #528: Wishlist item: the choice to use OneDrive instead of Dropbox for making automated backups

PeterEIER, 2018-10-03:
Automatic storage of backups on OneDrive would be great... I get 1TB cloud storage for free with my Office365 subscribtion, my free Dropbox account is filled up with DSMR backups but lots and lots of space left on my OneDrive.

So if possible think about supporting OneDrive next to Dropbox for automatic DMSR backups.

Thanks in advance for considering.

dennissiemensma, 2018-10-03:
Bedankt voor je verzoek. Ik heb ooit andere cloudproviders bekeken via #57 en #58. Ik kan daar wel een keer opnieuw naar kijken.

voipmeister, 2018-12-21:
I'd like support for OneDrive as well, for the same reasons as @PeterEIER. If you need any help testing or need a OneDrive account for development, I can provide you with one 😃

dennissiemensma, 2018-12-21:
Bedankt voor het aangeven van je voorkeur. Ik heb er destijds wel even naar gekeken, maar ik kreeg het niet werkend in OneDrive zelf (als ik het me goed herinner). Wellicht dat ik de komende weken nog even naar kan kijken en anders wordt het niet eerder dan het voorjaar.

voipmeister, 2018-12-21:
Ik heb mijn dropbox al opgeschoond, de reden dat ik het aanhoud op dit moment is het gebruik voor DSMR back-ups. Mijn aanbod blijft staan hoor, als je een OneDrive account nodig hebt kan ik je deze bieden. Ook over 3 maanden of later.

dennissiemensma, 2019-04-28:
In de volgende release komt overigens ook ondersteuning voor het versturen van een back-up per e-mail. Via je eigen e-mailserver dan (zelf instellingen invullen dus). De back-up is overigens wel beperkt tot de dag-statistieken die in het Archief zichtbaar zijn. Meer voor nood.

dennissiemensma, 2019-09-10:
Ik merk dat ik veel tijd kwijt ben aan implementaties van backups en syncen en alles (zie #690). Daarom ga ik deze functionaltieit niet code-technisch binnen dit project maken.

Als alternatief wil ik later kijken voor simpele tools die dit gewoon op de commandline kunnen syncen. Ook voor Dropbox en GDrive.

Voor OneDrive zou dit bijvoorbeeld een optie zijn: https://www.maketecheasier.com/sync-onedrive-linux/

dennissiemensma, 2020-02-27:
Ik heb hier opnieuw naar gekeken, maar ik kom wederom tot de conclusie dat er geen standaardmanier is om dit te ondersteunen. 
Er zijn wel diverse clients, tools en docker images beschikbaar voor Linux, maar die moeten gescreend en getest worden en ik zie dat persoonlijk niet als mijn taak. En ook niet als onderdeel van dit project.

Degenen die dit willen kunnen dit het beste zelf onderzoeken. In principe is elke tool afdoende die als daemon, container of cronjob in de achtergrond draait om data te syncen. **Neem daarbij wel in acht dat er ook onveilige software kan rondgaan, dus installeer niet zomaar iets.**

_Ik laat overigens de Dropbox-integratie er inzitten, omdat ik bestaande gebruikers niet iets wil afnemen wat doorgaans afdoende werkt._

closed 2020-02-27 by dennissiemensma