dsmrreader/dsmr-reader #530: API error

GreyEarl, 2018-10-06:
I have added a script to remotely extract the telegrams. This works, however I have an issue submitting to the API. The API is enabled and I'm using the API key in every submission. Maybe I missed some here, but this is what I get when submitting to the API:

```
Sending telegram to: http://10.0.0.21/api/v1/datalogger/dsmrreading
[!] API error: <html>
<head><title>404 Not Found</title></head>
<body bgcolor="white">
<center><h1>404 Not Found</h1></center>
<hr><center>nginx/1.10.3 (Ubuntu)</center>
</body>
</html>
```

GreyEarl, 2018-10-06:
Hi Dennis, thanks for your quick response! The URL shows a 404 in the browser as well, but the DSMR is working and reachable on 10.0.0.21 (port 8080). Could it be something with Nginx?

GreyEarl, 2018-10-06:
Oh wait ...

dennissiemensma, 2018-10-06:
Well then I think you should change the API URL In your script from:
```
http://10.0.0.21/api/v1/datalogger/dsmrreading
```

To:
```
http://10.0.0.21:8080/api/v1/datalogger/dsmrreading
```

The missing port might cause the issue.

GreyEarl, 2018-10-06:
Exactly, after I wrote my reply it struck me I'm not using the default port 80 for my instance, doh :) Thanks!
I get now a 'Failed to parse telegram', but I will check that first if I can fix it.

dennissiemensma, 2018-10-06:
Great, let me know if you need any other help.

dennissiemensma, 2018-10-06:
Help in debugging yourself can be found here: https://dsmr-reader.readthedocs.io/en/latest/troubleshooting.html

closed 2018-10-06 by dennissiemensma