dsmrreader/dsmr-reader #531: From valid telegrams to invalid in seconds.

GreyEarl, 2018-10-06:
I got a lot closer setting up the DSMR instance by seeing actual data in the dashboard. However, after a couple of valid telegram the application marks them as invalid:

```
Read telegram:
/XMX5LGBBLxxxx001xx57^M
^M
1-3:0.2.8(50)^M
0-0:1.0.0(181006115207S)^M
0-0:96.1.1(45303033363xxxxx373438323335313136)^M
1-0:1.8.1(001807.699*kWh)^M
1-0:1.8.2(002193.515*kWh)^M
1-0:2.8.1(000000.000*kWh)^M
1-0:2.8.2(000000.000*kWh)^M
0-0:96.14.0(0001)^M
1-0:1.7.0(00.231*kW)^M
1-0:2.7.0(00.000*kW)^M
0-0:96.7.21(00022)^M
0-0:96.7.9(00008)^M
1-0:99.97.0(8)(0-0:96.7.19)(170330181426S)(0000089786*s)(170429140233S)(0000000490*s)(170429140238S)(0000000490*s)(170429141523S)(0000000740*s)(170429141528S)(0000000740*s)(170604201126S)(0000001209*s)(170605203424S)(0000001485*s)(170609150855S)(0000000184*s)^M
1-0:32.32.0(00004)^M
1-0:52.32.0(00003)^M
1-0:72.32.0(00004)^M
1-0:32.36.0(00000)^M
1-0:52.36.0(00000)^M
1-0:72.36.0(00000)^M
0-0:96.13.0()^M
1-0:32.7.0(237.0*V)^M
1-0:52.7.0(238.0*V)^M
1-0:72.7.0(238.0*V)^M
1-0:31.7.0(000*A)^M
1-0:51.7.0(001*A)^M
1-0:71.7.0(000*A)^M
1-0:21.7.0(00.097*kW)^M
1-0:41.7.0(00.120*kW)^M
1-0:61.7.0(00.013*kW)^M
1-0:22.7.0(00.000*kW)^M
1-0:42.7.0(00.000*kW)^M
1-0:62.7.0(00.000*kW)^M
!AE9F^M

Sending telegram to: http://10.0.0.21:8080/api/v1/datalogger/dsmrreading
Read telegram:

Sending telegram to: http://10.0.0.21:8080/api/v1/datalogger/dsmrreading
[!] API error: Invalid data
Read telegram:

Sending telegram to: http://10.0.0.21:8080/api/v1/datalogger/dsmrreading
[!] API error: Invalid data
Read telegram:

Sending telegram to: http://10.0.0.21:8080/api/v1/datalogger/dsmrreading
[!] API error: Invalid data
```
As if there is no data at all? I checked my sending device, but it just spit correct P1 data every second. When I restart the relevant services in supervisor it works for a few seconds, showing correct data, but after that, nothing.

dennissiemensma, 2018-10-06:
Are the `Read telegram:` lines empty or did you remove the data yourself?

The `Invalid data` error occurs when no `telegram` field was found, in the POST data submitted to the API: https://github.com/dennissiemensma/dsmr-reader/blob/v1.24.0/dsmr_api/forms.py#L15

GreyEarl, 2018-10-06:
The lines are empty as is unfortunately.  Strange thing is when I do a simple wget to the P1 device on the same system as DSMR reader, I get telegrams flowing. So the script should be the culprit?

GreyEarl, 2018-10-06:
In addition to my above edit. It seems before getting invalid data, some telegrams looks like it ended abruptly. Here is the very last part of a new test:

```
Read telegram:
/XMX5LGBBLB2410010157

1-3:0.2.8(50)
0-0:1.0.0(181006224745S)
0-0:96.1.1(45XXXXXXXXXXXXXXXXXX6)
1-0:1.8.1(001811.253*kWh)
1-0:1.8.2(002193.515*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.247*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00022)
0-0:96.7.9(00008)
1-0:99.97.0(8)(0-0:96.7.19)(170330181426S)(0000089786*s)(170429140233S)(0000000490*s)(170429140238S)(0000000490*s)(170429141523S)(0000000740*s)(170429141528S)(0000000740*s)(170604201126S)(0000001209*s)(170605203424S)(0000001485*s)(170609150855S)(0000000184*s)
1-0:32.32.0(00004)
1-0:52.32.0(00003)
1-0:72.32.0(00004)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00000)
0-0:96.13.0()
1-0:32.7.0(236.0*V)
1-0:52.7.0(237.0*V)
1-0:72.7.0(236.0*V)
1-0:31.7.0(000*A)
1-0:51.7.0(001*A)
1-0:71.7.0(000*A)
1-0:21.7.0(00.100*kW)
1-0:41.7.0(00.138*kW)
1-0:61.7.0(00.008*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
!1AB7

Sending telegram to: http://10.0.0.21:8080/api/v1/datalogger/dsmrreading
Read telegram:
/XMX5LGBBLB2410010157

1-3:0.2.8(50)
0-0:1.0.0(181006224747S)
0-0:96.1.1(4XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX36)
1-0:1.8.1(001811.253*kWh)
1-0:1.8.2(002193.515*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.247*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00022)
0-0:96.7.9(00008)
1-0:99.97.0(8)(0-0:96.7.19)(170330181426S)(0000089786*s)(170429140233S)(0000000490*s)(170429140238S)(0000000490*s)(170429141523S)(0000000740*s)(170429141528S)(0000000740*s)(170604201126S)(0000001209*s)(1706052034
Sending telegram to: http://10.0.0.21:8080/api/v1/datalogger/dsmrreading
[!] API error: Failed to parse telegram
Read telegram:
24S)(0000001485*s)(170609150855S)(0000000184*s)
1-0:32.32.0(00004)
1-0:52.32.0(00003)
1-0:72.32.0(00004)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00000)
0-0:96.13.0()
1-0:32.7.0(236.0*V)
1-0:52.7.0(237.0*V)
1-0:72.7.0(236.0*V)
1-0:31.7.0(000*A)
1-0:51.7.0(001*A)
1-0:71.7.0(000*A)
1-0:21.7.0(00.100*kW)
1-0:41.7.0(00.138*kW)
1-0:61.7.0(00.008*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
!868A

Sending telegram to: http://10.0.0.21:8080/api/v1/datalogger/dsmrreading
[!] API error: Failed to parse telegram
Read telegram:

Sending telegram to: http://10.0.0.21:8080/api/v1/datalogger/dsmrreading
[!] API error: Invalid data
Read telegram:

Sending telegram to: http://10.0.0.21:8080/api/v1/datalogger/dsmrreading
[!] API error: Invalid data
Read telegram:

Sending telegram to: http://10.0.0.21:8080/api/v1/datalogger/dsmrreading
[!] API error: Invalid data
Read telegram:

Sending telegram to: http://10.0.0.21:8080/api/v1/datalogger/dsmrreading
[!] API error: Invalid data
Read telegram:

Sending telegram to: http://10.0.0.21:8080/api/v1/datalogger/dsmrreading
[!] API error: Invalid data
```

GreyEarl, 2018-10-07:
Hi Dennis,

I have been experimenting with other firmware for my ESP8266, which pushes the P1 readings on the network. For the last 15 minuten the telegram processing works stable within DSMR reader!

closed 2018-10-07 by GreyEarl

dennissiemensma, 2018-10-07:
Okay, great to hear!