dsmrreader/dsmr-reader #534: No new data shown, telegrams still received by Pi

raoulteeuwen, 2018-10-18:
### Jouw omgeving
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.2.5`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 3b...`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v4`

### Bug, wens of iets anders
Gisteravond leek er geen data meer bij te komen; in de grafieken was na een tijdstip niets nieuws te zien.

Ik heb op de command prompt gekeken of er wel telegrams binnen komen op de Pi, en dat doen ze.

Een collega hielp me en vond 

cat dsmrreader.log 
<oude foutmeldingen>
[2018-10-18 09:59:33,957] WARNING  @ services | Rejected telegram (base64 encoded): b'L1gxMDEwMS0xKDAwMDEoMSgNCikxMTEtMC0wLTEtMS0xMTAxMDAtIUMvWE1YNTEtMC4wMTAxNTEtKDEoDTEtMS0xMC0wOjk2LjEzLjEoKQ0KMC0wOjk2LjEzLjAoKQ0KMS0wOjMxLjcuMCgwMDIqQSkNCjEtMDo1MS43LjAoMDAxKkEpDQoxLTA6NzEuNy4wKDAwMSpBKQ0KMS0wLTE6MjQuMi4xKDE4MTAxODA5MDAwMFMpKDA0Mjk3LjI1NiptMykNCiEwRjk2DQo='
[2018-10-18 09:59:33,960] ERROR    @ services | Content or CRC data not found
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services.py", line 189, in telegram_to_reading
    verify_telegram_checksum(data=data)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services.py", line 123, in verify_telegram_checksum
    raise InvalidTelegramError('Content or CRC data not found')
dsmr_datalogger.exceptions.InvalidTelegramError: Content or CRC data not found
[2018-10-18 10:00:44,110] WARNING  @ services | Rejected telegram (base64 encoded): b'L1hNWDVMR0JCRkcxMDEyNDQ5Nzc1DQoNCjEtMzowLjIuOCg0MikNCjAtMDoxLjAuMCgxODEwMTgxMDAwMDVTKQ0KMC0wOjk2LjEuMSg0NTMwMzAzMzMwMzAzMDMzMzIzNzM3MzQzNDM0MzIzMTM2KQ0KMS0wOjEuOC4xKDAwMzEzNi42NzAqa1doKQ0KMS0wOjEuOC4yKDAwMjg5MC4yMjIqMS0wOjk5Ljk3LjAoMikoMC0wOjk2LjcuMTkpDQoxLTA6MzIuMzIuMCgwMDAwMCkNCjEtMDo1Mi4zNi4wKDAwMDAwKQ0KMTAtMDo5Ni4xMy4wKCkNCjEtMDo3MS43LjAoMDAxKkEpDQoxLTA6NDIuNy4wKDAwLjAzNyprVykNCjAtMTAtMToyNC4yLjEoMTgxMDE4MDkwMDAwUykoMDQyOTcuMjU2Km0zKQ0KIUNBMjcNCg=='
[2018-10-18 10:00:44,111] ERROR    @ services | CRC mismatch: 51751 / CA27 (telegram) != 63742 / F8FE (calculated)
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services.py", line 189, in telegram_to_reading
    verify_telegram_checksum(data=data)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services.py", line 140, in verify_telegram_checksum
    telegram_checksum, hex_telegram_checksum, calculated_checksum, hex_calculated_checksum
dsmr_datalogger.exceptions.InvalidTelegramError: CRC mismatch: 51751 / CA27 (telegram) != 63742 / F8FE (calculated)
[2018-10-18 10:05:34,674] WARNING  @ services | Rejected telegram (base64 encoded): b'L1hNWDVMR0JCRkcxMDEyNDQ5Nzc1DQoNCjEtMzowLjIuOCg0MikNCjAtMDoxLjAuMCgxODEwMTgxMDA0NDZTKQ0KMC0wOjk2LjEuMSg0NTMwMzAzMzMwMzAzMDMzMzIzNzM3MzQzNDM0MzIzMTM2KQ0KMS0wOjEuOC4xKDAwMzEtMDoyLjguMSgwMDA0NjMuNzY5KmtXaCkNCjEtMDoyLjguMigwMDEyMipzKQ0KMDAwKQ0KMS0wOjUyLjM2LjAoMDAwMDApDQowMS03LjEtMS0xLTEtMC0wLTAtIS9YTVg1TEdCQkZHMTAxMjQ0OTc3NQ0KDQoxLTM6MC4yLjgoNDIpDQowLTA6MS4wLjAoMTgxMDE4MTAwNDU2UykNCjAtMDo5Ni4xLjEoNDUzMDMwMzMzMDMwMzAzMzMyMzczNzM0MzQzNDMyMzEzNikNCjEtMDoxLjguMSgwMDMxMzYuNjcwKmtXaCkNCjEtMDoxLjguMigwMDI4OTAuMjI2KmtXaCkNCjEtMDoyLjguMSgwMDA0NjMuNzY5KmtXaCkNCjEtMDoyLjguMigwMDEyMDYuODQ0KmtXaCkNCjAtMDo5Ni4xNC4wKDAwMDIpDQoxLTA6MS43LjAoMDAuMDAwKmtXKQ0KMS0wOjIuNy4wKDAwLjMwNSprVykNCjAtMDo5Ni43LjIxKDAwMDA1KQ0KMC0wOjk2LjcuOSgwMDAwMikNCjEtMDo5OS45Ny4wKDIpKDAtMDo5Ni43LjE5KSgxODEwMTcwOTUyNDVTKSgwMDAwMDA2OTMzKnMpKDE4MDgzMDEzNDc1MVMpKDAwMDAwMDI3MDIqcykNCjEtMDozMi4zMi4wKDAwMDAwKQ0KMS0wOjUyLjMyLjAoMDAwMDApDQoxLTA6NzIuMzIuMCgwMDAwMCkNCjEtMDozMi4zNi4wKDAwMDAwKQ0KMS0wOjUyLjM2LjAoMDAwMDApDQoxLTA6NzIuMzYuMCgwMDAwMCkNCjAtMDo5Ni4xMy4xKCkNCjAtMDo5Ni4xMy4wKCkNCjEtMDozMS43LjAoMDAyKkEpDQoxLTA6NTEuNy4wKDAwMSpBKQ0KMS0wOjcxLjcuMCgwMDEqQSkNCjEtMDoyMS43LjAoMDAuMDAwKmtXKQ0KMS0wOjQxLjcuMCgwMC4wMTcqa1cpDQoxLTA6NjEuNy4wKDAwLjA3OSprVykNCjEtMDoyMi43LjAoMDAuNDAzKmtXKQ0KMS0wOjQyLjcuMCgwMC4wMDAqa1cpDQoxLTA6NjIuNy4wKDAwLjAwMCprVykNCjAtMToyNC4xLjAoMDAzKQ0KMC0xOjk2LjEuMCg0NzMwMzAzMTM5MzMzNDMwMzMzMjMwMzEzNTMwMzMzMTM2KQ0KMC0xOjI0LjIuMSgxODEwMTgwOTAwMDBTKSgwNDI5Ny4yNTYqbTMpDQohMUExRg0K'
[2018-10-18 10:05:34,675] ERROR    @ services | Content or CRC data not found
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services.py", line 189, in telegram_to_reading
    verify_telegram_checksum(data=data)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services.py", line 123, in verify_telegram_checksum
    raise InvalidTelegramError('Content or CRC data not found')
dsmr_datalogger.exceptions.InvalidTelegramError: Content or CRC data not found

In dsmr_backend.log zie ik - cat dsmr_backend.log

...
Command completed. Sleeping for 1 second(s)...

2018-10-17 07:56:53.725427+02:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2018-10-17 07:56:55.219701+02:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

<knip>

2018-10-17 21:12:08.790128+02:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2018-10-17 21:12:09.991307+02:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2018-10-17 21:12:11.201347+02:00: Calling backend services
 >>> Uncaught exception :: FATAL:  the database system is shutting down
FATAL:  the database system is shutting down
 >>> <bound method AppConfig._on_backend_called_signal of <AppConfig: dsmr_datalogger>> ::   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/dispatch/dispatcher.py", line 210, in send_robust
    response = receiver(signal=self, sender=sender, **named)

  File "/home/dsmr/dsmr-reader/dsmr_datalogger/apps.py", line 20, in _on_backend_called_signal
    dsmr_datalogger.services.apply_data_retention()

  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services.py", line 299, in apply_data_retention
    settings = RetentionSettings.get_solo()

[... 871 lines cut ...]
    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 216, in ensure_connection
    self.connect()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 194, in connect
    self.connection = self.get_new_connection(conn_params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/postgresql/base.py", line 168, in get_new_connection
    connection = Database.connect(**conn_params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/psycopg2/__init__.py", line 130, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
Command completed. Sleeping for 1 second(s)...

Detected signal #15, exiting on next run...
Exited due to signal detection
Starting infinite command loop...
2018-10-17 21:12:55.877651+02:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2018-10-17 21:12:58.405958+02:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

Enig idee wat de oorzaak/oplossing is of kan zijn?

raoulteeuwen, 2018-10-18:
Mmmm, i already tried lots of things before sending in the above report ... but after doing that, i decided to update hassbian and home assistant ... and after a reboot, i see one new data point coming ... and than nothing again. I remember i also added https://www.home-assistant.io/components/sensor.dsmr/ around the time the problems started ... so i just deactivated that code, rebooted, and now i get new data in the graphs ... so for now, it seems to work again...

dennissiemensma, 2018-10-19:
Bedankt voor je melding. Het zou ermee te maken kunnen hebben dat beide applicaties van dezelfde seriele poort proberen te lezen. De een of de ander stopt dan met data ontvangen en dan gebeuren er rare dingen.

Je kunt DSMR-reader configureren om via MQTT alle telegrammen door te sturen naar Home Assistant. Dan heb je in principe de gegevens beschikbaar in beide systemen. Ik heb alleen geen exacte configuratie voor je, ik weet alleen dat er gebruikers zijn die dat werkend hebben (ik gebruik zelf geen HA).

dennissiemensma, 2018-10-20:
Voor nu sluit ik dan deze. Mocht je toch nog tegen issues aanlopen dan hoor ik het graag.

closed 2018-10-20 by dennissiemensma