dsmrreader/dsmr-reader #535: All time highs empty

swerveshot, 2018-10-19:
### Environment specifics
* DSMR-reader version: `v1.25`
* Hardware: `Odroid U3`
* DSMR protocol: `v4`

### Description
The all time highs feature does not display any data. I noticed this issue a few releases ago (not sure which one) so it's not an issue specifically releated to the v1.25 release.

[image]

dennissiemensma, 2018-10-19:
Bedankt voor je melding. Wil je eens in de database kijken wat er daar staat?

```
sudo su - postgres
psql dsmrreader
select * from dsmr_stats_electricitystatistics;
```

swerveshot, 2018-10-19:
Absoluut! Kijk eens aan.

` id | highest_usage_l1_timestamp | highest_usage_l2_timestamp | highest_usage_l3_timest
amp | highest_return_l1_timestamp | highest_return_l2_timestamp | highest_return_l3_tim
estamp | highest_usage_l1_value | highest_usage_l2_value | highest_usage_l3_value | hig
hest_return_l1_value | highest_return_l2_value | highest_return_l3_value               
----+----------------------------+----------------------------+------------------------
----+-----------------------------+-----------------------------+----------------------
-------+------------------------+------------------------+------------------------+----
---------------------+-------------------------+-------------------------              
  1 |                            |                            |                        
    |                             |                             |                      
       |                        |                        |                        |    
                     |                         |                                       
(1 row)`

dennissiemensma, 2018-10-19:
Hmm dat is vrij weinig. Eventueel kun je proberen om die data te legen, hij zou dan een nieuwe moeten maken bij de eerstvolgende meting:

```
sudo su - postgres
psql dsmrreader
truncate table dsmr_stats_electricitystatistics;
```

Alleen vermoed ik dat je op hetzelfde uitkomt.

* Klopt het dat je een 1-fasen meter hebt? Mogelijk zit het verschil daar in, gezien het bij mij met 3-fasen wel werkt. Ik zal eens kijken.

swerveshot, 2018-10-19:
Dat maakt helaas niets uit. Nog steeds hetzelfde resultaat.

dennissiemensma, 2018-10-20:
In dat geval mag je wel even een branch draaien met wat debug prints. Ik ben benieuwd wat er uit komt:

```
sudo su - dsmr
git fetch
git checkout -b 535-electricity-statistics-debug origin/535-electricity-statistics-debug
./post-deploy.sh
```

Ik ga er vanuit dat je de standaard datalogger gebruikt. Er komt extra output in de log te staan:
```
CTRL + D
 sudo supervisorctl tail -f dsmr_datalogger
```

Je ziet dan iets als:
```
update_electricity_statistics() - Start
update_electricity_statistics() - Reading trigger: 2018-10-20 07:02:18+00:00
update_electricity_statistics() - Reading phase_currently_delivered_l2     01.593
update_electricity_statistics() - Reading phase_currently_returned_l3     00.000
update_electricity_statistics() - Reading phase_currently_delivered_l3     00.000
update_electricity_statistics() - Reading phase_currently_returned_l1     00.000
update_electricity_statistics() - Reading phase_currently_delivered_l1     00.014
update_electricity_statistics() - Reading phase_currently_returned_l2     00.000
update_electricity_statistics() - Current check: phase_currently_delivered_l2 / highest_usage_l2
update_electricity_statistics() -     reading_value / highest_value:   01.593 / 3.184
update_electricity_statistics() - Current check: phase_currently_returned_l3 / highest_return_l3
update_electricity_statistics() -     reading_value / highest_value:   00.000 / 0
update_electricity_statistics() - Current check: phase_currently_delivered_l3 / highest_usage_l3
update_electricity_statistics() -     reading_value / highest_value:   00.000 / 0
update_electricity_statistics() - Current check: phase_currently_returned_l1 / highest_return_l1
update_electricity_statistics() -     reading_value / highest_value:   00.000 / 0
update_electricity_statistics() - Current check: phase_currently_delivered_l1 / highest_usage_l1
update_electricity_statistics() -     reading_value / highest_value:   00.014 / 2.070
update_electricity_statistics() - Current check: phase_currently_returned_l2 / highest_return_l2
update_electricity_statistics() -     reading_value / highest_value:   00.000 / 0
update_electricity_statistics() - Dirty: False
update_electricity_statistics() - End
```

Graag ontvang ik een kopie van zo'n blok, zodat ik kan zien wat die precies doet.

swerveshot, 2018-10-21:
Ga ik uitvoeren, maar voor ik dat doe..... Is het de bedoeling dat ik dan nog eerst naar de `~/dsmr-reader` directory ga en de VirtualEnvironment activeer met `source ~/.virtualenvs/dsmrreader/bin/activate`?

dennissiemensma, 2018-10-21:
Correct. Ik ga er vanuit dat je die dingen in de bashrc hebt gezet, maar als dat niet zo is dan zul je dat elke keer handmatig moeten doen.

swerveshot, 2018-10-22:
Done, hier is de output.
```
update_electricity_statistics() - Start
update_electricity_statistics() - Reading trigger: 2018-10-22 07:06:53+00:00
update_electricity_statistics() - Reading phase_currently_returned_l2     None
update_electricity_statistics() - Reading phase_currently_delivered_l1     None
update_electricity_statistics() - Reading phase_currently_returned_l3     None
update_electricity_statistics() - Reading phase_currently_delivered_l2     None
update_electricity_statistics() - Reading phase_currently_returned_l1     None
update_electricity_statistics() - Reading phase_currently_delivered_l3     None
update_electricity_statistics() - Current check: phase_currently_returned_l2 / highest_return_l2
update_electricity_statistics() -     reading_value / highest_value:   0 / 0
update_electricity_statistics() - Current check: phase_currently_delivered_l1 / highest_usage_l1
update_electricity_statistics() -     reading_value / highest_value:   0 / 0
update_electricity_statistics() - Current check: phase_currently_returned_l3 / highest_return_l3
update_electricity_statistics() -     reading_value / highest_value:   0 / 0
update_electricity_statistics() - Current check: phase_currently_delivered_l2 / highest_usage_l2
update_electricity_statistics() -     reading_value / highest_value:   0 / 0
update_electricity_statistics() - Current check: phase_currently_returned_l1 / highest_return_l1
update_electricity_statistics() -     reading_value / highest_value:   0 / 0
update_electricity_statistics() - Current check: phase_currently_delivered_l3 / highest_usage_l3
update_electricity_statistics() -     reading_value / highest_value:   0 / 0
update_electricity_statistics() - Dirty: False
update_electricity_statistics() - End
```

Grappig; de output lijkt in UTC. Weet niet of dat nog interessant is.

swerveshot, 2018-10-22:
Logisch dat je daar vanuit staat. Staat ook keurig in de documentatie. :monkey_face:
Ben geswitched naar de branch die je hebt klaargezet en dit is de output.
```
update_electricity_statistics() - Start
update_electricity_statistics() - Reading trigger: 2018-10-22 07:06:54+00:00
update_electricity_statistics() - Reading phase_currently_returned_l2     None
update_electricity_statistics() - Reading phase_currently_delivered_l1     None
update_electricity_statistics() - Reading phase_currently_returned_l3     None
update_electricity_statistics() - Reading phase_currently_delivered_l2     None
update_electricity_statistics() - Reading phase_currently_returned_l1     None
update_electricity_statistics() - Reading phase_currently_delivered_l3     None
update_electricity_statistics() - Current check: phase_currently_returned_l2 / highest_return_l2
update_electricity_statistics() -     reading_value / highest_value:   0 / 0
update_electricity_statistics() - Current check: phase_currently_delivered_l1 / highest_usage_l1
update_electricity_statistics() -     reading_value / highest_value:   0 / 0
update_electricity_statistics() - Current check: phase_currently_returned_l3 / highest_return_l3
update_electricity_statistics() -     reading_value / highest_value:   0 / 0
update_electricity_statistics() - Current check: phase_currently_delivered_l2 / highest_usage_l2
update_electricity_statistics() -     reading_value / highest_value:   0 / 0
update_electricity_statistics() - Current check: phase_currently_returned_l1 / highest_return_l1
update_electricity_statistics() -     reading_value / highest_value:   0 / 0
update_electricity_statistics() - Current check: phase_currently_delivered_l3 / highest_usage_l3
update_electricity_statistics() -     reading_value / highest_value:   0 / 0
update_electricity_statistics() - Dirty: False
update_electricity_statistics() - End
```
Zie inderdaad dat hier de highest_value steeds 0 is. En het valt me op dat de tijd UTC lijkt te zijn. Weet niet of dat nog interessant is.

dennissiemensma, 2018-10-22:
Dankje, dat is helder. Ik haal de statistieken uit de fasen-informatie, maar dat is wel iets wat handmatig ingeschakeld moet worden bedenk ik me nu.

Kan het zijn dat je in de admin-interface bij `Datalogger configuration`  de optie `Track electricity phases` uit hebt staan? Zo ja, kun je die inschakelen om te zien of dat het oplost?

swerveshot, 2018-10-22:
Ik ga eens controleren. Dan nog weer een n00b vraag. Zou je me ook kunnen vertellen hoe ik weer naar de master branch switch terug kom? Of kostte de maintenance branch geen extra diskspace? Had namelijk de indruk dat de vrije ruimte rap afnam.

dennissiemensma, 2018-10-22:
De extra branch kost vrijwel geen extra ruimte als het goed is. Je kunt terug met:

```
sudo su - dsmr
git checkout master
./post-deploy.sh
```

swerveshot, 2018-10-22:
Okay, het viel wel mee met de diskspace dus laat het voorlopig nog wel draaien zo.

Het aanpassen van de `Track electricity phases` optie leverde inderdaad een verschil op in de logging. Ik krijg nu deze info.

```
update_electricity_statistics() - Start
update_electricity_statistics() - Reading trigger: 2018-10-22 21:55:34+00:00
update_electricity_statistics() - Reading phase_currently_delivered_l1     00.040
update_electricity_statistics() - Reading phase_currently_delivered_l2     None
update_electricity_statistics() - Reading phase_currently_returned_l2     None
update_electricity_statistics() - Reading phase_currently_returned_l1     00.000
update_electricity_statistics() - Reading phase_currently_returned_l3     None
update_electricity_statistics() - Reading phase_currently_delivered_l3     None
update_electricity_statistics() - Current check: phase_currently_delivered_l1 / highest_usage_l1
update_electricity_statistics() -     reading_value / highest_value:   00.040 / 0.044
update_electricity_statistics() - Current check: phase_currently_delivered_l2 / highest_usage_l2
update_electricity_statistics() -     reading_value / highest_value:   0 / 0
update_electricity_statistics() - Current check: phase_currently_returned_l2 / highest_return_l2
update_electricity_statistics() -     reading_value / highest_value:   0 / 0
update_electricity_statistics() - Current check: phase_currently_returned_l1 / highest_return_l1
update_electricity_statistics() -     reading_value / highest_value:   00.000 / 0
update_electricity_statistics() - Current check: phase_currently_returned_l3 / highest_return_l3
update_electricity_statistics() -     reading_value / highest_value:   0 / 0
update_electricity_statistics() - Current check: phase_currently_delivered_l3 / highest_usage_l3
update_electricity_statistics() -     reading_value / highest_value:   0 / 0
update_electricity_statistics() - Dirty: False
update_electricity_statistics() - End
```

Ik heb geen idee hoe deze setting uit is gegaan.

De statistics page toont nu een hoogste waarde, maar als ik me niet vergis was er ook ooit een all-time low. Die zie ik nu nog niet verschijnen. Is dat een kwestie van tijd misschien?

[image]

Btw, ik hoop dat je niet plat bent gespammed vandaag. Ik heb verschillende keren een bericht gepost totdat ik doorhad dat er [een grote storing bij GitHub was](https://status.github.com/messages).

dennissiemensma, 2018-10-23:
Fijn dat het nu werkt. Ik zal kijken of ik dit duidelijker kan maken in de interface, gezien ik er geen rekening mee had gehouden.

De "all-time low" heb ik in een eerdere versie weggehaald toen ik dit op de schop heb gegooid. De oude manier van meten hield namelijk geen rekening met retentie (hij deed ook telkens een live berekening voor de hoogste en laagste waardes). 
De huidige versie slaat die hoogste waardes nu apart op, zoals je misschien al kon zien in de debug log. Dit overleeft retentie en is tevens supersnel met weergeven. Wellicht dat ik het ooit toevoeg voor "all-time low", maar eigenlijk zegt het niet heel veel en heb ik het daarom weggelaten.

swerveshot, 2018-10-23:
Okay Dennis het is een duidelijk verhaal. Jammer van de gesneuvelde "all-time low" feature. Een van de redenen waarom ik dsmr-reader zo'n gave toepassing vind is de mogelijkheid om sluipverbruik te herkennen en zo mijn totale energieverbruik omlaag te krijgen. Om dan te kunnen zien dat je weer een nieuw record hebt verbroken als 's avonds alle apparaten weer uitschakelen is dan een mooie bonus. Dus ben het niet echt eens met je stelling dat het een getal is dat niet zoveel zegt. Maar goed ik snap dat je keuzes moet maken in de ontwikkeling van je project en bepaalde functies zwaarder wegen dan de andere.

Enorm bedankt wederom voor de hulp en mocht je het issue nog open willen laten staan voor de "all-time low" is dat mooi, maar wil je 'm sluiten is dat ook best.

dennissiemensma, 2018-10-23:
Je bent de eerste die ik er over hoor namelijk, maar als er behoefte voor is, kan ik het zeker wel implementeren hoor.

swerveshot, 2018-10-23:
Hahahaha, eentje moet de eerste zijn natuurlijk. Mij zou je er erg blij mee maken. Maar als het zo'n niche functie waarom zou je er dan moeite voor doen? Is er een eenvoudige wijze om in github mensen te laten stemmen op een feature request? En anders zou ik zeggen; sluit het issue joh. Je hebt vast meer op je backlog staan.

dennissiemensma, 2018-10-24:
Ik heb de feature ingebouwd voor de volgende release. Eventueel kunnen je wisselen naar `development` om het eerder te gebruiken.

closed 2018-10-24 by dennissiemensma

dennissiemensma, 2018-10-26:
Goed om te horen! Ik was vergeten instructies voor je achter te laten om te wisselen naar `development`.

Voor de applicatie maken symlinks niet uit, deze volgt gewoon de regels van Linux.

dennissiemensma, 2018-10-28:
Dit zit in de nieuwe release.

Hiermee kun je weer terug naar `master`: https://github.com/dennissiemensma/dsmr-reader/issues/535#issuecomment-431914592

swerveshot, 2018-10-28:
Je houdt het tempo er lekker in. :ok_hand:
Als je wilt dat ik iets voor je test; laat maar weten.