dsmrreader/dsmr-reader #536: Opschonen oude data ivm retentie lijkt PVOutput upload te storen

pyrocumulus, 2018-10-20:
* DSMR reader *: `v1.24.0`
* Database *: MySQL, remote op NAS
* Hardware *: `RaspberryPi 2b`
* DSMR-protocol versie van slimme meter *: `v5`

Gisteravond heb ik voor het eerst de retentie aangepast van alles naar een half jaar. Ik kwam erachter dat mijn database al ruim 7 GB groot was. Beetje gortig ;-)

Maar wat ik vanmorgen zag bij PVOutput:
[screenshot_20181020-115935_chrome](https://user-images.githubusercontent.com/1721382/47254239-b47ef180-d45f-11e8-98a0-5f83e38efa04.jpg)

Het lijkt erop dat het opschonen iets doet dat de upload in de weg zit? Ik zie geen missende metingen in DSMR zelf.

Of denk je dat het iets anders is? Weet namelijk niet of de tijd past bij het opschoonproces namelijk. Verder geen groot issue maar viel wel direct op als je de avond daarvoor de retentie hebt aangepast :)

dennissiemensma, 2018-10-20:
Bedankt voor je melding. Ik heb er niet direct een verklaring voor. Het zou er inderdaad mee te maken kunnen hebben, al zou de retentie alleen tussen middernacht en 6:00 moeten lopen (hij kan natuurlijk wel eerder gestart zijn en uitlopen).

Je zou eventueel debug-logging kunnen inschakelen: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#logging

Mocht het morgenochtend wéér gebeuren, dan is goed te zien wat die precies doet (in de log van `dsmr_backend`). En dus mogelijk ook wat een oorzaak is.

pyrocumulus, 2018-10-20:
Oké, dat lijkt wel aan de late kant te zijn dan. Dat zei mijn gevoel ook al maar ik vond het simpelweg te toevallig (PV nerd als ik ben, ik check die grafiek elke dag).

Ik heb debug logging ingeschakeld, ga ik morgen even zien of ik een herhaling zie en zoja wat het debug log daarover zegt.

Ik kom erop terug :)

pyrocumulus, 2018-10-21:
Vanmorgen hetzelfde verhaal, paar gaten in de grafiek. Waaronder specifiek 7:45. Na het bekijken van de log (zie een uitsnede [hier](https://gist.github.com/pyrocumulus/ebf38081b0afe953cfd53a4e2356152a) van relevante periode) is de oorzaak wel redelijk duidelijk. De PVOutput upload raakt - vermoedelijk door de intensieve retention cleanup - out-of-sync. 

Reconstructie:
De eerste upload in de log wordt om 7:34 uitgevoerd en stelt de volgende upload te moeten plannen op 7:35, dus 1 minuut later. Dat is op zich al gek genoeg maar de volgende upload wordt pas daadwerkelijk uitgevoerd op 7:41. Deze upload zegt de volgende te plannen op 7:45 maar die komt pas op 7:48. En die laatste timestamp wordt door de 5-min interval afronding, effectief 7:50. En dan krijg je een gaatje :-)

Ik zie tussendoor volop retention cleaning up staan, tot na 8 uur lijkt het wel. Enig idee hoe ik die retention zo kan gaan doen, dat deze niet teveel storing in de gewone bezigheden brengt? Misschien iets minder 'uren' verwerken per trigger moment? Ik denk namelijk dat de backend in deze situatie soms net iets teveel te doen heeft. Overigens zijn het paar een paar gaatjes, dus het is niet alsof het helemaal kapot is ;-)

pyrocumulus, 2018-10-22:
Vandaag weer hetzelfde maar de gaten komen ook nog na 10h voor. Tot hoe laat loopt die cleanup door?

dennissiemensma, 2018-10-22:
Bedankt voor de informatie!

* Het backend-proces doet slechts 1 ding, dus als die bezig is met opschonen, kan die niets anders doen, zoals uploaden.
* Momenteel schoont die al maximaal 24 uur op, echter zie je goed dat die de meeste tijd kwijt is met het zoeken naar data die weg kan:

```
[2018-10-21 07:48:52,864] DEBUG    Retention | Cleaning up: 2017-05-27 20:00:00+00:00 (GasConsumption)
<<< ZOEKEN NAAR METINGEN >>>
[2018-10-21 07:55:25,062] DEBUG    Retention | Cleaning up: 2017-05-24 19:00:00+00:00 
(DsmrReading)
[2018-10-21 07:55:25,307] DEBUG    Retention | Cleaning up: 2017-05-24 20:00:00+00:00 (DsmrReading)

```
Het opschonen zelf kost een fractie van een seconde (per uur opschoning).

* Het opschonen zal in het begin heel traag zijn en naarmate er minder data is, steeds sneller. Een database van 7 GB is dan ook exceptioneel eerlijk gezegd.

* Mogelijk zit er nog een bug in de code. Ik zie dat de code kijkt of het al 6:00 AM is geweest, echter doe ik daar geen expliciete check op tijdzone en in de wintermaanden kan dat tot 2 uur verschil zijn t.o.v. UTC. Alleen is het wel vreemd dat die dat ná 10 uur nog doet.

dennissiemensma, 2018-10-22:
Ik heb verder niet concreet een oplossing voor je, alleen een handmatige fix. Je kunt eventueel tijdelijk hierin wat aanpassen (vergeet niet het proces te herstarten), zodat je wat meer controle hebt over wanneer die opschoont:

https://github.com/dennissiemensma/dsmr-reader/blob/master/dsmr_datalogger/services.py#L318

Je kunt hem op `4` ipv `6` zetten om te zien of die inderdaad UTC pakt. Dan zou die om 6 uur onze tijd moeten stoppen (of beter gezegd, niet opnieuw moeten starten).

(Ik zie ook dat hij checkt met `> 6` ipv `>= 6`, dat helpt ook niet mee).

dennissiemensma, 2018-10-22:
En wat je ook nog kan doen is `MAX_HOURS_CLEANUP` (daaronder) **hoger** zetten, zodat die meer tegelijkertijd opschoont. Want nu zoekt die effectief 6 minuten voor 24 uur aan metingen. Wellicht kan die 6 minuten beter gebruikt worden om bijvoorbeeld 2, 3 of zelfs 7 dagen per keer op te schonen. Je kunt er zelf wat mee spelen.

dennissiemensma, 2018-10-22:
Ik zal hier later nog verder naar kijken om die tijdstip-check goed te fixen. Sowieso klopt die nu niet (hij doet effectief tot 7:00 en mogelijk zelfs tot 9:00 als de tijdzone niet klopt.

pyrocumulus, 2018-10-23:
Thanks voor uitgebreide info! Ik ga de eindtijd inderdaad even terugzetten naar >= 4 om te kijken of dat een UTC dingetje is. En zal ook eens spelen met die `MAX_HOURS_CLEANUP` om te zien of dat het zoeken niet veel duurder maakt.

Ik had al zo'n idee dat de backend inderdaad maar één ding tegelijkertijd doet. Wat voor de toekomst misschien ook een mogelijkheid is, is de retention in een los supervisor proces draaien. Zoals dat voor mqtt ook gedaan is. Dan zou de backend in elk geval door lopen.

dennissiemensma, 2018-10-23:
Ik heb wat wijzigingen gedaan om bovenstaande te fixen. Voor de volgende release.

* Tijdzone in de code was inderdaad UTC.
* Bovenstaande instellingen zijn straks los in te stellen in de settings.py.

closed 2018-10-23 by dennissiemensma

dennissiemensma, 2018-10-28:
Staat in nieuwe release.

dennissiemensma, 2018-10-28:
Is de rententie inmiddels bij qua achterstand?

pyrocumulus, 2018-10-28:
Sinds een paar dagen zijn de gaten uit de grafiek in elk geval weg :-) kan
nu even niet kijken maar ik denk dat de retentie al bij is inderdaad.

Ben benieuwd naar de volgende release, mocht het dan weer voorkomen dan kan
ik makkelijk zelf spelen met die settings.

Sent from a smartphone

pyrocumulus, 2018-10-28:
Ik kan eerlijk gezegd niets omtrent retentie vinden in de huidige backend log, ik vermoed dus dat er niets meer is om op te ruimen dan. Zou ook verklaren dat de gaten weg zijn want de backend heeft het nu lang zo druk niet meer 👍 

Ik beschouw het als opgelost!

dennissiemensma, 2018-10-29:
Top, fijn om te horen.