dsmrreader/dsmr-reader #540: Statistieken werken niet

RichieB2B, 2018-10-26:
### Jouw omgeving
* DSMR-reader versie: `v1.25`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 3B+`
* DSMR-protocol versie van slimme meter: `v5`

### Bug, wens of iets anders
Als ik op `statistics` in het linker menu klik krijg ik deze foutmelding:
```
Server Error

Exception:

MultipleObjectsReturned: get() returned more than one EnergySupplierPrice -- it returned 2!
Traceback:

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/handlers/base.py", line 126, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/views/generic/base.py", line 69, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/views/generic/base.py", line 89, in dispatch
    return handler(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/views/generic/base.py", line 150, in get
    context = self.get_context_data(**kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/views/statistics.py", line 41, in get_context_data
    context_data['energy_prices'] = EnergySupplierPrice.objects.by_date(today)

  File "/home/dsmr/dsmr-reader/dsmr_consumption/models/energysupplier.py", line 12, in by_date
    start__lte=target_date,

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/query.py", line 407, in get
    (self.model._meta.object_name, num)
```

dennissiemensma, 2018-10-26:
Kan het zijn dat je twee energiecontracten hebt ingesteld die elkaar overlappen? Daar lijkt de melding namelijk op.

RichieB2B, 2018-10-27:
Ja, inderdaad. Ik heb 1 energiecontract voor gas en een andere voor elektriciteit. Die overlappen qua looptijd met elkaar. Merge ik de contracten (wat niet logisch is, want ze hebben een andere looptijd) tot 1 contract dan werkt alles wel, ook #538

dennissiemensma, 2018-10-28:
Je bent tot nu toe de tweede die ik er over hoor. Wellicht dat ik het alsnog ooit een keertje lostrek.

glsf91, 2019-02-02:
Ik loop ook tegen dit probleem aan. Ook 2 leveranciers.
Dus nummer 3 :-)

dennissiemensma, 2019-09-29:
Ik ben hier van het weekend een paar uur mee aan het stoeien geweest om te kijken of het makkelijk door te voeren is en ik heb besloten het voor nu te laten voor wat het is. Ik begrijp dat het voor jullie drieen wat onlogisch blijft werken maar:

* Op een aantal plekken waar ik prijzen weergeef of bereken, moet ik dubbel de prijzen ophalen, samenvoegen en rekening mee houd. Hoewel dat niet complex is, wordt de code er wat omslachtig van.
* De contractpagina wordt er aardig complexer van. Ik kan waarschijnlijk contracten met gelijke looptijden samenvoegen (effectief het huidige model), maar zodra ze deels overlappen kan ik er visueel niet iets logisch van maken. Verder ook hier aardig wat dubbele code om door beide type prijzen heen te lopen.
* Het invoeren van de prijzen zal voor alle andere gebruikers dubbel moeten. Hoewel dat qua interface vast op te lossen is met een handmatige pagina, past dat niet in het huidige standaard django-admin framework dat ik gebruik. Daarmee kan ik relatief snel nieuwe instellingen of beheerschermen toevoegen, al verdienen ze niet de schoonheidsprijs voor logische indeling of gebruikersgemak.

Dus ik ben het in principe met jullie eens dat het los zou moeten, maar in de praktijk heeft dat wat veel gevolgen.

closed 2019-09-29 by dennissiemensma

RichieB2B, 2019-09-29:
Bedankt voor de upstate. Als je de prijzen voor gas en elektra splitst hoef je die toch niet in de code weer samen te voegen? Ook overlap hoef je geen rekening mee te houden. Elektra contracten overlappen niet met een ander elektra contract. En een gas contract overlapt niet met een ander gas contract. Ook de admin pagina klinkt simpel: 1 voor elektra prijzen en 1 voor gas prijzen. Dus echt splitsen i.p.v. iets cosmetisch wat je nu van plan was zoals ik het lees.

dennissiemensma, 2019-09-30:
Op alle plekken in de code moet ik beide prijzen weten, dus ik moet die weer samenvoegen. En qua admin bedoel ik dus juist anderom, dat straks een hoop gebruikers hetzelfde contract meerdere keren moeten invoeren, omdat het dan gesplitst is. Dat is wel op te lossen qua interface, maar dat is op dit moment niet mogelijk. Ik heb nu niet echt een oplossing die voor iedereen goed is.