dsmrreader/dsmr-reader #541: Extreme amount of duplicated notifications

GreyEarl, 2018-10-27:
This morning/night I received hundereds of daily consumption notifications through Pushover. All duplications. I even got a warning from Pushover that the designated channel exceeded the 7500 message limit.

tjla, 2018-10-27:
Same problem (using Pushover) here, >7000 messages,

michielvisser, 2018-10-27:
Here as well, my API of Prowl stopped after 1000 messages with an error stating quota exceeded. I have not seen this issue before. BTW, at the moment I am stil running an older version of DSMR reader - 1.21.1.

Michiel

dennissiemensma, 2018-10-27:
Bedankt voor jullie meldingen. Ik ervaarde vannacht hetzelfde en vroeg me al af of ik de enigste was. 

De oplossing voor nu is om in DSMR-reader notificaties uit te zetten, totdat ik er achter ben waar dit opeens vandaan komt en ik een fix heb om te voorkomen dat dit ooit weer gebeurt.

dennissiemensma, 2018-10-27:
Het heeft inderdaad te maken met de wisseling zomertijd/wintertijd. Ik heb dit 5 maanden terug gerefactored toen NotifyMyAndroid ermee stopte en ik Pushover heb toegevoegd. 

Daarbij vraag ik aan het systeem, ná het versturen van de notificatie, om voor mij de volgende dag om 2:00 te berekenen. Die gebruik ik als check-tijd om te bepalen of er vandaag al een notificatie is verstuurd. Dat gaat altijd goed, totdat die aankomende nacht tegenkomt, omdat die dat niet weet of ik 2:00 zomertijd of wintertijd bedoel. Python geeft daar simpelweg een foutmelding voor. 
En doordat die eerst de notificatie verstuurt en daarna de 'check-tijd' een dag opschuift, komt die telkens een niet-aangepaste check-tijd tegen en blijft die het proberen. Normaal gesproken een wat lelijke situatie met niet zoveel gevolgen, maar nu wel omdat het daadwerkelijk een notificatie verstuurt naar iedereen en dat vrij hinderlijk is.

Ik zal kijken hoe ik dit goed oplos. Waarschijnlijk zet ik voor nu het notificatie-tijdstip simpelweg op 6:00, zodat dit niet meer kan gebeuren (en als het wel gebeurt dan kan iemand sneller reageren ipv wakker te worden met duizenden notificaties).

dennissiemensma, 2018-10-27:
Ik laat het ticket trouwens nog even openstaan bedenk ik me net, want er zijn vast meer die hier last van hebben en dit ticket mogelijk nog niet gezien hebben.

GreyEarl, 2018-10-27:
Awesome! Bedankt voor de uitgebreide toelichting en het snelle handelen naar een oplossing!

closed 2018-10-28 by dennissiemensma

dennissiemensma, 2018-10-28:
Fix staat nu in nieuwere release.