dsmrreader/dsmr-reader #543: Ongeconfigureerde MQTT blijft zichzelf herstarten

RichieB2B, 2018-10-28:
### Jouw omgeving
* DSMR-reader versie: `v1.25`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 3B+`
* DSMR-protocol versie van slimme meter: `v5`

### Bug, wens of iets anders
Ik gebruik MQTT (nog) niet, dus ik heb het niet geconfigureerd. Hierdoor blijft het `dsmr_mqtt` proces zichzelf herstarten.
Uit `/var/log/syslog`:
```
Oct 28 17:44:28 pi supervisord[414]: 2018-10-28 17:44:28,049 INFO exited: dsmr_mqtt (exit status 1; not expected)
Oct 28 17:44:28 pi supervisord[414]: 2018-10-28 17:44:28,419 INFO spawned: 'dsmr_mqtt' with pid 25634
Oct 28 17:44:29 pi supervisord[414]: 2018-10-28 17:44:29,424 INFO success: dsmr_mqtt entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
```
Uit `/var/log/supervisor/dsmr_mqtt.log`:
```
[2018-10-28 17:38:04,947] WARNING  MQTT | No hostname found in settings, restarting in a minute...
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/__init__.py", line 371, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/__init__.py", line 365, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/base.py", line 288, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/base.py", line 335, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 40, in handle
    self.data = self.initialize()
  File "/home/dsmr/dsmr-reader/dsmr_mqtt/management/commands/dsmr_mqtt.py", line 25, in initialize
    self.mqtt_client = dsmr_mqtt.services.broker.initialize()
  File "/home/dsmr/dsmr-reader/dsmr_mqtt/services/broker.py", line 23, in initialize
    raise StopInfiniteRun()
dsmr_backend.mixins.StopInfiniteRun
```
Het zou netter zijn als het MQTT proces zou herkennen dat het niet geconfigureerd is en dan een lange sleep doet om daarna te kijken of de hostname nu wel is gezet.

dennissiemensma, 2018-10-29:
Ik kon hier destijds geen nette oplossing voor vinden. De opzet zoals die nu via Supervisor is gedaan, heeft nog wel de voorkeur, dus voor nu laat ik dit voor wat het is.

Wat je wel kan doen, is simpelweg `dsmr_mqtt` stoppen in Supervisor zodat die niets meer doet. Je kunt zelfs `autostart` uitzetten in het proces in `/etc/supervisor/conf.d/dsmr-reader.conf`. Vergeet niet om `sudo supervisorctl reread` en `sudo supervisorctl update` te doen daarna.

RichieB2B, 2018-10-29:
Zou dit dan niet de default moeten zijn met een melding om `dsmr_mqtt` aan te zetten zodra de hostname wordt ingevuld?

dennissiemensma, 2018-10-29:
Ja dat zou kunnen, maar dan vereist het ook handwerk in de Supervisor-config om `autostart` weer aan te zetten.

dennissiemensma, 2019-09-30:
Aangepast voor v2.6 release

closed 2019-09-30 by dennissiemensma

dennissiemensma, 2019-10-07:
Uitgebracht in [v2.6](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v2.6.0).