dsmrreader/dsmr-reader #547: DSMR never stops retrying to make a PostgreSQL, when it fails

ualex73, 2018-11-01:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `1.26.0`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `Intel NUC (Docker)`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v5`

My PostgreSQL became corrupt (root-cause unknown), but when DSMR Reader tries to make its nightly backup at 01:00, it failed and it tries to continue to do it over and over again - i only saw it around 09:30.

The first PostgreSQL logging:
---
2018-11-01 01:01:44 CET [1167-10] LOG:  server process (PID 24556) was terminated by signal 11: Segmentation fault
2018-11-01 01:01:44 CET [1167-11] DETAIL:  Failed process was running: COPY public.dsmr_datalogger_dsmrreading (id, "timestamp", electricity_delivered_1, electricity_returned_1, el
ectricity_delivered_2, electricity_returned_2, electricity_currently_delivered, electricity_currently_returned, extra_device_timestamp, extra_device_delivered, processed, phase_cur
rently_delivered_l1, phase_currently_delivered_l2, phase_currently_delivered_l3, phase_currently_returned_l1, phase_currently_returned_l2, phase_currently_returned_l3) TO stdout;
2018-11-01 01:01:44 CET [1167-12] LOG:  terminating any other active server processes
2018-11-01 01:01:44 CET [25240-1] dsmrreader@dsmrreader WARNING:  terminating connection because of crash of another server process
2018-11-01 01:01:44 CET [25240-2] dsmrreader@dsmrreader DETAIL:  The postmaster has commanded this server process to roll back the current transaction and exit, because another ser
ver process exited abnormally and possibly corrupted shared memory.
2018-11-01 01:01:44 CET [25240-3] dsmrreader@dsmrreader HINT:  In a moment you should be able to reconnect to the database and repeat your command.
2018-11-01 01:01:44 CET [23388-1] dsmrreader@dsmrreader WARNING:  terminating connection because of crash of another server process
2018-11-01 01:01:44 CET [23388-2] dsmrreader@dsmrreader DETAIL:  The postmaster has commanded this server process to roll back the current transaction and exit, because another ser
ver process exited abnormally and possibly corrupted shared memory.

dennissiemensma, 2018-11-02:
Bedankt voor je melding. Ik denk dat ik dit wel kan verbeteren.

Sowieso vind ik het geen slecht idee om MQTT te gebruiken voor status-berichten (zoals wanneer de datalogger achterloopt of, in dit geval, backups mislukken).

ualex73, 2018-11-04:
Dank je 👍 

Ik doe nu 2x per dag een backup, dan heb ik een wat betere backup dan toen het nu gebeurde. Als je de MQTT toevoegd, kan je misschien ook gradaties invoeren? dus iets van warning, error, etc ... Dan kan ik nog een beetje filteren en silence aanzetten e.d. voordat ik het uitstuurt naar mijn mobiel ;-)

dennissiemensma, 2018-11-04:
Ja ik denk dat het alleen 'kritieke' notificaties zullen zijn, wanneer er echt iets aan de hand is.

dennissiemensma, 2019-09-10:
Uiteindelijk ga ik zaken zoals Dropbox sync en backups buiten dit project trekken. Ik ga er nog wel ondersteuning voor bieden, maar dan in de vorm van bestaande 3rd party tools die op de commandline werken (en te schedulen zijn in de crontab). Daar is dan ook zelfs een MQTT-melding aan vast te hangen denk ik in jouw geval. (mosquito_pub aanroepen bij foute exitcode oid).

Voor nu ga ik hier geen betere foutafhandeling voor maken.

closed 2019-09-10 by dennissiemensma