dsmrreader/dsmr-reader #548: Re-insert data from a partial backup/copy

ualex73, 2018-11-01:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.2.6`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `Intel NUC (Docker`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v5`

I had to recover my DSMR Reader database from a dump of yesterday (at 01:00), so i am missing some data of yesterday/today. Before i did the recovery, i made a dump of the database, so i have some data of those 2 days.

Is there any way of importing that data into the DSMR Reader database? Without having to wipe the existing data? I am not an expert in postgresql sorry.

ualex73, 2018-11-04:
Ja, het wordt dus lastig. De ID's zullen nu wel gebruikt zijn.

Wat gebeurd er eigenlijk als ik deze data in 1-2 weken weer toevoegd? Wordt die dan nog verwerkt of genegeerd? Ik dit er aan te denken om een perl/python scriptje te schrijven om dit uit te proberen ;-)

dennissiemensma, 2018-11-04:
Als het goed is kun je de data gewoon toevoegen, maar verwerkt hij het niet zomaar, omdat die de dagen al gehad heeft (nieuwere dagen bedoel ik dan).

Je kunt dan wel in de DB alle dag-, uurstatistieken en consumptie vanaf die datum verwijderen (mits je alle metingen nog hebt!) en vervolgens de metingen vanaf diezelfde datum in bulk op 'overwerkt' zetten. Dan zou hij ze gewoon moeten verwerken op volgorde. Het kan dan wel even een paar uur duren, afhankelijk van de hoeveelheid metingen en de snelheid van je hardware.

dennissiemensma, 2018-11-25:
@ualex73 ben je hier nog uit gekomen? 

Bij iemand anders is het me geluk om de dag- en uurstatistieken achteraf nog toe te voegen, ongeacht de ID's, deze strip je dan handmatig uit de export. In principe zou dat ook voor alle andere data kunnen, maar het vereist wat handwerk qua exporteren en find-replacen in de dump zelf, per tabel.

ualex73, 2018-11-25:
Om eerlijk te zijn, heb ik er geen tijd voor gehad en dus nog niet geprobeerd. Staat nog wel op mijn todo/wishlist.

dennissiemensma, 2019-04-21:
@ualex73 is het inmiddels wel gelukt? Anders laat ik dit voor nu voor wat het is.

closed 2019-05-04 by dennissiemensma