dsmrreader/dsmr-reader #549: Dag Dennis,

alwisiebers, 2018-11-03:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.X.Y`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi X...`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v...`

### Bug, wens of iets anders
* Bijvoorbeeld stappen hoe een probleem te reproduceren is...
* Of simpelweg een beschijving van een feature die je graag wenst te zien...

alwisiebers, 2018-11-03:
Nu het vervolg. Ikheb het al een keer eerder gevraagd en zou het nog steeds fijn vinden als de temperatuur als grafiek ook wordt weergegeven in het archief en niet alleen als gemiddelde in de kop. Hoop dat dit mogelijk wordt. Met een vriendelijke groet, Alwi

closed 2018-11-03 by alwisiebers

dennissiemensma, 2018-11-03:
Ik heb je oude ticket heropend. Ik kan je niets beloven.