dsmrreader/dsmr-reader #550: Vaak meldingen over database in recovery mode

cranberryNL, 2018-11-06:
* DSMR-reader versie: v1.26Y
* Hardware: RaspberryPi 3
* DSMR-protocol versie: 5

De laatste weken heb ik regelmatig problemen om DSMR gegevens te bekijken.  Ik zie dan dergelijke foutmeldingen:

Server Error
Exception:

OperationalError: FATAL:  the database system is in recovery mode
FATAL:  the database system is in recovery mode

Een paar keer F5/CTRL+F5 of een reboot van de RPi lost het probleem meestal voor enige tijd weer op, maar helaas is het altijd maar tijdelijk. In de tussenliggende tijd heb ik o.a. versie 1.24 en 1.25 van de software al gebruikt en dat lijkt geen verschil te maken.
Wat kan de oorzaak hiervan zijn? SD-kaart corruptie? D.m.v. 'dmesg' zie ik geen meldingen die wijzen op hw-problemen of disk errors.

dennissiemensma, 2018-11-08:
Bedankt voor je vraag. Wegens ziekte was ik niet in de gelegenheid om eerder te reageren.

De melding lijkt te komen van de database zelf, PostgreSQL. Als ik in google zoek op de melding kan ik zelf niet veel vinden, maar mogelijk is het wel gerelateerd aan je kaartje, of het OS.

* Heb je al een reboot gedaan en bleven de gegevens van vóór de reboot bewaard?
* Je zou nog even kunnen kijken in de logfile van de DB zelf, meestal in: `/var/log/postgresql/`

cranberryNL, 2018-11-08:
No problem, ik vind het razend interessant, maar uiteindelijk eet ik er geen boterham minder om als DSMR niet helemaal top werkt. Voor jou is het ongetwijfeld ook 'hobby' naast andere, bezigheden... ;)

De RPi heb ik al heel vaak herstart, maar ik zie eigenlijk geen verbetering.
De PostgreSQL log heb ik snel gescand. Het ontbreekt me aan ervaring om een gefundeerde conclusie te kunnen trekken daaruit, maar als ik zaken zie zoals 'invalid record length at 3/BC4C0148: wanted 24, got 0', dan kan ik me voorstellen dat het onderliggende storagesysteem (lees; de SD-kaart) niet helemaal lekker meer is. Het Als ik de verschillende logs bekijk, lijkt het probleem ook toe te nemen. 
[postgresql-9.6-main.log.zip](https://github.com/dennissiemensma/dsmr-reader/files/2563088/postgresql-9.6-main.log.zip)

Overigens ben ik, mede omdat ik de RPi 3 eigenlijk wat langzaam vind voor voor DSMR én omdat ik ook andere zaken met de/een RPi wil gaan doen, bezig om een wat zwaardere machine (HTPC) in te richten met Debian. Hierop wil ik straks DSMR ook op laten draaien. Data komt dan op een SSD te staan.

dennissiemensma, 2018-11-09:
Bedankt voor je aanvulling. In dat geval verwacht ik dat de issues helemaal verholpen worden wanneer je switcht naar de SSD. 

Als ik de logfile zo bekijk, lijken de meldingen inderdaad vrij ernstig. Ik heb ze zelf nooit gehad in deze mate. **Zorg er in ieder geval voor dat je sowieso apart een goede back-up hebt, voordat de corruptie helemaal teveel wordt.**

Met name deze meldingen zijn zorgwekkend, die lijken aan te geven dat de complete DB crasht:
```
2018-11-08 18:30:07.571 UTC [412] LOG:  server process (PID 25739) was terminated by signal 11: Segmentation fault
2018-11-08 18:30:07.571 UTC [412] LOG:  terminating any other active server processes
2018-11-08 18:30:07.571 UTC [25737] dsmrreader@dsmrreader WARNING:  terminating connection because of crash of another server process
```

dennissiemensma, 2018-11-09:
Mocht je tzt nog hulp nodig hebben bij je nieuwe systeem, dan hoor ik het graag. De packages/commands voor installeren zijn als het goed is gelijk, dus ik verwacht wel dat je er verder uit komt.

closed 2018-11-09 by dennissiemensma