dsmrreader/dsmr-reader #552: dsmr installatie op de PI

hansv50, 2018-11-09:
Een newbee ben ik, maar wel enthousiast. Weet weinig meer van linux.

Ik probeer het dsmr project op mijn rasp pi 2 te installeren, Nu probeer ik stap 6 Virtualenv te maken.
De eerste stap (Make sure you are still acting as dsmr user (if not then enter: sudo su - dsmr)
Create folder for the virtualenv(s) of this user  mkdir ~/.virtualenvs) ging goed, de folder bestaat.

Bij de 2e stap: virtualenv ~/.virtualenvs/dsmrreader --no-site-packages --python python3
krijg ik de melding: -su: virtualenv: opdracht niet gevonden.

Ben ik iets vergeten te installeren of doe ik iets anders fout?

met vriendelijke groet,
Hans

dennissiemensma, 2018-11-10:
Hallo Hans, bedankt voor je vraag. Zou je een screenshot kunnen maken van de situatie? Ik vermoed dat het een klein dingetje is namelijk.

dennissiemensma, 2018-11-10:
Je screenshot is helaas niet zichtbaar. Wil je het nogmaals proberen? Wellicht direct in Github?

hansv50, 2018-11-10:
Hoi,
houd het liever bij de mail want Github snap ik niet zo veel van.
In de bijlage nu het screenshot, hopen dat het beter gaat.

dennissiemensma, 2018-11-10:
Ik denk dat de bijlage per mail wordt weggehaald. Je kunt hier gewoon een reactie plaatsen, net zoals bijvoorbeeld op Facebook.

Als alternatief mag je ook de inhoud van de complete terminal copy-pasten hier. Het wordt anders erg lastig om te achterhalen waar het in zit.

hansv50, 2018-11-11:
Goed, via github dan (werkt eigenlijk best makkelijk!)
Nogmaals het screenshot waar ik vastloop:

[image]

Wat ik nu krijg snap ik niet, maar in preview ziet er goed uit.

dennissiemensma, 2018-11-11:
Aah duidelijk. Het lijkt erop dat het programma `virtualenv` mist, heb je die geinstalleerd in stap 2?

Als `pi` gebruiker kun je dit uitvoeren om te dubbelchecken of alles geinstalleerd is:
```
sudo apt-get install -y postgresql postgresql-server-dev-all nginx supervisor git python3 python3-pip python3-virtualenv virtualenvwrapper
```

Let erop dat de regel doorloopt, misschien is dat ook wel het issue geweest? Als alternatief kun je het in twee stappen doen:
```
sudo apt-get install -y postgresql postgresql-server-dev-all nginx supervisor git 

sudo apt-get install -y python3 python3-pip python3-virtualenv virtualenvwrapper
```

hansv50, 2018-11-11:
Dennis,
Geweldig bedankt, het werkt.
Ik dacht dat ik de instructies zorgvuldig had uitgevoerd, toch niet dus.

Ik laat het je weten als alles werkt en bestel dan een PI 3+

Nogmaals hartelijk bedankt voor je geduld en hulp

Met vriendelijke groet,
Hans Verbeek

Van: Dennis Siemensma
Verzonden: zondag 11 november 2018 10:41
Aan: dennissiemensma/dsmr-reader
CC: Hans; Author
Onderwerp: Re: [dennissiemensma/dsmr-reader] dsmr installatie op de PI (#552)

Aah duidelijk. Het lijkt erop dat het programma virtualenv mist, heb je die geinstalleerd in stap 2?
Als pi gebruiker kun je dit uitvoeren om te dubbelchecken of alles geinstalleerd is:
sudo apt-get install -y postgresql postgresql-server-dev-all nginx supervisor git python3 python3-pip python3-virtualenv virtualenvwrapper
Let erop dat de regel doorloopt, misschien is dat ook wel het issue geweest? Als alternatief kun je het in twee stappen doen:
sudo apt-get install -y postgresql postgresql-server-dev-all nginx supervisor git 

sudo apt-get install -y python3 python3-pip python3-virtualenv virtualenvwrapper

hansv50, 2018-11-12:
> *Van: *Dennis Siemensma <notifications@github.com>
> *Aan: *dennissiemensma/dsmr-reader <dsmr-reader@noreply.github.com>
> *CC: *Hans <hjverbeek@gmail.com>; Author <author@noreply.github.com>
> .[image:

Dennis,
Heb het bijna voor elkaar, maar ik ben erachter dat ik een 2.x meter heb
(de data komt met 9600 baud). Dat kan ik veranderen, maar ik kan de
dataloggerconfigaratie niet vinden.
Kun (wil) je me nog een keer helpen, dan is als ik inlog op 192.168.1.1xx
misschien de 'bad gateway 502, verdwenen?

Al vast bedankt voor je geduld,
Hans

hansv50, 2018-11-12:
Hallo Dennis,
Inmiddels ben ik een aantal stappen verder en kan ik de website (192.168.1.113) openen. Dit ziet er zo uit:

[image]

niet zoals het bedoeld is denk ik.
Heb ook in het configuratiescherm de slimme meter gewijzigd (Ik heb dsmr 2.2), ziet er zo uit:

[image]

Maar ik heb nog steeds geen data.

Ik heb getest met putty met de volgende commando's:
cu -l /dev/ttyUSB0 -s 9600 --parity=none

dan zie ik de volgende regels:

[image]

Ongetwijfeld zie ik iets over het hoofd, maar wat?

Zou je me willen helpen?

met vriendelijke groet,
Hans

dennissiemensma, 2018-11-12:
* Voor het uiterlijk van de applicatie lijkt het erop dat de `static files` missen of niet goed werken. Ik neem aan dat je bij de installatie deze stappen hebt uitgevoerd?
```
sudo mkdir -p /var/www/dsmrreader/static
sudo chown -R dsmr:dsmr /var/www/dsmrreader/

sudo su - dsmr
./manage.py collectstatic --noinput
```

* Qua datalogger zou je eens logging kunnen aanzetten en de logsfiles bekijken. Meer informatie staat hier: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#logging

hansv50, 2018-11-12:
Dennis, het werkt!!

Uiterlijk is nu goed, mogelijk heb ik de opdracht './manage.py
collectstatic --noinput' niet als dsmr-user uitgevoerd. Krijg ook data
binnen, een kwestie van geduld lijkt me.

Nogmaals bedankt voor alle geduld en ondersteuning. Succes met de verdere
ontwikkeling van dit prachtige project.

met vriendelijke groet,
Hans

dennissiemensma, 2018-11-13:
Fijn om te horen dat het gelukt is en bedankt voor je bijdrage!

Zorg er wel voor dat je goede back-up's hebt, want de SD-kaartjes wisselen erg qua levensduur en het zou zonde zijn als je al je data op den duur kwijt raakt.

hansv50, 2018-11-13:
Hoi, ben zojuist aan de dropbox backup begonnen. Gaat ook weer lukken.
Hans

(in github heb ik het item gesloten)