dsmrreader/dsmr-reader #554: Kosten voor specifiek stuk grafiek zien

javl, 2018-11-17:
Net dsmr-reader ingesteld op een Raspberry Pi Zero W die ik nog had liggen en werkt tot nu toe perfect. Het is super interessant om live te kunnen zien wat je verbruik is.

### Wens
Op de dashboard pagina zie je nu je verbruik van de afgelopen x minuten waar je eventueel zelf nog een selectie in kan maken. Wat mij heel tof lijkt is om de verbruikskosten van die selectie te zien. Zo zou je in staat moeten zijn om bij een grootverbruiker (zeg een oven die grote pieken veroorzaakt in je gebruik, iets wat ik net had) een inschatting te maken van de kosten van het gebruik.

Ik ga binnenkort eens in de code duiken om te kijken hoe alles in elkaar zit, maar dacht ik post dit vast even mocht je hier zelf al ideeën over hebben.

dennissiemensma, 2018-11-18:
Bedankt voor je suggestie. Of dat mogelijk is hangt vooral af van de grafiek-software die ik gebruik, eCharts. Het is een vrij uitgebreid pakket, dus wellicht is e.e.a. te realiseren. Ik zal er een keer naar kijken.

dennissiemensma, 2019-11-08:
Ik heb een tijdje geleden nog gekeken naar features van de grafiek-software. Hoewel er veel mogelijk is, is het ook vrij complex en breekbaar. Voor nu laat ik deze wens dan voor wat het is.

closed 2019-11-08 by dennissiemensma