dsmrreader/dsmr-reader #555: Foutmelding terugzetten database

cranberryNL, 2018-11-22:
Omgeving: DSMR-reader 1.26 op een RPi 3.

Helaas is m'n database corrupt geraakt. Ik zag me dus genoodzaakt de restore procedure zoals ergens in de wiki beschreven uit te voeren.

Helaas lijkt er in de backup ook problematische waarden terecht te zijn gekomen. Ik krijg nl. deze melding:
ERROR:  numeric field overflow
DETAIL:  A field with precision 9, scale 3 must round to an absolute value less than 10^6.
CONTEXT:  COPY dsmr_datalogger_dsmrreading, line 10249, column electricity_delivered_1: "24880000000000000000000000000000000000000000000000000000000000000.000"

Iemand enig idee hoe ik zulke waarden uit de backup kan krijgen?

cranberryNL, 2018-11-22:
In een verloren uurtje heb ik vanmiddag even de .sql backupfile met Notepad++ bekeken. Daarin valt me o.a. op dat ID's niet uniek zijn. Een aantal ID's kom ik meerdere malen tegen.
Verder staat er als laatste regel van de backupfile ongeldige data. Ik ga proberen deze eruit te halen en kijken of ik de backup dan wel kan terugzetten.

Het issue komt trouwens met alle backups van deze week voor. Wellicht is het gerelateerd aan feit dat de database vaak in recovery mode is. Omdat het de laatste tijd weer redelijk stabiel draaide, heb ik nog geen prioriteit gegeven aan de migratie naar andere hardware (Intel NUC met SSD).

Ter leringhe ende vermaeck heb ik het data-gedeelte uit de backup even toegevoegd.

[meter_data.zip](https://github.com/dennissiemensma/dsmr-reader/files/2608648/meter_data.zip)

dennissiemensma, 2018-11-22:
Wil je eens een mailtje sturen naar github apenstaartje dennissiemensma punt nl? Volgens mij heb ik je dag en uur-statistieken van 12 juli t/m 19 november, maar dat kun je zelf het beste beoordelen. Ik wil je ook geen valse hoop geven.

dennissiemensma, 2018-12-24:
Ik heb inmiddels een eerste opzet werkend voor een (uitgeklede) backup per e-mail. De bijlage was een DB-dump van ~400 KB met daarin (mijn) 3 jaar gegevens qua dag- en uurstatistieken. De belangrijkste historische data.

Ik moet alleen nog wel een hoop refactoren zodat het allemaal automatisch gaat. Verder zorg ik voor diverse opties, zoals een e-mail per dag, week, twee weken of maand.

dennissiemensma, 2019-04-28:
@cranberryNL het heeft een tijd geduurd, maar ik heb deze feature inmiddels zo goed als klaar. Wil je hem zelf ook nog testen voor die tijd? Ik hoor graag of je daar behoefte aan hebt.

cranberryNL, 2019-04-29:
Heel mooi Dennis! Graag test ik even. Als je aangeeft hoe ik dat kan doen, zal ik dat op zo kort mogelijke termijn even doen.

dennissiemensma, 2019-04-29:
Top. Ik zal hem eerst zelf nog even extra uitproberen, maar dan zal ik je tzt een seintje geven als het goed genoeg werkt om te testen.

dennissiemensma, 2019-04-29:
@cranberryNL je kunt overigens alvast updaten naar de laatste release versie, 2.0.2, mocht je dat nog niet gedaan hebben. Let op dat je wel minimaal Python 3.5 draait, anders crasht de applicatie.

Hiermee zit je dan alvast zo dicht mogelijk tegen de nieuwe release aan. Later kun je dan wisselen van branch, maar ik laat het hier eerst even een dagje draaien.

cranberryNL, 2019-04-29:
Prima, ik wacht ff op en seintje van jou. Ik draai overigens al op v2.0.2, dus dat komt helemaal goed!

dennissiemensma, 2019-04-30:
Ik kom toch nog wat dingetjes tegen, vooral in foutsituaties, dus ik moet dat eerst verbeteren.

dennissiemensma, 2019-04-30:
Tevens lijkt het me goed dat het mogelijk is om via Gmail de mailtjes te kunnen versturen. Daarvoor moet ik nog eea inbouwen: https://developers.google.com/gmail/imap/xoauth2-protocol

dennissiemensma, 2019-05-13:
Ik ben al enige tijd bezig met een verbeterde interface voor de instellingen (eerst alleen voor de e-mailinstellingen van dit issue). 

Echter kost dat dusdanig veel tijd om de interface goed, duidelijk en foutbestendig te maken, dat ik dit voor nu laat zitten. Opzich wel jammer, want de admin-interface van dit project is inmiddels verre van ideaal voor een aantal instellingen, maar het is dan maar even niet anders.

Ik zal proberen om de documentatie dan helder genoeg te maken voor deze feature, en om toch een knop of iets te maken om de e-mailinstellingen te kunnen testen.

dennissiemensma, 2019-05-13:
Todo:
* [x] Backend
* [x] Interface
* [ ] Docs voor gebruik

dennissiemensma, 2019-05-15:
@cranberryNL het heeft even geduurd, maar inmiddels kun je het uitproberen.

Normaal gesproken tak ik dergelijke features af van de laatst gereleasde versie (`v2.0.2`), maar in dit geval is het uiteindelijk onderdeel geworden van de `development` branch en daarmee dus ook indirect de eerstvolgende release. Dit betekent dat je dus praktisch de complete volgende release hiermee draait.

Overigens is het goed getest en draai ik hem zelf al een paar dagen. 

1) Desnietemin graag **dubbelchecken of je een goede DB-backup** hebt, mocht er om wat voor reden dan ook iets onverwachts gebeuren.

2) Vervolgens staat hier alvast een kort verhaaltje in de DEV-versie van de documentatie. Deze is tevens in een nieuw jasje, dus niet schrikken, de inhoud is gelijk:
* https://dsmr-reader.readthedocs.io/nl/development/admin/email.html
* https://dsmr-reader.readthedocs.io/nl/development/admin/email_backup.html

3) Daarna wisselen van versie qua applicatie kan met:
```
sudo su - dsmr
git fetch
git checkout -b development origin/development
./deploy.sh
```

Als het goed is draai je dan versie `v2.1b15`.

4) Ik hoor graag of het lukt. 

Afhankelijk van je eigen e-mailsetup of wat je ook gebruikt kan het mee- of tegenvallen.

dennissiemensma, 2019-05-15:
Terugrollen kan overigens met:

```
sudo su - dsmr
sh dsmrreader/provisioning/downgrade/2.1.0-2.0.2.sh
git checkout master
./deploy.sh
```

Je zou dan weer op `v2.0.2` moeten zitten. IK verwacht niet dat je het nodig gaat hebben overigens.

closed 2019-05-20 by dennissiemensma

dennissiemensma, 2019-05-20:
Gereleased in [v2.1.0](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v2.1.0)

cranberryNL, 2019-05-22:
Hoi Dennis,

Sorry voor mijn late reactie. Pas eergisteren heb ik tijd vrij kunnen maken de nieuwe versie en onderstaande functionaliteit te testen. Tot op heden lijkt alles prima te werken. 

Veel dank voor je inzet en de toevoegingen!

Met vriendelijke groeten,
Aza

From: Dennis Siemensma 
Sent: Wednesday, May 15, 2019 9:32 PM
To: dennissiemensma/dsmr-reader 
Cc: cranberryNL ; Mention 
Subject: Re: [dennissiemensma/dsmr-reader] Ondersteuning voor back-up per e-mail (#555)

@cranberryNL het heeft even geduurd, maar inmiddels kun je het uitproberen.

Normaal gesproken tak ik dergelijke features af van de laatst gereleasde versie (v2.0.2), maar in dit geval is het uiteindelijk onderdeel geworden van de development branch en daarmee dus ook indirect de eerstvolgende release. Dit betekent dat je dus praktisch de complete volgende release hiermee draait.

Overigens is het goed getest en draai ik hem zelf al een paar dagen.

  1.. Desnietemin graag dubbelchecken of je een goede DB-backup hebt, mocht er om wat voor reden dan ook iets onverwachts gebeuren.

  2.. Vervolgens staat hier alvast een kort verhaaltje in de DEV-versie van de documentatie. Deze is tevens in een nieuw jasje, dus niet schrikken, de inhoud is gelijk:

  a.. https://dsmr-reader.readthedocs.io/nl/development/admin/email.html 
  b.. https://dsmr-reader.readthedocs.io/nl/development/admin/email_backup.html 
  3.. Daarna wisselen van versie qua applicatie kan met: 
sudo su - dsmr
git fetch
git checkout -b development origin/development
./deploy.sh
Als het goed is draai je dan versie v2.1b15.

  4.. Ik hoor graag of het lukt. 
Afhankelijk van je eigen e-mailsetup of wat je ook gebruikt kan het mee- of tegenvallen.