dsmrreader/dsmr-reader #556: Mogelijkheid om jaren te vergelijken in archief

luchtenburger, 2018-11-24:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.X.Y`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi X...`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v...`

### Bug, wens of iets anders
* Bijvoorbeeld stappen hoe een probleem te reproduceren is...
* Of simpelweg een beschijving van een feature die je graag wenst te zien...

dennissiemensma, 2018-11-25:
Bedankt voor je verzoek. Ik denk dat ik wat context mis van wat je exact bedoelt. Momenteel kun je namelijk al in de Compare/Vergelijk-pagina dagen, maanden of jaren met elkaar vergelijken. Zie een voorbeeld hieronder:

[image]

Zou je je verzoek willen verduidelijken?

luchtenburger, 2018-11-25:
Ik ben op zoek naar de mogelijkheid om trends te zien over meerdere jaren. In het archief kun je nu kiezen voor dagen, maanden of jaren, ik zou graag nog een niveau hoger willen. 
In dat hogere niveau zou je dan een x aantal (of alle beschikbare?) jaren kunnen zien in één grafiek.
Ik hoop dat het zo duidelijk is, anders knutsel ik even een voorbeeld in Excel in elkaar.

dennissiemensma, 2020-11-10:
Ik ben hier de afgelopen twee jaar niet aan toegekomen en ik zie daar op korte termijn ook geen ruimte voor. Hoewel er zeker wat te zeggen is voor de feature, denk ik dat je hiervoor het beste kunt kijken naar een tool waar je zelf wat meer invloed hebt op de data(aggregatie) en hoe dat getoond wordt. 

Een suggestie is bijvoorbeeld de data uit DSMR-reader koppelen/exporteren naar InfluxDB en dan visueel plotten met Grafana. Dan ben je ook wat minder afhankelijk van een implementatie met je wensen, gezien het vooral aggregatie van data is.
In de eerdere jaren van DSMR-reader was ik zelf nog niet zo bekend met dergelijke tools, anders had ik het vermoedelijk initieel al opgeslagen in iets als ElasticSearch of InfluxDB (ipv een relationele database) en dan daar met een visualizer zoals Kibana of Grafana de grafieken in plotten.

closed 2020-11-10 by dennissiemensma