dsmrreader/dsmr-reader #562: Feature Request: API voor live gas verbruik

Proxx, 2018-12-06:
ik zou graag ook mijn gas verbruik inzichtelijk hebben via de api

**GET - consumption/gas-live**

```
{
    "timestamp": "2018-12-06T09:39:00Z",
    "currently_returned": 0,
    "currently_delivered":xxx,
    "cost_per_hour": 0.xx,
}
```

dennissiemensma, 2018-12-06:
Bedankt voor je verzoek. Ik heb het gasverbruik destijds bewust niet opgenomen als 'live' omdat het verre van live is. Voor oude meters is er pas na een uur een update beschikbaar en voor nieuwere meters na een kwartier. 

Heb je ook al gekeken of het dagtotaal (tot nu toe) wellicht een optie voor je is? Die is wat minder onderhevig aan veranderingen. Gedurende de dag zie je dan je gasverbruik oplopen.

https://dsmr-reader.readthedocs.io/nl/latest/api.html#get-consumption-today

Mocht je toch een vertraagde versie willen hebben voor 'live' gas, dan zal ik het overwegen.

Proxx, 2018-12-07:
het klopt mijn meter geeft ong elke 10min. een gasmeting geloof ik.
met die dagtotalen heb ik vandaag gespeeld om zo tot het zelfde resultaat te komen alleen lukt dit (nog) niet. als je de mogelijkheid ziet om het in DSMR api te verwerken heel graag als het is niet te werk is,

dennissiemensma, 2018-12-07:
Prima. Ik laat de feature dan openstaan om een keer in te bouwen.

dennissiemensma, 2019-04-19:
Ik heb hier nu even opnieuw naar gekeken en ik zie nu waarom ik dit destijds niet gebouwd heb.

De telegrammen uit de meter geven het huidige verbruik aan in een apart veld. Voor gas is dat niet aanwezig, alleen de gasstand.

Deze gegevens zijn echter ook beschikbaar via de API. Alleen zie ik nu dat die API-calls nooit zijn opgenomen in de documentatie op Read The Docs. Hah. Dat is best wel spijtig. Ze staan overigens wel in de auto-documentatie op `/docs/v2/#consumption-gas-list`.

Ik zal kijken of ze nog steeds werken, want in principe kun je daarmee makkelijk je "live" verbruik uitrekenen. Daarna zal ik ze sowieso toevoegen aan de documentatie op Read The Docs.

dennissiemensma, 2019-04-19:
Volgens mij werken ze nog steeds goed, dus kun je ze gebruiken. Alle telegrammen worden eerst in losse telegram-records opgeslagen. Daarna wordt elektriciteit en gas in aparte records gesplitst, ook wel 'consumption' genoemd binnen het project. Deze records zijn via de API te benaderen, maar waren dus niet ontsloten in de API docs.

De 'gas consumption' is degene die je wilt weten. Dit zijn alle records die elk uur (of in jouw geval waarschijnlijk sneller) aangemaakt worden. Via de API is het dan een kwestie van:
* Resultset limiteren tot 2
* Aflopend sorteren op tijdstip (meest recente eerst)

Call: `/api/v2/consumption/gas?limit=2&ordering=-read_at`

```
{
    "count": 28793,
    "next": "http://raspberrypi/api/v2/consumption/gas?limit=2&offset=2&ordering=-read_at",
    "previous": null,
    "results": [
        {
            "id": 28857,
            "read_at": "2019-04-19T08:00:00+02:00",
            "delivered": "2850.538",
            "currently_delivered": "0.233"
        },
        {
            "id": 28856,
            "read_at": "2019-04-19T07:00:00+02:00",
            "delivered": "2850.305",
            "currently_delivered": "0.000"
        }
    ]
}
```

Vervolgens kun je dan zelf de twee meterstanden van elkaar aftrekken om je 'live' gasverbruik in te zien.

dennissiemensma, 2019-04-19:
Mocht je hier desondanks niet uitkomen of nog aanvullingen hebben dan hoor ik het graag! Ik maak even een apart issue om deze API-calls op te nemen in de documentatie.

dennissiemensma, 2019-04-19:
Nog een aanvulling. Ik zie dat je simpelweg gewoon het laatste record kan opvragen, omdat daar het meest recente verbruik in zit. Dus:

`/api/v2/consumption/gas?limit=1&ordering=-read_at`

En dan het veld `"currently_delivered": "0.233"` (uit bovenstaand voorbeeld).

dennissiemensma, 2019-04-19:
Ik overweeg om alsnog een API-call te maken, die bovenstaande eigenlijk onderwater doet.

dennissiemensma, 2019-04-19:
De docs voor de bestaande API-call staan nu op de dev-docs: 

https://dsmr-reader.readthedocs.io/nl/development/api.html#get-consumption-gas

dennissiemensma, 2019-04-21:
Gemerged naar `development` voor volgende release.

closed 2019-04-21 by dennissiemensma

joep2000, 2019-05-13:
Hi Dennis,
Is het veld 'gas_currently_delivered' ook beschikbaar als een MQTT veld?
Mijn meter spuugt de waarde elke 5 minuten uit, en daar kan je al wat meer mee!

dennissiemensma, 2019-05-13:
Hallo Joep, dat is wellicht mogelijk. Ik denk dat je daarvoor dan het beste even een apart issue kunt maken hier.