dsmrreader/dsmr-reader #564: Dropbox doesn't sync anymore

raoulteeuwen, 2018-12-16:
### Jouw omgeving
* DSMR-reader versie: `v1.26.1`
* Hardware: `RaspberryPi 3b...`
* DSMR-protocol v4 or v5 ... (how can i check?)

Mijn Rpi werd een paar weken terug ontoegankelijk. Heb daarop nieuwe SD-kaart klaar gemaakt en daarmee Rpi weer opgestart. Op zich draait oa Home Assistant en DSMR, maar de dropbox-sync werkt niet meer. Backups worden wel lokaal aangemaakt. Pi is al paar keer gereboot. Waar kan ik nog naar kijken?

dennissiemensma, 2018-12-16:
Bedankt voor je melding. Ik denk dat je het beste even kunt kijken in de logs of je daar activiteit ziet:
Hiervoor zul je even logging moeten aanzetten:
https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#logging

Daarna kun je forceren dat de applicatie checkt of er iets geupload moet worden:
```
sudo su - postgres
psql dsmrreader

# Voer deze query uit om de waarden op een dag terug te zetten:
update dsmr_backup_dropboxsettings set latest_sync = now() - interval '24 hours', next_sync = now() - interval '24 hours';

# Nu kun je deze doen, om te kijken of de applicatie ze bijgewerkt heeft. Doorgaans
# duurt het slechts een paar seconden voordat deze waarden weer 'terug' gezet zijn vandaag:
select latest_sync, next_sync from dsmr_backup_dropboxsettings;

# Houd er rekening mee dat de tijden waarschijnlijk in UTC zijn 
# (achteraan de tijd staat dan iets van "+00").
```

Daarna kun je in de logs kijken of zoeken: `sudo zgrep Dropbox /var/log/supervisor/dsmr_backend.log*`

Ik heb het net even 2x geprobeerd en zie ik twee resultaten:
```
pi@raspberrypi:~ $ sudo zgrep Dropbox /var/log/supervisor/dsmr_backend.log*
/var/log/supervisor/dsmr_backend.log:[2018-12-16 12:41:45,228] INFO      - Uploading file to Dropbox: /home/dsmr/dsmr-reader/dsmrreader/../backups/dsmrreader-postgresql-backup-Sunday.sql.gz
/var/log/supervisor/dsmr_backend.log:[2018-12-16 12:43:31,723] INFO      - Uploading file to Dropbox: /home/dsmr/dsmr-reader/dsmrreader/../backups/dsmrreader-postgresql-backup-Sunday.sql.gz
```

Ik hoor graag je hier hier iets in terugvindt.

raoulteeuwen, 2018-12-16:
Dank voor je snelle reactie ... Fyi: ik zit zelf altijd nog te pielen met in welke prompt ik zit en hoe ik er dan uit komt: exit, quit, /q, logout ... 

Ik heb het loglevel aangepast. Ik kreeg na het update-statement een UPDATE 1. Na de select een ascii-tabel met ca de huidige tijd in kolom 1. Liep wel een uur achter. raspi-config gedaan en timezone goed gezet. Nogmaals eea geprobeerd, maar tijd bleek nog fout te staan.

Rpi uit en aan gezet. Daarna op URL waar ik DSMR had "502 Bad Gateway". Met sudo netstat -tulpn denk ik dan te zien waar er evt servers draaien. Gezocht naar nginx of postgres-vermeldingen. Ik zie wel een

 0.0.0.0:8180            0.0.0.0:*               LISTEN      546/nginx: master p 

maar op <ip=adres>:8180 (waar dsmr draaide), krijg ik nu dus bad gateway :-( ... one of those days hier, want na update van HA gisteren ook al veel dingen die niet meer werken.

raoulteeuwen, 2018-12-16:
Ik zie nu dat ik 1 ding vergeten ben: na aanpassen settings.py ivm DEBUG-level, post-deply-sh draaien. Als ik dat nu probeer krijg ik

 --- Checking whether VirtualEnv is activated.
     [i] ----- Activating 'dsmrreader' VirtualEnv...
./post-deploy.sh: line 10: /home/pi/.virtualenvs/dsmrreader/bin/activate: No such file or directory
     [!] FAILED to switch to 'dsmrreader' VirtualEnv (is it installed?)

dennissiemensma, 2018-12-16:
In reactie op je laatste bericht, heb je wel eerst `sudo su - dsmr` gedaan? Je kunt dat nog eerst proberen. Als het goed is gaat het dan beter.

raoulteeuwen, 2018-12-16:
Dank. Sorry, voor jou ws heel normaal. Leverde nu

  File "/home/dsmr/dsmr-reader/dsmrreader/settings.py", line 40
    LOGGING['loggers']['commands']['level'] = 'DEBUG'
    ^

Dus heb settings.py aangepast. (ik had alleen # weg gehaald)

Nu krijg ik uit de select

          latest_sync          |           next_sync           
-------------------------------+-------------------------------
 2018-12-15 17:31:30.892401+00 | 2018-12-15 17:31:30.892401+00
(1 row)

Hoe forceer ik een next-sync in de toekomst? Of pakt hij bij een next-sync in het verleden die op enig moment alsnog? Als ik een date opvraag op de pi krijg ik momenteel de goede datum en tijd.

dennissiemensma, 2018-12-16:
Als de Dropbox-instellingen goed staan (volgens mij alleen een API-key), dan pakt hij inderdaad de sync-datum op elk moment uit het verleden (vandaar de -24 uur). Dat geldt voor de meeste features in het backend-proces.

Dat verwerken zou echter vrij vlot moeten gaan. Als die na enkele minuten nog niet is bijgewerkt, dan lijkt het erop dat die het niet doet. Begrijp ik het goed dat wanneer je nu opnieuw de SELECT uitvoert, je nu nog steeds dezelfde datum/tijden ziet (15 december)?

raoulteeuwen, 2018-12-16:
jeetje, dit voelt heel stom .... sorry

dennissiemensma, 2018-12-16:
Maakt niet uit, ik ben blij dat je de oorzaak hebt kunnen vinden! En aan de andere kant, het debuggen is wat lastiger via de logs en niet heel makkelijk. Mogelijk dat ik in de toekomst wat betere foutafhandeling in de interface zelf kan laten zien.

Vergeet niet om je `settings.py` weer aan te passen en de regel mbt DEBUG logging uit te schakelen en een reload te doen.

closed 2018-12-16 by dennissiemensma