dsmrreader/dsmr-reader #565: Mogelijke verbetering bij vergelijken periodes

voipmeister, 2018-12-21:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.26.1`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 3`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v...`

### Verbetering
Bij het vergelijken van 2 periodes wordt niet gekeken/gemeld of de periodes wel compleet zijn. De verbetering zou dan (mogelijk) bestaan uit het tonen van (in dit geval) twee meldingen:
* Let op, de data voor 2017 is niet compleet! (eerst gevonden data is van 14-07-2017)
* Let op, de data voor 2018 is (nog) niet compleet

Voorbeeld: midden in juli 2017 ben ik begonnen met de huidige installatie. Als ik 2017 met 2018 vergelijk, gaat de vergelijking mank:
* Er zijn pas gegevens vanaf 14 juli 2017
* 2018 is nog niet afgelopen

Het screenshot is van alleen de vergelijking van juli 2017 en juli 2018, ter onderbouwing van het eerste punt:

[image]

En hierbij de archiefdata van juli 2017:

[image]

Als ik nog wat mag verduidelijken of meer mag meednken hoor ik het graag :-)

dennissiemensma, 2018-12-21:
Bedankt voor je suggestie. Dat is inderdaad iets waar ik geen rekening mee heb gehouden. Het lijkt me goed om die melding dan toe te voegen.

voipmeister, 2019-01-24:
Gezien mijn meeste recente SD-kaart crash en de herinstallatie + restore (ik had nog geen tijd genomen de setup te verbeteren) toch even mijn eigen suggestie 'bevestigen' in de zin dat ik nu 2 weken mis i.v.m. kapotte recente backups. Bij vergelijkingen heb ik dus een gat van 2 weken helaas, dat ga ik vast vergeten als ik 2019 met 2018 ga vergelijk begin 2020 ;-)

dennissiemensma, 2020-03-10:
Ik heb dit in de volgende release opgelost met een melding:

[image]

closed 2020-03-10 by dennissiemensma

dennissiemensma, 2020-03-19:
Uitgebracht in [v3.7](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v3.7.0)