dsmrreader/dsmr-reader #567: buitentemperatuur via externe thermometer

hsleewis, 2018-12-24:
Zou het mogelijk zijn om een externe thermometer uit te lezen? Ik merk dat de internet weer services er toch altijd naast zitten. En volgens mij is de ingebouwde weather provider ook per 1 januari niet meer bereikbaar.

dennissiemensma, 2018-12-24:
Bedankt voor je verzoek. Ik begrijp de wens heel goed, maar ik kan helaas geen ondersteuning toevoegen voor het importeren van externe data. De huidige integratie met Buienradar had ik, achteraf gezien, niet moeten toevoegen aan DSMR-reader. Dit project focust puur op het uitlezen van de slimme meter en het kunnen bekijken/exporteren van die data op zoveel mogelijk manieren.

Ik wist trouwens niet dat Buienradar er per 1 januari mee stopt. Kan ik dat ergens teruglezen?

hsleewis, 2018-12-24:
Excuus,.. Ik heb 't verkeerd,.. het weersysteem van tuinbewatering systeem (draait op dezelfde pi) is weather.com en die stoppen er mee, dacht eerst dat DSMR ook dat systeem gebruikte. :)

dennissiemensma, 2018-12-24:
Aah precies, dat had ik al wel een keertje meegekregen (jammer trouwens dat ze ermee stoppen). In dat geval laat ik dit voor wat het is en focus ik mij op de andere openstaande wensen. Groet!

closed 2018-12-24 by dennissiemensma